collabora/libsurvive #4: Improve Light Disambiguator

cnlohr, 2016-12-16:
The light disambiguator is right now very poor.  Located in survive_process.c it is rather simplistic, and something terrible happens when you power on the watchmen controllers.

Would anyone mind taking a whack at this to help me out?

[code]
static void handle_lightcap( struct SurviveObject * so, struct LightcapElement * le )
{
	struct SurviveContext * ct = so->ctx;

	if( le->type != 0xfe || le->length < 50 ) return;
	//le->timestamp += (le->length/2);

	if( le->length > 2100 ) //Pulse longer indicates a sync pulse.
	{
		int32_t deltat = (uint32_t)le->timestamp - (uint32_t)ct->last_photo_time;
		if( deltat > 2000 || deltat < -2000 )		//New pulse. (may be inverted)
		{
			if( le->timestamp - ct->last_photo_time > 80000 )
			{
				ct->last_photo_time = le->timestamp;
				ct->total_photo_time = 0;
				ct->total_photos = 0;
				ct->total_pulsecode_time = 0;
				ct->lightproc( so, le->sensor_id, -1, 0, le->timestamp, deltat );
			}
		}
		else
		{
			ct->total_pulsecode_time += le->length;
			ct->total_photo_time += deltat;
			ct->total_photos++;
		}
	}
	else if( le->length < 1200 && le->length > 40 && ct->total_photos )
	{
		int32_t dl = (ct->total_photo_time/ct->total_photos);
		int32_t tpco = (ct->total_pulsecode_time/ct->total_photos);
		//Adding length 
		int32_t offset_from = le->timestamp - dl - ct->last_photo_time + le->length/2;

		//Long pulse-code from IR flood.
		//Make sure it fits nicely into a divisible-by-500 time.
		int32_t acode = (tpco+125)/250;
		if( acode & 1 ) return;
		acode>>=1;
		acode -= 6;

		if( offset_from < 380000 )
		{
			ct->lightproc( so, le->sensor_id, acode, offset_from, le->timestamp, le->length );
		}
	}
	else
	{
		//Runt pulse.
	}
}
[/code]

jpicht, 2016-12-16:
I would guess the best way of recovering the sync pulses is to build something like a PLL and maybe even additionally sync on the sequence of the pulses (iirc there are different length sync pulses, that come in a well known sequence?). Standard PLL implementations probably will have a problem with shadowing all the sensors / the sensor used to synchronize the clock.

It could also be helpful to identify the times many sensors are triggered at the "same time", but this might not work really well with the handheld controllers, because iirc the "Ben compression" omits pulses if too many sensors are triggered in a very rapid succession.

cnlohr, 2016-12-16:
(1) I just realized I am using system's sync time.  That's why turning on the controllers messes everything up.  I will switch to having each time domain have its own timer monitor.

(2) @jpicht: I think this is something that would take a fair bit of thought and effort to implement.  If I record raw pulses in a variety of distances, would you be up for trying to improve the disambiguator?  Additionally, do you have a vive?  Perhaps you could take the readings yourself?

Charles

LeeCookGHA, 2016-12-17:
On a lot of microcontrollers there is a specific bit count assembly instruction does the '86 have the equivalent?
If so, you could keep track of which are lit bitwise and just trigger a flood count based on a lit threshold with minimal processing.

cnlohr, 2016-12-17:
Because I am unfamiliar with this, it would take me a fair bit of time to implement.  Sounds like this is something you'd be well suited to making work and doing a push... I will still try to collect some good raw light data and post it here.

cnlohr, 2016-12-17:
I've taken lightcap data from the watchman, since that seems to be the hardest thing to deal with at the moment.   I held the watchman still a ways away from both of the lighthouses, then I moved it close to one, then close to the other.  @jpicht @LeeCookGHA  if you could poke through this and see if you can improve the disambiguator, that would be awesome!

[raw_light_data_from_watchman.csv.zip](https://github.com/cnlohr/libsurvive/files/659085/raw_light_data_from_watchman.csv.zip)

jpicht, 2016-12-20:
I will have a poke at it. What do the columns mean? I would guess it's

<controller name> <sensor id> <???> <length> <"absolute" time of rising/falling edge?!>

cnlohr, 2016-12-20:
It's 254 here, hard-coded, always.  It's almost always 254 from the HMD.  I'm not sure what it actually means.  But other than that that's all correct.

cnlohr, 2016-12-20:
@jpicht I am not sure what's actually going on there.  Mostly with the output. I am expecting to get the time from sync pulse to light passing over each sensor.  What I "think" I'm seeing is the time between sync pulses?

jpicht, 2016-12-20:
Well, yeah, it is a very primitive approach to detect the sync pulses by their frequency and length. The number displayed is related to the detected frequency.

cnlohr, 2016-12-20:
So, currently it ignores the sweep pulses? (Which is ok since that is part of developing a better disambiguator, just trying to understand)

jpicht, 2016-12-20:
Yes, exactly. The version I pushed yesterday night was all about detecting the sync pulses. An actually usable implementation of your interface is in the works at the moment.

cnlohr, 2016-12-20:
Ah! Cool! Ok!  Good to know I may not need to write that myself!!!  Thank you!

cnlohr, 2016-12-20:
One other note - please be as sensitive as you can to run-time.  This code will need to be very tight as it is run more frequently than the code that accepts this code's output, and this will need to be a part that runs on an embedded microcontroller.

jpicht, 2016-12-20:
Yeah, I know :) I've watched ALL THE STREAMS. Not live though, as in Germany the time offset makes that kind of inconvenient ...

That's why I'm doing it in C instead of C++ (which I am way more familiar with, really starting to miss the std containers!). This way it should be easily portable to the ESP.

I will not put too much emphasis on runtime though, before I know it really is going to be better than your current approach. If we know it works, we can start crunching it down.

jpicht, 2016-12-20:
Do have any data with only one lighthouse? I am currently a little confused ...

cnlohr, 2016-12-20:
I think I need to record a LOT more data for this, like HMD, WM, etc.  Still want to do both lighthouses.  What would having only one benefit?

jpicht, 2016-12-20:
I am currently trying to manually check if the results my code produces make sense. But two lighthouses both flashing away make that kind of extra hard ...

cnlohr, 2016-12-20:
You should be able to filter fairly easily based on the length-of-sync-pulse.

jpicht, 2016-12-20:
I pushed my current version to the disambiguate branch in my fork. Seems pretty stable to me. The output should be a little more self-explanatory now.

Now I only need to integrate it with your code, so someone who has a vive can test in the real world.

jpicht, 2016-12-20:
I think the order of the pulses is not always correct. Could you check, if the pulses are fed into handle_lightcap in the right order, after they come from the Ben-decoder?

cnlohr, 2016-12-20:
Perhaps that was the same issue I was running into with my code.  It seemed that the sync pulses were coming in at an unusual order, and it made the relative timings "jump" around.  I hope we do not have to descend back into the depths of the bencoder.

jpicht, 2016-12-20:
Your code seems to mistake these pulses for sync pulses if they are >2100 ticks otherwise it ignores them in two different ways.

I am thinking about creating a buffer to get them in order. Just always process the oldest and then add the current one to the buffer. A buffer size of 8 should be enough for this. Would add latency to the event chain though ...

jpicht, 2016-12-20:
The pulses in a batch seem to be ordered by the falling edge (timestamp + length) instead of the leading edge.

cnlohr, 2016-12-20:
Interesting... That is something I may be able to do something about, though admittedly, it would take re-recording.  Thank you for figuring that out!!!

cnlohr, 2016-12-20:
(1) Does your current disambiguator, implemented as-is assume the pulse times are at the end or beginning?  Will something need to be undone once I correct the edges?

(2) Since this is almost exclusively your work, please include a copyright notice at the top with year, your name, and a note that the file is licensed under the MIT license.

jpicht, 2016-12-20:
(1) It is implemented without any reordering. The pulse order is assumed to be correct.

(2) done & pushed.

I sorted the data with a python script for testing purposes: [raw_light_data_from_watchman.sorted.csv.zip](https://github.com/cnlohr/libsurvive/files/664799/raw_light_data_from_watchman.sorted.csv.zip)

cnlohr, 2016-12-21:
Hmm, this is not working in the final system at the moment.  Even on the HMD.  I guess I am concerned you were expecting the light pulse times going into the headset to be centered differently.

cnlohr, 2016-12-21:
I've made a bunch of changes trying to massage the algorithm in more, but it's still not working :-/.  Needs more comments in future.  I've set the preprocessor directive USE_OLD_DISAMBIGUATOR.  I'm going to try to push to my master and collect more data.  Maybe from there you can help determine what's wrong with the new disambiguator.

cnlohr, 2016-12-21:
[new_lightcap_data.csv.zip](https://github.com/cnlohr/libsurvive/files/665514/new_lightcap_data.csv.zip)

Here's my new lightcap data, includes for HMD and watchmen dongles. @jpicht  see if you can get this data to behave correctly?

jpicht, 2016-12-21:
Hm ... really inconvenient, that I cannot test it myself... I will have a look at the new data. It worked just fine with the old data I had. :(

Some guesses:
- the buffer size might be to small (`DIS_NUM_VALUES`)
- the threshold for the lock might be too high (`disambiguator_step`, `if (d->scores[idx] >= 30)`)
- the maximum jitter to low (`disambiguator_find_nearest`, third parameter)

jpicht, 2016-12-21:
Just had a _very_ brief look at the pulses that arrive at the HMD and I can already tell that you probably have something reflective in the room. That gives you a spurious sync pulse ~20000 ticks probably before, maybe after the "real" sync pulse. Also the jitter seems to be about twice as high as in the older WM data.

It should be possible to remedy that by also locking onto the expected sequence of pulse lengths. Will have a closer look at that tonight.

LeeCookGHA, 2016-12-21:
I’ve done a little research and I think this is the way each device needs to be decoded:
For this valid sync pulses are (in ticks) S:SKIP, D:DATA, A:AXIS:
2880:3024, // SYNC_TYPE_0 - PS0, PD0, PA0
3380:3524, // SYNC_TYPE_1 - PS0, PD0, PA1
3880:4024, // SYNC_TYPE_2 - PS0, PD1, PA0
4380:4524, // SYNC_TYPE_3 - PS0, PD1, PA1
4880:5024, // SYNC_TYPE_4 – PS1, PD0, PA0
5380:5524, // SYNC_TYPE_5 – PS1, PD0, PA1
5880:6024, // SYNC_TYPE_6 – PS1, PD1, PA0
6380:6524  // SYNC_TYPE_7 – PS1, PD1, PA1

// RE-SYNC to LH process (also start-up waiting for the LH “babble” to settle down….)
// If any of the *Checks* fail then re-start the process

Wait for a valid sync pulse length then
	Take a snapshot of the arrival time (T0) and the type (S0)
Ignore all pulses for 12000 ticks
Wait for a valid sync pulse length then
	Take a snapshot of the arrival time (T1) and the type (S1)
*Check* T1 is within 18720: 21600 of T0
If S0 is SYNC_TYPE_0 or SYNC_TYPE_2
	*Check* S1 is SYNC_TYPE_4 or SYNC_TYPE_6
If S0 is SYNC_TYPE_1 or SYNC_TYPE_3
	*Check* S1 is SYNC_TYPE_5 or SYNC_TYPE_7
If S0 is SYNC_TYPE_4 or SYNC_TYPE_6
	*Check* S1 is SYNC_TYPE_0 or SYNC_TYPE_2
If S0 is SYNC_TYPE_5 or SYNC_TYPE_7
	*Check* S1 is SYNC_TYPE_1 or SYNC_TYPE_3
// ###### LASER SWEEPS HAPPEN HERE #######
 *Check* no pulses of greater length than 1200 ticks arrive between (T0+63450) and (T0+398400)
Wait for a valid sync pulse length then
	Take a snapshot of the arrival time (T2) and the type (S2)
*Check* T2 is within 398832: 401136 of T0
If S0 is SYNC_TYPE_0 or SYNC_TYPE_2
	*Check* S2 is SYNC_TYPE_1 or SYNC_TYPE_3
If S0 is SYNC_TYPE_1 or SYNC_TYPE_3
	*Check* S2 is SYNC_TYPE_4 or SYNC_TYPE_6
If S0 is SYNC_TYPE_4 or SYNC_TYPE_6
	*Check* S2 is SYNC_TYPE_5 or SYNC_TYPE_7
If S0 is SYNC_TYPE_5 or SYNC_TYPE_7
	*Check* S2 is SYNC_TYPE_0 or SYNC_TYPE_2

// If it managed to get to here without a failed check then T2 and S2 become T0 and S0 of a newly started cycle
// The new cycle should adhere to the rules above (continue to do the checks?)
// S0 is output from the first lighthouse, S1 from the second lighthouse
// Angles are measured +/- around a centre of S0+192000 ticks (4000us)
// The sync type (corresponding to particular lengths) define the current axis (PA0/PA1) and whether the lighthouse is “skipping” that cycle (PS0/PS1)

// The lighthouse also outputs data encoded within the pulses (PD0/PD1)
// The data is meant to follow this format:
Pre-amble : 18bits   : “000000000000000001” – NO BIT STUFFING
Data Len  : 16bits   : uint16, little endian, number of bytes in the data field
Data      : VAR bits : message data with a padding byte appended ZERO byte if Data Len is odd
CRC       : 32bits   : little endian CRC of data field (excluding padding and bit stuffed bits)
Excluding the Preamble after every 16th bit within the message there is a bit-stuffed “1” bit.

LeeCookGHA, 2016-12-21:
It would be really interesting to know if the relative angles between the lighthouses are calculated and encoded within that sync pulse data stream!

LeeCookGHA, 2016-12-22:
After watching Alans Hackaday presentation, and especially after his comment in the other thread here, I'm now **really** curious about what is encoded in the sync pulses of the lh datastream.
Hoping for more, I figured it would just be serial numbers etc.
Have you seen anything like it being relayed by the HMD or do we need to pull the info from the sync data?

vk2zay, 2016-12-22:
@LeeCookGHA the bases do not send the angles of other bases they can see in their OOTX broadcasts.  I considered but never implemented this feature as it is not really required.  The angles might be useful for a couple of things, but they don't give you a scale for the Universe so would not help too much computing the relative position of the base stations - they might however be useful for locking down the orientation of the base stations which is more variable in the current implementation than I would like.

@nairol's website includes a description of the OOTX frame.  We currently implement its decode in the disambiguation layer, but previously we did it in the tracked object firmware...  It got moved host side to simplify things, it may end up back in the tracked object eventually.  The 5 calibration parameters per beam are probably worth a look once you guys get to tracking for real.  You will need to use the phase term at the very least and probably the tilt and gibbousness before you get good results.  Curvature is a bit of a hack on double-rotor bases and should be pretty low for HTC units so you may be able to just ignore it.

LeeCookGHA, 2016-12-22:
@vk2zay I need to stop thinking relative...  of course the likes of axis tilt are going to have to be known if you're positioning absolutely.
You say phase, is that the phase difference between the different axis wrt to the zero points? I'm struggling to think of another phase but I would have thought that could be removed in the lighthouse by shifting the lock position?

nairol, 2016-12-22:
@LeeCookGHA: [This](https://github.com/nairol/LighthouseRedox/blob/master/docs/Base%20Station.md#base-station-info-block) is the description of the OOTX payload that vk2zay mentioned. Just in case you haven't seen it yet.

Btw. I know my repo is not really up to date with all the stuff you guys figured out in the live streams and beyond. I'm on it. Currently trying to make a simple SVG explaining the "Arcane Bencoding". :)

vk2zay, 2016-12-22:
@LeeCookGHA phase of the beam relative to the sync.  Tilt is beam tilt, not axis tilt.

nairol, 2016-12-22:
Regarding writing a disambiguator for TDM base stations the following timing diagram might be helpful:  
[image]

The time between the rising edge of the master sync pulse and the rising edge of the slave sync pulse should always be 20000 ticks.

Lasers are turned on at 32.0° and stay on for 124.0°. (On at 71111 ticks, off at 346667 ticks, measured from the rising edge of the sync pulse)

The order in which sync pulses should be received is:
skip:0 axis:0
skip:1 axis:0
-- Laser sweep by master base station, axis 0 --
skip:0 axis:1
skip:1 axis:1
-- Laser sweep by master base station, axis 1 --
skip:1 axis:0
skip:0 axis:0
-- Laser sweep by slave base station, axis 0 --
skip:1 axis:1
skip:0 axis:1
-- Laser sweep by slave base station, axis 1 --

LeeCookGHA, 2016-12-22:
@nairol That is very cool - watching! I'm pretty sure that the source I used for the post earlier was probably copied from you but without the content details.
That's a hellova lot of config to figure out. Lets hope Charles' bases are sufficiently different that it is obvious.
Heh, I now see why Alan doesn't mind pitching in. This thing is hugely complex to keep the accuracy over the full volume. I feel like I did in school chemistry when the teachers said "actually, *this* is what an atom looks like and, oh, expect worse to come"....

LeeCookGHA, 2016-12-22:
@vk2zay Thank you.
@nairol I had some of that but other bits I got from the timing info in Charles files. Is there any reason you are aware of that the sync pulses from the same base would gradually drift in start time and reduce in length? That is why I only picked up the first declared reading from a unit, as the others were coming in they reduced in length, sometimes to the point where they were actually a valid length of the next sync down.

cnlohr, 2016-12-22:
Hmm... I thought my code gave timing from the center-of-the-pulse, not the beginning.  I guess I'm gonna need to nail that down.  Make sure it's universally beginning-of-pulse on both systems.  I don't know why that would be.  All of my tests are static.  Additionally, my disambiguator seems to work well with the HMD, just not the watchmen... But I really need to double-check my start- mid- times.

LeeCookGHA, 2016-12-23:
@cnlohr This is the definition of light_process_function:
( struct SurviveObject * so, int sensor_id, int acode, int timeinsweep, uint32_t timecode, uint32_t length );
but in the code:
ct->lightproc( so, le->sensor_id, -1, 0, le->timestamp, deltat );
the deltat is being output as the length?

Also, what's the purpose of dl in the offset_from assignment:
int32_t offset_from = le->timestamp - dl - so->last_photo_time + le->length/2;
if offset_from is timeinsweep?

vk2zay, 2016-12-23:
@nairol I have a similar timing diagram on my whiteboard at the moment:
[image]
The timing in 48 MHz ticks for each rotor of a base is roughly:
[image]

cnlohr, 2016-12-23:
Will read @vk2zay thing soon.  I wanted to chime in now saying: No reflections.  Just 90% sure something is wrong with my bencoding.  Things are jumping around far more than they should.

cnlohr, 2016-12-23:
Just wanted to let people know, there is at least one problem with the watchman data.  There looks to be another.  But, it was not reflections.

Here's some watchman data from the most recent run... ```	printf( "%s %d %d %d %d %d\n", so->codename, le->sensor_id, le->type, le->length, le->timestamp, le->timestamp-tsl );```
[watchman_lightcap_data.csv.zip](https://github.com/cnlohr/libsurvive/files/670483/watchman_lightcap_data.csv.zip)

jpicht, 2016-12-23:
Oh! I never knew the both stations were putting out sync pulses each cycle! So the "reflection" I though I saw was actually the second base station. That makes way more sense! The timing diagram really helps a lot to understand what is going on.

Who is currently implementing this in C? Is this still something that somebody needs to do? If nobody is on it, I would have another go at it.

cnlohr, 2016-12-23:
I have a rather bad implementation, but it seems to be working for everything except the watchmen.  But, I'm starting to think it may yet be something wrong with my bencoding.  I've already found one bug that totally wrecked the data.  There appears to be something else going on since I am having a hard time resolving the sync pulses.

jpicht, 2017-01-03:
My version was supposed to be better at deciding if a pulse is a sync pulse or not, when the HMD is close to the lighthouse. Which works pretty well. But when I compare it's performance to your current implementation I don't see the need for that, you really have to be pretty close to get real problems.

I was playing around a bit with your implementation and I found out that the timings for one of the lighthouses had considerably more jitter. I was able to fix that by tracking master and slave timings separately.

I did not send you a pull request, but my current version is up on my clone at: https://github.com/jpicht/libsurvive/tree/disambiguator

cnlohr, 2017-01-09:
Not sure - do you feel your current version is better or worse than mine?

jpicht, 2017-01-10:
I think it does not have to be done the way I did - it's more like a proof of concept - but referencing the sweep to the flash of the sweeping lighthouse should help (and it does for me) with the stability of the angle measurements.

It will introduce a static offset for the slave lighthouse, that needs to be taken into account when trying to find the lighthouses, but your current code always uses the sync flash of the slave lighthouse as a reference, so you probably take that into account already?

cnlohr, 2017-01-10:
Hmm, I don't know if I do or don't.  The data is very stable, though.  Additionally if there is a static offset, it doesn't really effect the system since the angles of the lighthouses can have constant offsets.  I will admit - if that is the case of what I am doing, it is nonideal.

mwturvey, 2017-01-11:
Just a quick note: I don't think a static offset is benign.  It becomes important when calculating the angle that the lighthouse makes relative to two points on a tracked object.  I assume that's also why the "phase" offset is listed in the [Base Station Info Block](https://github.com/nairol/LighthouseRedox/blob/master/docs/Base%20Station.md#base-station-info-block).  The error introduced from a static offset will be very small in the center of the lighthouse's range, but will get more extreme toward the "edges" of the lighthouse's range.

cnlohr, 2017-01-11:
I am pretty sure that because the lighthouses aren't cameras, it doesn't matter.  What I am worried about has more to do with jitter that could be introduced due to syncing off the wrong thing... and, I am pretty sure that would induce a fair bit of phase error.

mwturvey, 2017-01-11:
I agree that getting the jitter out and syncing off of the right thing is critical, so I don't want to distract this thread too much from that.  But if we write the disambiguator without regard to adding a static offset, I'm pretty sure we're going to have other problems later.  
Let me give an example of why I think it's so important.  In order to determine the position of the lighthouse, we will need to know the angle between two points and the lighthouse.  Assume a lighthouse at the origin facing +Y.  Now, given two points at (1, 10, 0) and (-1, 10, 0), there will be a vertical sweep delta of 0° (obviously) and a horizontal sweep delta of ~11.4°.  And, the angle the two points make with respect to the lighthouse is also ~11.4°.  Now, if we move both points along the Z axis by 100, they now are at (1, 10, 100) and (-1, 10, 100).  They still have a vertical sweep delta of 0°, and a horizontal sweep delta of 11.4°.  *But* the angle they make with respect to the lighthouse is now only ~ 1.14°.  Without knowing the absolute sweep angles, we cannot determine the angle between two points and the lighthouse.  ([Spreadsheet showing my work](https://docs.google.com/spreadsheets/d/1P-6fb-7_N7c-aqH2HFrgyKTUAlAKp33z6F3ajHPq3_M/edit?usp=sharing))

LeeCookGHA, 2017-01-11:
@mwturvey While I agree that the angles without a static offset makes the absolute location easier I don't think it is absolutely (sorry ;) )necessary.  The HMD will always be able to determine the relative angles between the individual sensors and the LH, which is enough to give you a bearing, attitude and distance to the LH (in the HMD frame of reference). It comes down to which reference frame you build it against, device or LH.
I would probs argue that the devices reference frame may be the easiest as you would then have a single reference frame for both LH while you are trying to work out the relative positions.

commit ee433e4a (2017-02-13): Closes #15 Closes #4 These are now rock solid.

Prepare the way for the OOTX Data.  Gonna hav eto get @axlecrusher on it.
closed 2017-02-13 by cnlohr