collabora/libsurvive #5: Agenda for Dec 17 Livestream

cnlohr, 2016-12-17:
* Get point and normals for watchmen
* Visualize lighthouse probability data in voxeltastic
* Improve watchman disambiguator
* Collect more data as people want it

nairol, 2016-12-18:
Regarding the calibration data of the controllers. You should be able to use the same method to download them as with the HMD.
Here is some example data from my Vive and an explanation how to do this using "lighthouse_console" on Windows: https://github.com/nairol/LighthouseRedox/tree/master/data/calibration%20data

cnlohr, 2016-12-18:
What is up with the mura correction stuff?? That looks wicked! == per-pixel brightness calibration.  People in chat are describing it as a correction for the AMOLED dispaly?

cnlohr, 2016-12-18:
@nairol  - how do I actually get the profile from the controllers? I've tried it wirelessly.  Does it not work wirelessly?

nairol, 2016-12-18:
Never tried it myself. It should be able to get the data wirelessly because that is what SteamVR does every time you start it.
I guess I'll have to sniff the USB traffic again. I'll report back once I know more...

nairol, 2016-12-18:
Pcaps: https://github.com/nairol/LighthouseRedox/tree/master/data/Vive%20USB%20protocol/System

cnlohr, 2016-12-18:
Now, I wish I could understand what's up with that control out message.  It's so jank.

cnlohr, 2016-12-20:
* Got points/normals for watchmen after livestream by getting config now.
* Visualized
* People are working on a better disambiguator
* Will be collecting more data!  

closing issue.

closed 2016-12-20 by cnlohr