collabora/libsurvive #7: Read watchman configuration

cnlohr, 2016-12-18:
We need a way of reading the configuration from the watchmen.  Right now I'm stuck at sending the commands needed to read from them via libusb.  It seems that there is a control_out message that's being sent that seems to trigger a read config from the controllers.

bjj, 2016-12-18:
You have to retry the failed USB commands.  They will STALL (which will be an error on the host) while the radio round-trip is complete.  Then they will succeed.

cnlohr, 2016-12-18:
Hmm... I can totally retry, since it's what I'm usually doing for other operations like reading the HMD config, but, what is kinda trippin' me out is this control message found in @nairol's dump.

[image]

I guess I'll just try spamming things more :-p

bjj, 2016-12-18:
It's just a GET FEATURE request to an endpoint whose descriptors call for
report IDs.

cnlohr, 2016-12-19:
It looks like in the packet dump the data from/to was switched around.  When I ignore that and just treat it like a normal control message everything's fine and I can pull the data.

commit 0d6e2a3d (2016-12-19): Closes #7 - retry the control out message
closed 2016-12-19 by cnlohr