collabora/libsurvive #8: Common data for project

cnlohr, 2016-12-19:
I figured I'd make a thread for files I will be using for various static analysis tools for this project.

Config files, points and normals for my vive: [charles_all_configs.zip](https://github.com/cnlohr/libsurvive/files/660286/charles_all_configs.zip)

wasdennnoch, 2017-02-04:
Screenie from the stream http://puu.sh/tODMa/72d2b565c9.jpg

mateon1, 2017-02-04:
HTC vive coordinate reference: [image]
Screenshot 2: [image]

cnlohr, 2017-03-03:
Fresh datas from the calibrator....  (With Josh's vive, need to find his cal data)

[ptinfo.csv.zip](https://github.com/cnlohr/libsurvive/files/815531/ptinfo.csv.zip)

Format is:
```fprintf( ptinfo, "%d %d %d %d %f %f %f %f %f %f\n", sen, lh, axis, count, avgsweep, avglen*1000000, stddevang*1000000000, stddevlen*1000000000, max_outlier_length*1000000000, max_outlier_angle*1000000000 );```

cnlohr, 2017-03-14:
Object poses taken for HMD, WM1 and WM0 based on Charles Poser...

Distances from LH1, LH2 to object
3.85, 2.96
3.75, 3.8
3.2, 3.55

[objposes.csv.zip](https://github.com/cnlohr/libsurvive/files/840354/objposes.csv.zip)

cnlohr, 2017-03-20:
Another test, this one is better than the above one.  It includes information about distances from lighthouses to objects.
[full_test_triangle_on_floor.zip](https://github.com/cnlohr/libsurvive/files/853966/full_test_triangle_on_floor.zip)

mwturvey, 2017-03-28:
[image]

mwturvey, 2017-03-28:
[image]

cnlohr, 2017-03-28:
Our definitions of +z is confusing to me now.  Is +Z down when holding the controller upright?

mwturvey, 2017-03-28:
Yeah, I was surprised by this.  That's from IMU data.  (I assume/ hope that the config data uses the same reference frame as the IMU).  When you hold the controller horizontal, with the word "Vive" facing up, positive Z is facing the floor.

mwturvey, 2017-03-28:
I just double checked the model of the tracker, and +Z from the model appears to be the same as what I got from the IMU.  Given that, I'm confident in the tracker's designation above.  And given that I used the same methodology for the controller, I'm pretty sure it's correct, too.  Once my HMD shows up in the mail, I'll verify against it using the same method, just to be sure.

mwturvey, 2017-05-16:
Here's another sample, this time with just a single watchman that was connected via USB.  

One lighthouse is about 6 feet off the floor, the other is about 5 feet off the floor (~ 1.8m and 1.5 m).  The controller is roughly 2.5m away from each lighthouse.  Both lighthouses are placed on opposite sides of the same wall, about 3m apart.  This sample is using my disambiguator.

[calinfo.zip](https://github.com/cnlohr/libsurvive/files/1003286/calinfo.zip)

[output.zip](https://github.com/cnlohr/libsurvive/files/1003295/output.zip)

mwturvey, 2018-03-11:
[image]

cnlohr, 2018-03-15:
here's some more data...  from HMD only.  (This is just light data, strictly.  For Charles and his new disambiguator experiment)

[lightdata.txt.zip](https://github.com/cnlohr/libsurvive/files/1813965/lightdata.txt.zip)

cnlohr, 2018-03-23:
Here's a graph I made of distance from lighthouse and an incident angle.
[Relationship between Distance And Pulse Length.ods.zip](https://github.com/cnlohr/libsurvive/files/1840195/Relationship.between.Distance.And.Pulse.Length.ods.zip)
 some

cnlohr, 2018-03-24:
Here is a set of data logs from the orange pi, no calibration, no posers.

[orange_pi_data_logs.zip](https://github.com/cnlohr/libsurvive/files/1844104/orange_pi_data_logs.zip)

cnlohr, 2018-03-24:
Here's a pure lightcap...
[time] [sensor] [length] [tick]
[lightcap-orangepi-slow-fast-on-floor-three.txt.zip](https://github.com/cnlohr/libsurvive/files/1844152/lightcap-orangepi-slow-fast-on-floor-three.txt.zip)

jdavidberger, 2018-03-25:
[lightcap-reformat.zip](https://github.com/cnlohr/libsurvive/files/1845699/lightcap-reformat.zip)

Same thing as lightcap-orangepi-slow-fast-on-floor-three.txt.zip but in a format that can be used as a playback file

jdavidberger, 2018-03-26:
[image]
[image]
[cleaner_data.zip](https://github.com/cnlohr/libsurvive/files/1846217/cleaner_data.zip)

Cleaner / debugged data with pictures. Note that things get very squirrely when you get too close to the sensor, but otherwise it follows what you'd expect

cnlohr, 2018-04-01:
[data_for_albino.zip](https://github.com/cnlohr/libsurvive/files/1866126/data_for_albino.zip)

This is a data record sitting on the floor, pointed upward, then picking the vive up and moving around very slowly, going to the extremes of the lighthouses.

jdavidberger, 2018-04-02:
Visualization for calibration values. 

I'm mostly adding these so I can link them in the wiki

jdavidberger, 2018-04-02:
[logs_for_calibration.zip](https://github.com/cnlohr/libsurvive/files/1866913/logs_for_calibration.zip)

[calibration_pass.tar.xz.zip](https://github.com/cnlohr/libsurvive/files/1866924/calibration_pass.tar.xz.zip)
Rename that last one to omit the zip -- github doesn't like tar.xz?

closed 2020-08-17 by jdavidberger