collabora/libsurvive #11: Bencoding.  Apparently, I was doing it wrong.

cnlohr, 2016-12-23:
Sorry to bring it back - there's definitely something wrong with the bencoding.

consider the following output:

Why is the time jumping around so heavily? 

LED: 10 (adv 0) / st: 305346930 en: 305347134 len: 204 / TC: 1
LED: 4 (adv 2) / st: 305346536 en: 305346806 len: 270 / TC: 5
LED: 2 (adv 0) / st: 305346589 en: 305346786 len: 197 / TC: 4
LED: 9 (adv 0) / st: 305346258 en: 305346521 len: 263 / TC: 7
LED: 5 (adv 0) / st: 305346031 en: 305346209 len: 178 / TC: 9
LED: 8 (adv 0) / st: 305345537 en: 305345832 len: 295 / TC: 11
---
LED: 21 (adv 0) / st: 298517640 en: 298520448 len: 2808 / TC: 1
---
LED: 6 (adv 0) / st: 305555807 en: 305559276 len: 3469 / TC: 1
---
LED: 1 (adv 0) / st: 298724883 en: 298725324 len: 441 / TC: 1
LED: 2 (adv 0) / st: 298724288 en: 298724723 len: 435 / TC: 3
LED: 0 (adv 0) / st: 298723812 en: 298724254 len: 442 / TC: 5
LED: 12 (adv 0) / st: 298536725 en: 298543135 len: 6410 / TC: 7
LED: 1 (adv 0) / st: 298517026 en: 298520501 len: 3475 / TC: 9
---
LED: 4 (adv 1) / st: 305825810 en: 305826127 len: 317 / TC: 2
LED: 3 (adv 1) / st: 305825560 en: 305825969 len: 409 / TC: 3
LED: 2 (adv 0) / st: 305825198 en: 305825386 len: 188 / TC: 5
LED: 1 (adv 2) / st: 305575490 en: 305581915 len: 6425 / TC: 9
LED: 7 (adv 0) / st: 305575932 en: 305581878 len: 5946 / TC: 8

Names for possible future mention:  but danweecc, Zmathue, CaiB, mobius, MobiusHorizons  (but not yet)

cnlohr, 2016-12-23:
Something still smells funky....

```	printf( "%s %d %d %d %d %d\n", so->codename, le->sensor_id, le->type, le->length, le->timestamp, le->timestamp-tsl );```
WM0 1 254 5456 20507665 206574
WM0 10 254 3467 20527390 19725
WM0 13 254 1781 20528242 852
WM0 17 254 2344 20528453 211
WM0 9 254 428 20691880 163427
WM0 8 254 423 20692095 215
WM0 4 254 408 20692426 331
WM0 10 254 458 20692706 280
WM0 22 254 296 20692721 15
WM0 0 254 3967 20907685 214964
WM0 10 254 4963 20927412 19727
WM0 3 254 4339 20927981 569
WM0 13 254 2457 20928322 341
WM0 17 254 3041 20928419 97
WM0 2 254 304 21082930 154511
WM0 1 254 326 21083791 861
WM0 0 254 345 21084123 332
WM0 18 254 236 21084403 280
WM0 5 254 304 21085288 885
WM0 0 254 3471 21307663 222375
WM0 13 254 2104 21328317 20654
WM0 17 254 3292 21328328 11
WM0 10 254 5469 21327369 -959
WM0 3 254 4826 21327959 590
WM0 0 254 385 21531114 203155
WM0 2 254 346 21531511 397
WM0 1 254 371 21531875 364
WM0 18 254 301 21532048 173
WM0 5 254 365 21532208 160
WM0 1 254 4948 21707686 175478
WM0 10 254 3967 21727396 19710
WM0 13 254 2426 21728172 776
WM0 17 254 2171 21728250 78
WM0 17 254 359 21730918 2668
WM0 8 254 327 21898558 167640
WM0 22 254 258 21899018 460
WM0 9 254 351 21899600 582
WM0 10 254 376 21899890 290
WM0 4 254 324 21901070 1180
WM0 1 254 6459 22107662 206592
WM0 13 254 2352 22128280 20618
WM0 10 254 3464 22127359 -921
WM0 17 254 2457 22128302 943
WM0 9 254 429 22291854 163552
WM0 8 254 424 22292072 218
WM0 4 254 413 22292400 328
WM0 10 254 456 22292682 282
WM0 22 254 308 22292698 16
WM0 1 254 3968 22507664 214966
WM0 17 254 2696 22528273 20609
WM0 13 254 2640 22528746 473
WM0 10 254 4970 22527372 -1374
WM0 3 254 4328 22527963 591
WM0 2 254 304 22682913 154950
WM0 1 254 323 22683776 863
WM0 0 254 343 22684109 333
WM0 18 254 229 22684392 283
WM0 5 254 303 22685272 880
WM0 1 254 4459 22907662 222390
WM0 10 254 5463 22927347 19685
WM0 17 254 3377 22928224 877
WM0 13 254 1856 22928464 240
WM0 1 254 369 23131870 203406
WM0 18 254 300 23132043 173
WM0 1 254 5961 23307657 175614
WM0 10 254 3967 23327375 19718
WM0 17 254 3028 23328249 874
WM0 13 254 2198 23328334 85
WM0 9 254 352 23499578 171244
WM0 1 254 366 35931833 12432255
WM0 18 254 303 35932002 169
WM0 5 254 367 35932161 159
WM0 1 254 4947 36107632 175471
WM0 2 254 35 36130200 22568
WM0 1 254 40 36130200 0
WM0 0 254 43 36130200 0
WM0 6 254 49 36130200 0
WM0 20 254 51 36130200 0
WM0 10 254 2970 36127299 -2901
WM0 22 254 256 36298930 171631
WM0 10 254 377 36299797 867
WM0 4 254 325 36300978 1181
WM0 1 254 5460 36507611 206633
WM0 13 254 2415 36528235 20624
WM0 17 254 2106 36528552 317
WM0 10 254 3473 36527253 -1299

LeeCookGHA, 2016-12-23:
I can't watch the video atm but looking through the code and the comments I think you should reconsider the part you've flagged as wacky:
160 		type &= 0x7f; 
161 		if( type == 0x68 ) doimu = 1; 
162 		type &= 0x0f; 
163 		if( type == 0x00 && qty ) { type = POP1; qty--; } 

The 0x60 part of the doimu check will always be present (0xF0 to enter the type statement and masked with ~0x90 over the course of the block).  This essentially leaves 0x08 to define if IMU data is present *but*, you don't process it right away...

So, if IMU data is present you set a flag, mask out the 0x60 and leave the 0x01 in the type at the end of the block.

This means if (0xF8 == type&0xF8) at the start of handle_watchman you're essentially saying that there is no-way the next type could be 0xe8 or 0xe1?  Whereas, if there was no IMU data there could be?  Is that definitely the case?

It would sit much better with me if the IMU data were done in the block rather than by a flag and then do the standard if( qty ) check to pop the next type...

cnlohr, 2016-12-23:
It seems odd I am not getting any mismatching parameter count warnings thrown.  If I was misinterpreting that there would be an extra photodiode flag and would start yelling at me.  

I will admit, I have not taken too close of a look at it and it did feel rather funky at the time when I coded it.

LeeCookGHA, 2016-12-26:
Sorry, hopefully the right issue this time.  I Can't see anything in #8 which has something like "raw_light_data_from_watchman.csv", or even better, a hex of the raw data and the output of the most recent Bencoding decoding interleaved.

cnlohr, 2016-12-26:
@LeeCookGHA Your wish is my command.
@MobiusHorizons I think you would do a good job

Format: 

_WM0 (or 1)   [RAW HEX
_WM0 Time: Popped time from end, newtime = new local time as we count back in time.
_WM0 Event: Actual event.
WM0: Event as it is received by the rest of the system.

[watchman_extra_debug_data.txt.zip](https://github.com/cnlohr/libsurvive/files/672604/watchman_extra_debug_data.txt.zip)

cnlohr, 2016-12-26:
This one also contains HMD data to compare against.  And yes, it really does look like something is gong wonky with the sync flashes.
[watchman_extra_debug_data_with_hmd.txt.zip](https://github.com/cnlohr/libsurvive/files/672640/watchman_extra_debug_data_with_hmd.txt.zip)

cnlohr, 2016-12-26:
Also, @Zmathue and @CaiB any way you can compare the raw data in there with what your decoders say?

CaiB, 2016-12-26:
@cnlohr I never did write a decoder for the data, I simply plotted what I discovered about the packet in Excel and correlated that to known input (AVR) to try and figure out how the encoding behaved, so I'm not sure what I can do here...

cnlohr, 2016-12-26:
>.< ok thanks.  Sorry about that.  Got you guys confused.

CaiB, 2016-12-26:
Hey, no problem.

I tried to have a look through your decoder code, but as usual C just confuses me, I'm more of a C# person. Sorry I can't be of more help :(

cnlohr, 2016-12-26:
It's a-okay.  I am sure there will be other parts of the project that you can work stuff out for.  Heck, who knows, may even support a C# port someday :)

cnlohr, 2016-12-26:
Hmm... So I found something that looks like what might have been causing my issues.

I get this from the watchman.
```
WM0 3 254 3468 228144868 232328
WM0 12 254 2831 228145456 588
WM0 12 254 6457 228164388 18932
```

This corresponds to the following event:
```
HMD 5 0 3381 398740715 223734
HMD 1 0 3411 398740690 -25
HMD 2 0 3415 398740686 -4
...
HMD 29 0 6355 398760248 19614
HMD 12 0 6357 398760247 -1
HMD 8 0 6352 398760255 8
```

I don't understand why the watchman contains a second, invalid pulse, but, it seems that it can be detected because it overlaps, in time, almost entirely with the first pulse.  

This happens later, too.

```
WM0 3 254 2947 229344948 180554
WM0 12 254 2332 229345511 563
WM0 19 254 2336 229345512 1
WM0 17 254 2198 229345641 129
WM0 20 254 5514 229364784 19143
...
WM0 12 254 5972 229364370 -414
WM0 6 254 284 229509720 145350
```

For HMD event

```
HMD 5 0 2863 399940792 179776
HMD 1 0 2887 399940773 -19
HMD 2 0 2892 399940768 -5
...
HMD 29 0 5876 399960232 19514
HMD 12 0 5879 399960231 -1
HMD 8 0 5875 399960237 6
```

I will try to detect these funkypulses and throw them out.

nairol, 2016-12-26:
Found a bug in my decoder thanks to your data files. (Still have to fix it)

Btw. the time values (last 3 bytes) look strange in your decoded lines.
They shouldn't be bigger than 0x00FFFFFF (16777215). Where do you get the MSB from? I just set it to 0 in my tests. If you get them from the packet header (that's not included in the raw data that you posted) then I can't compare them.

cnlohr, 2016-12-26:
Oh, sorry.  I was trying to match the times up with the IMU times initially.  You can ignore the MSBs for your purposes.

nairol, 2016-12-28:
Ok, I can confirm that my de-bencoder outputs the same light pulse data as yours given the same raw input data. (Tested with watchman_extra_debug_data.txt.zip that you posted in this thread)

cnlohr, 2016-12-28:
Excelent.  That raises a lot of my confidence levels and puts the trouble and rough edges squarely in issue #4 .

closed 2016-12-28 by cnlohr