collabora/libsurvive #12: Sometimes data just stop flowing from the HMD

cnlohr, 2016-12-24:
don't know why, but sometimes, the HMD randomly decides not to give light data. Even when sending the light on magic.

geekmaster, 2016-12-24:
That is the problem with magic. Perhaps it needs to be repeated until you know it is on (open loop control). Or perhaps there is a status that tells it when to send the command. Or some other unknown.

cnlohr, 2016-12-24:
I can click the "o" button to resend magic.  When it's not working, I can spam it all day long to no avail.  But, if I unplug, replug, wait, then click "o" seems to usually work.  I think I'll leave this issue here in case someone does want to give it a crack.

geekmaster, 2016-12-24:
Can you send it a reset command or a power off command instead of power cycling it? I will have time to test this in a few days...

cnlohr, 2016-12-24:
I have tried sending the "off" magic commands and then "on" to no avail.  I do not need to power cycle, just unplug USB and replug it.

LeeCookGHA, 2016-12-25:
Are you able to produce a file which interleaves a hex dump of the raw data lines with the output of the decodes?  The longer the better, but over 12 seconds if possible to try and get a couple of full OOTX in.
With 4 kids, Christmas anf family around the country, it's going to be busy, but I'm sure I'll be able to grab some time to look at it over the holidays if no-one else is.

cnlohr, 2016-12-25:
I think this is the wrong issue for that, @LeeCookGHA - but, you can get that from the data in #8 for the data we're trying to get from the disambiguator.

jpicht, 2016-12-30:
I do not get any useful data from the HMD:

HMD 254 254 0 -1980420239 -1980747918
HMD 255 255 65535 -2104229889 -123809650
HMD 255 255 65535 327679 2104557568
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0
HMD 254 254 0 -1979884091 -1980211770
HMD 255 255 65535 -2104229889 -124345798
HMD 255 255 65535 327679 2104557568
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0
HMD 254 254 0 -1979347936 -1979675615
HMD 255 255 65535 -2104229889 -124881953
HMD 255 255 65535 327679 2104557568
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0
HMD 254 254 0 -1978811782 -1979139461
HMD 255 255 65535 -2104229889 -125418107
HMD 255 255 65535 327679 2104557568
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0
HMD 254 254 0 -1978275634 -1978603313
HMD 255 255 65535 -2104229889 -125954255
HMD 255 255 65535 327679 2104557568
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0
HMD 254 254 0 -1977739479 -1978067158
HMD 255 255 65535 -2104229889 -126490410
HMD 255 255 65535 327679 2104557568
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0
HMD 254 254 0 -1977203325 -1977531004
HMD 255 255 65535 -2104229889 -127026564
HMD 255 255 65535 327679 2104557568
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0
HMD 254 254 0 -1976667177 -1976994856
HMD 255 255 65535 -2104229889 -127562712
HMD 255 255 65535 327679 2104557568
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0
HMD 254 254 0 -1976131022 -1976458701
HMD 255 255 65535 -2104229889 -128098867
HMD 255 255 65535 327679 2104557568
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0
HMD 255 255 65535 327679 0

Is the current master broken, or is something wrong with my vive?

jpicht, 2017-01-03:
How often does this problem happen for you? I get it like on every second attempt to read data. Sometimes pressing O helps, sometimes it doesn't.

cnlohr, 2017-01-09:
Maybe one out of every 10 times, then it doesn't work for a while and randomly when I reconnect and press 'O' it just works immediately.

jpicht, 2017-01-10:
I just spend like 15 Minutes unplugging and messing around with the order of connecting the cables until I got it actually working ...

Do you have any idea where I could start to look to find out what's going on there?

cnlohr, 2017-01-10:
No.  That is why it is infuriating to me as well.  It might take taking some usbpcap wireshark dumps from Windows to more understand what the HMD reset looks like.

Can you characterize activities that make it crash regularly?

jpicht, 2017-01-10:
From what I observed:
- it (virtually) never works if it is not currently receiving a video signal.
- once it is running it never stops while a program is receiving the data
- if I stop the application, the longer time I wait before I start an application the more likely it is to not respond
- it sometimes seems I can take longer time between program stop and start, when I issue an "F" key before killing the program, but this could very well be just by chance
- the HMD has multiple modes, that show at least in different video output capabilities. When it is in the mode where only the right eye get's an image (recognized as a 720x480 display) it seems more reluctant to start. The only reliable way to get out of that mode and to get it into the "3d" mode (both eyes get an image) in linux is to boot into windows, run steam vr, and without full power loss boot into linux. It seems to retain that mode through a reboot most of the time ... but sometimes it goes into the one eye mode. I could not yet find out why and when.

commit eb9893e9 (2017-01-22): Closes #14 and closes #12 - fix syncing off correct word and don't throw out good data from HMD.
closed 2017-01-22 by cnlohr