collabora/libsurvive #15: Disambiguation with watchmen

cnlohr, 2017-01-24:
Need to figure this disambiguation with the watchmen thing out.  Syncing is frequently wrong.
[image]

jpicht, 2017-01-25:
What does that graph actually show?

cnlohr, 2017-01-25:
It shows that when a base station is scanning, most of the time, the value of the scanned time is the same, but, every now and then it gets a read time much earlier.  In fact this happens almost 30% of the time.  I believe the second peak is the correct one.  But, for some reason the disambiguator is sometimes saying "no it happened at 380000 ticks, not 380400 ticks!" which is wrong.  Forgot to mention: X axis is ticks/4.

commit ee433e4a (2017-02-13): Closes #15 Closes #4 These are now rock solid.

Prepare the way for the OOTX Data.  Gonna hav eto get @axlecrusher on it.
closed 2017-02-13 by cnlohr