collabora/libsurvive #19: Decode OOTX stream

cnlohr, 2017-02-07:
Need to use @nairol 's documentation and work with the OOTX streams. https://github.com/nairol/LighthouseRedox/blob/master/docs/Light%20Emissions.md

axlecrusher, 2017-02-10:
I am working on this.

cnlohr, 2017-02-13:
[josh.txt.zip](https://github.com/cnlohr/libsurvive/files/770289/josh.txt.zip)

@axlecrusher -> This is an output from the current libsurvive.  Can you see if your code can parse out the lines with -1 and -2 for the sensor id?  They are the primary and secondary lighthouses's sync codes respectively.  I also do the decoding from time to i,j,k within the code, too.  If you can parse this, it will be very easy to put your code in!

cnlohr, 2017-02-13:
Another thing we may want to consider is making the ootx stuff only happen when in "calibrate" mode or something to keep the core of libsruvive as absolutely lean and mean as possible.  Just something to think about.  It would mean your code would stay at least out of the core libsurvive.  Maybe it could be used with a room calibration executable.

axlecrusher, 2017-02-13:
Yes I was thinking there is no reason for ootx decoder to run all the time

cnlohr, 2017-02-13:
Not just run, but, code not exist.  I am trying to make sure there is a mechanism to actually run the whole algorithm on a really limited chip.  It just won't have to deal with room setup.  That can all be provided by a config file.  Only thing it might be useful for is a sanity check like "yo, lighthouse down wrong.  fix it."

axlecrusher, 2017-02-13:
your data output seems to be working fine for me now.

LeeCookGHA, 2017-02-14:
Are you sure about not running the ootx all the time?  Some of the parameters will be def be static (based on the optics or physical setup such as tilt and linearity etc) but it could, with certain implementations, be easier to have some of the parameters as dynamic (phase for example).

cnlohr, 2017-02-14:
@nairol does the OOTX cal data change dynamically or is it static?  Would be nice to cut it out of the picture for most of the time if possible.

I agree that if we find out it changes, we'll have to re-evaluate.

cnlohr, 2017-02-14:
@LeeCookGHA , @mwturvey  just pointed out your desire for custom hardware, so, once I get it going, I would be amenable to having a constantly running OOTX engine.

nairol, 2017-02-14:
@cnlohr Most of it is static. The only things that can change (but usually don't) are:

**fw_version** changes after firmware updates. Hopefully. :)

**sys.unlock_count** increases each time the rotors desynchronize from the internal clock. E.g. quick movement or rotation of an active BS causes this.

**accel.dir** changes when the gravity vector changes. Can happen by bumping the BS (like the unlock counter) and also if the BS was rotated while powered off.

**mode.current** changes if the user presses the button on the back.

**sys.faults** changes if the BS detects a hardware fault. It is a bit field: 4=Laser0, 8=Laser1, 16=Motor0, 32=Motor1 (Not sure about the other bits.)

Everything else is constant.

closed 2017-04-19 by cnlohr