collabora/libsurvive #24: Calibration Architecture

cnlohr, 2017-02-13:
Need to set up architecture so we can, in a single, consistent application:

(1) collect data
(2) store data to file
(3) run data processor
(4) lighthouse find
(5) zero floor and test colinearity with watchmen
(6) write out a buffer of "world" values which contains:
  * Lighthouses X, Y, Z + Quaternion
  * RMS Quality of camfind solution (in arbitrary units)
  * OOTX Decoded information from lighthouses for fixing beam angles.

Said buffer will be read into a datastructure for the rest of the system to use.

cnlohr, 2017-02-14:
For floor calibration, consider making a triangle with the headset and two watchmen.

mwturvey, 2017-02-14:
For floor calibration: 
If you set the headset on the floor looking up, how consistent will the height of the HMD's "origin" across a wide variety of HMDs?  I.e. is there rubber or anything that would make this unreliable?

For the watchmen, if you set one on the floor, is the orientation always consistent?  What about on thick carpet?

cnlohr, 2017-02-14:
I think the watchmen: yes. we're good.  headset: don't know. Maybe we will need to take accelerometers into account.  Then we could use as little as one watchmen's height.

cnlohr, 2017-04-29:
Looks like we're going to use the IMU in the system to determine the "down" vector.

mwturvey, 2017-05-21:
Just committed an update for the tori poser.  When calibrating, it now determines "down" from the IMU, not from the orientation of the tracked object.

cnlohr, 2017-05-22:
Did you do the thing where it also determines the "world" output for the cameras?  Also -- just to document it somewhere, we will need a "rebase" function, to allow for the re-zeroing of origin.  Though that can certainly wait.

closed 2020-03-04 by jdavidberger