collabora/libsurvive #35: Identify minimal hardware that will be able to output to Vive

cnlohr, 2017-03-11:
It looks like the RPi3 won't be able to output to the vive's HDMI at needed resolution.

Investigate OrangePi.

VKoskiv, 2017-03-11:
RPi 3 pixel clock reference: http://elinux.org/RPiconfig

ThatsEli, 2017-03-11:
odroid﻿ c2:  http://odroid.com/dokuwiki/doku.php?id=en:c2_hdmi_autosetting

ChristophHaag, 2017-03-11:
As one data point, my Chromebook with Tegra K1 can output to the OSVR HDK2 display with 2160x1200 at 90Hz.

Fun fact: My Ivy Bridge i7 3632QM can not.

emanuele6, 2017-03-11:
Asus Tinker Board: https://www.asus.com/uk/Single-board-Computer/TINKER-BOARD/

ThatsEli, 2017-03-11:
@emanuele6 The Tinker board can do 253.17 MHz    3880 x 2175 @ 30.000000000 Hz    (exact)

ThatsEli, 2017-03-11:
Tinker board : 253 MHz
odroid﻿ c2 : 204 MHz
Cubieboard : 142 MHz
BeagleBoard : 125 MHz
Intel Galileo & Intel Edison : 20 MHz

The Tinker boardseems to be the best(with 250MHz) :/

StareInTheAir, 2017-03-11:
@ThatsEli Are you sure about those numbers? The odroid c2 supports HDMI 2.0 and therefore 4k@60 Hz (see http://www.hardkernel.com/main/products/prdt_info.php?g_code=G145457216438). The max clock rate of HDMI 2.0 is 600 MHz, so that should be more than enough. But I'm also no expert at HDMI.

ThatsEli, 2017-03-12:
@StareInTheAir Mh, you are right, I calculated some of the numbers and found some of them(including odroid﻿ ), I calculated 530 MHz, that would be great! Thank you for notifying!

geekmaster, 2017-03-12:
The HD4000 video output can only drive 2560 wide when using DisplayPort output. Two of my laptops have DP, so now I need to find a Vive DP cable to try it.

I am also working on a windows monitor.inf file for Vive EDID override. Only partly working now, probably because HD4000 over HDMI... For some applications, lower resolution and/or lower framerate is okay -- lots of folks liked their Rift DK1 other than screendoor effect, and the high res displays in Vive and Rift CV1 cure that (even when driving at lower resolutions).

cnlohr, 2017-03-12:
Looks like you can't easily get the Odroid C2.  Any chance anyone has one already?  Or, would be able to buy and test libsurvive on it?

geekmaster, 2017-03-12:
They are available on ebay for twice the amazon "out of stock" price.

http://www.ebay.com/itm/ODROID-C2-Project-Board-Quad-Core-2GHz-2GB-RAM-HDMI-2-0-IR-Gigabit-New-/351854231558

I began to order one, but they are adding on SALES TAX. ??? Never paid sales tax before except from an ebay store in my home state.

cnlohr, 2017-03-13:
You mean libsurvive compatible... The vive still "requires a VR-capable computer" :-p

geekmaster, 2017-03-13:
It seems that the odroid-c2 only runs at 1.5GHz or less despite claims it is 2GHz. The board manufacturer has requested info about this from the chipset manufacturer, and plans to change marketing literature to 1.5Ghz. I do not know how this will affect libsurvive, though it may limit scene complexity.

http://forum.odroid.com/viewtopic.php?f=141&t=23044

cnlohr, 2017-03-13:
"scene complexity"  I think all we'd be able to get on one of those things is line-based stuff anyway. :-p

geekmaster, 2017-03-13:
Even my original raspi could do stereoscopic skyboxes on my rift dk1 (though I did not publish it). In my case, I used 3d 360 photograph pairs, converted to a pair of skyboxes. I used a vertex shader to bounce the skybox off a sphere (as Paul Bourke suggested) and it worked great on my Pi. Also, the Pi software API can use the same framebuffer for the video decoder and the skybox renderer, so we can also do skyboxed video, even on an old Pi. Certainly on an odroid-c2. And BTW, doom ran great on the Pi as well. Many thinks can be rendered small and then scaled up to 2560x1200, which eliminates screendoor. The main problem is you cannot read onscreen small text (or rine details) when upscaling resolution. Vector graphics can be compelling for some applications though, when you really want high resultion. But you can mix vector foregrounds with video skybox backgrounds for some interesting content without needed the latest nvidia GPU.

cnlohr, 2017-03-13:
I really think even simple lines could provide an awesome bunch of games.   I actually don't know anything about getting GPU accelerated stuff running on the raspi 3 -- let alone the odroid.  Have any resources?

geekmaster, 2017-03-13:
Interestingly, the game that won the first Oculus competition for the DK1 was a vector-based game.

Regarding raspi gpu resources, my code is on one of my 300+ unsorted hard drives that are in storage. I need to index all of them (some day)...

I played a bit with the raspi native GPU assembler (not compiler) code a few years ago -- it should be MUCH more advanced by now. And with my new faster pi 3, I really need to see what it can do in VR land (at least with a DK1, if not a vive). I do know that the pi 3 can do 4K output (but only at 15Hz). Inadequate pixel clock...

I was also using my razer hydras on my raspi, besides DK1 tracking and video.

BTW, I am "geekmaster" in the oculus and mtbs3d (and other tech) forums. I have not been active in the forums for a few years (health reasons), but I am getting back into it. I have a Rift CV1 and HTC Vive, and Gear VR, and my Virtuix Omni should arrive next week. I have a big reason to complete my various projects in my remaining time (I have already outlived my prognosis by many months, with hopefully enough time left to make some useful contributions to VR and robotics). I am very well stocked in hardware and components -- I wish I had some assistants to help me with physical lifting and such. Getting old sucks -- avoid it if you can. ;-)

cnlohr, 2017-03-13:
Glad to have you, @geekmaster working on this all.

geekmaster, 2017-03-14:
I am just glad to not be completely on my own like the olden days. Thinks like github, forums, social networks, etc. all make it a lot easier to have remote "assistants". I really like your live coding sessions (when I have time to sit through them). I should probably do something like that with my projects. Sadly, my LTE wireless ISP TOS says "no live streaming". It has half minute dropouts during primetime when I go over 23GB/month too (no bandwidth limits or caps, just "deprioritized" so best time is late nights). Sadly, my steam vr apps for the vive go over my 23GB limits every month (a 13GB update has been in progress a couple of days now, plus dozens of many hundred MB updates -- why do steam apps do such damn HUGE updates, way too often?). I forked tons of stuff to my github -- now I just need to upload some of my many personal projects. My worry is maintaining it, with potential pull requests and such (and learning github to a useful degree). @cnlohr thanks for getting me interested in libsurvive (via hackaday). Regarding hackaday: https://hackaday.com/2006/09/12/floppy-connector-mmc-card-slot/  https://hackaday.com/2016/07/09/a-jailbreak-for-every-kindle/

VKoskiv, 2017-03-14:
LibSurvive - Redefining "VR-capable computer"

geekmaster, 2017-03-14:
FYI, my experiments show that you can get by with a lower framerate in VR if you have your eyes dark-adjusted and the VR content is low contrast with relatively smooth edges. At least in my case. So it might be worth seeing what kind of VR apps could be RasPi compatible.

I will let you know when my odroid-c2 arrives. Meanwhile, I will try to get the vive video working on a laptop with only HD4000 graphics (using display port output). My Virtuix Omni arrived at my local airport, so I am scrambling to get a "wife compatible" 5x7 ft area setup for it. I will use it with my Rift CV1, so their is room to free walk in the HTC Vive in a different room. It sucks because we have too darn much furniture and rather narrow rooms (ranch style house). My living room is already full of exercise equipment squeezed in (for my required physical therapy). I hope the Omni can replace some of the (boring) exercise equipment. ;-)

StareInTheAir, 2017-03-14:
I got a C2, but unfortunately no Vive. But I still did some testing and found out a few things. I'm just gonna write it down how it happened.

I first tried [Arch Linux](https://archlinuxarm.org/platforms/armv8/amlogic/odroid-c2) with Gnome 3, didn't work, gdm crashed while starting. So I moved to [Ubuntu MATE](http://odroid.com/dokuwiki/doku.php?id=en:c2_release_linux_ubuntu) and had more success. I have the C2 connected to a 4k monitor (LG 27MU67-B). The desktop started in 4k 30hz. I couldn't set a different resolution with the builtin "Display" settings GUI. The only thing I discovered changed the resolution was `echo 2160p60hz420 > /sys/class/display/mode`. Then I had 4k@60Hz output which is only possible with HDMI 2.0. `echo 2160p60hz > /sys/class/display/mode` (4:4:4 chroma) doesn't work for me although the monitor supports it (tested with a GTX 1070 under Windows). This probably is a monitor issue, as [this thread](http://forum.odroid.com/viewtopic.php?f=144&t=21952#p149773) suggests. `/sys/class/display/mode` only changes the HDMI output resolution, the MATE desktop remains in it's original size.

I then played around with HDMI resolutions. Unfortunately it seems the C2 doesn't support custom resolutions like the one used on the Vive. If you write an "invalid" resolution like `echo 2160x1200p30hz > /sys/class/display/mode` nothing happens. The only supported resolutions seem to be:

```bash
grep -i hz /media/boot/boot.ini
# setenv m "480i60hz" # Interlaced 60Hz
# setenv m "480i_rpt" # Interlaced for Rear Projection Televisions 60Hz
# setenv m "480p60hz" # 480 Progressive 60Hz
# setenv m "480p_rpt" # 480 Progressive for Rear Projection Televisions 60Hz
# setenv m "576i50hz" # Interlaced 50Hz
# setenv m "576i_rpt" # Interlaced for Rear Projection Televisions 50Hz
# setenv m "576p50hz" # Progressive 50Hz
# setenv m "576p_rpt" # Progressive for Rear Projection Televisions 50Hz
# setenv m "720p50hz" # 50Hz
# setenv m "720p60hz" # 60Hz
# setenv m "1080i60hz" # Interlaced 60Hz
setenv m "1080p60hz" # Progressive 60Hz
# setenv m "1080i50hz" # Interlaced 50Hz
# setenv m "1080p50hz" # Progressive 50Hz
# setenv m "1080p24hz" # Progressive 24Hz
# setenv m "2160p30hz"    # Progressive 30Hz
# setenv m "2160p25hz"    # Progressive 25Hz
# setenv m "2160p24hz"    # Progressive 24Hz
# setenv m "smpte24hz"    # Progressive 24Hz SMPTE
# setenv m "2160p50hz"    # Progressive 50Hz
# setenv m "2160p60hz"    # Progressive 60Hz
# setenv m "2160p50hz420" # Progressive 50Hz with YCbCr 4:2:0 (Requires TV/Monitor that supports it)
# setenv m "2160p60hz420" # Progressive 60Hz with YCbCr 4:2:0 (Requires TV/Monitor that supports it)
# setenv m "640x480p60hz"
# setenv m "800x480p60hz"
# setenv m "480x800p60hz"
# setenv m "800x600p60hz"
# setenv m "1024x600p60hz"
# setenv m "1024x768p60hz"  
# setenv m "1280x800p60hz"
# setenv m "1280x1024p60hz"
# setenv m "1360x768p60hz"
# setenv m "1440x900p60hz"
# setenv m "1600x900p60hz"
# setenv m "1680x1050p60hz"
# setenv m "1600x1200p60hz"
# setenv m "1920x1200p60hz"
# setenv m "2560x1080p60hz"
# setenv m "2560x1440p60hz"
# setenv m "2560x1600p60hz"
# setenv m "3440x1440p60hz"
```

2D performance under Ubuntu is not great because of missing hardware acceleration. Kodi and Android do have hardware acceleration for video playback. Adding acceleration to X11 should be possible for someone skilled or Hardkernel, but there are no signs of it (see http://forum.odroid.com/viewtopic.php?f=135&t=19434).

On a more positive note: libsurvive compiles on the C2 with Ubuntu after `apt install libusb-1.0-0-dev libz-dev`. As I said, I don't have a Vive, so all I get is
```
Loading driver DriverRegHTCVive (0x7f80ea596c) (1)
Info: Did not find device HMD (0bb4:2c87.0)
Info: Did not find device Lighthouse (28de:2000.0)
Info: Did not find device Watchman 1 (28de:2101.0)
Info: Did not find device Watchman 2 (28de:2101.1)
Info: Did not find device Tracker 0 (28de:2022.0)
Info: All enumerated devices attached.
*CALLING 0x419010 0x419e60
Info: Powered unit on.
Driver DriverRegHTCVive reports status 0
Available posers:
	 PoserCharlesSlow
	*PoserDummy
	 PoserDaveOrtho
Totals 3 posers.  Using selected poser (or first!).
Error: Error: cannot find all devices needed for calibration.
```

If anyone has ideas how to get custom HDMI resolutions working, tell me how and I will test it.

geekmaster, 2017-03-14:
That odroid thread does not confirm that it worked beyond 60Hz. Another thread suggests a kernel change and recompile for custom resolutions: http://com.odroid.com/sigong/blog/blog_list.php?bid=115

cnlohr, 2017-03-15:
Some people are suggesting a raspi might be able to do it. 

See this thread: https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=24679&start=50

```
hdmi_pixel_freq_limit=300000000
```

cnlohr, 2017-03-15:
....and this.... https://www.raspberrypi.org/forums/viewtopic.php?f=38&t=79330
```
At turbo mode and increasing the pixel clock limit, I got 4K@50hz!!!
But I view that the VPU is running at full tilt.
```

cnlohr, 2017-03-17:
After spending a very long time chasing my tail, I think I'm doing something fundamentally wrong, since Rpi mode 4 group 2 displays on the vive's right eye (640x480) but, when I manually set the modeline, no dice.  See this gist: https://gist.github.com/cnlohr/152d4b7f35a7a76e0688ee4305d81a0b - I really think this may be possible.

geekmaster, 2017-03-28:
The odroid-c2 kernel already has a predefined pixel clock of 297.60MHz, which should be close enough to the 297.00MHz that the Vive EDID shows. I am working on kernel patches (19 files) now, but I will be busy the next couple of days. I hope to complete this task and test it soon.

cnlohr, 2017-03-28:
As a side-note.  Just found this that lets you update the raspi modelines on the fly.  https://github.com/raspberrypi/firmware/issues/637

ChristophHaag, 2017-07-14:
Seeing https://www.kickstarter.com/projects/librecomputer/libre-computer-board-next-gen-4k-sbc-dev-board-for made me immediately think of this thread.

cnlohr, 2017-07-19:
awesome! Man. I need to get back into this.  I did get more of my sensors working last night.

cnlohr, 2018-06-18:
Here is the info for getting Raspberry Pi 3 running on the HTC Vive: https://gist.github.com/cnlohr/152d4b7f35a7a76e0688ee4305d81a0b

https://gist.github.com/cnlohr/152d4b7f35a7a76e0688ee4305d81a0b#gistcomment-2621874 and immediately above.

I'm happy with the raspi, now and will be sticking with it despite it's AWFUL issues surrounding draw calls.

closed 2018-06-18 by cnlohr

VKoskiv, 2018-06-18:
Props for raspi support. It's a super common hardware configuration.