collabora/libsurvive #81: Hardware to connect directly to VR Controllers (ie without Vive)?

mungewell, 2017-06-21:
Hi, recently found the project and, although I don't own a Vive, I think what you are doing is really cool (and interesting)....enough of the worship :-)

In your recent 'package from Alan' video you showed some USB dongles which look a lot like the one shipped with the regular/non-VR controllers. I was able to to confirm that those contain the nRF24L01 (just like the Vive headset), and also remember something about the original controllers being shipped with a dongle to enable additional controllers to be attached to the PC.

Found the following page which confirms this, and also details the f/w updating procedure.
https://developer.valvesoftware.com/wiki/SteamVR/HowTo_Update_Firmware

Perhaps the regular controller dongles can be re-flashed the same way to enable VR controllers without the headset?

The Lighthouse base stations and VR Controllers are available as separate items, from the steam store.

There is some documentation of the VR-Controller HID protocols here:
https://github.com/nairol/LighthouseRedox/blob/master/docs/USB%20Protocol.md#wireless-controller-dongle

I also found this project which has reversed some of the protocols used for the non-VR controller:
https://dennis-hamester.gitlab.io/scraw/protocol/

mungewell, 2017-06-22:
3rd hand confirmation...
https://steamcommunity.com/app/358720/discussions/0/357286532029364178/

mwturvey, 2017-06-22:
Hey mungewell, welcome!  I'm not sure if you've come across it yet, but the HTC tracker ships with a similar dongle.  AFAIK, it's effectively identical to the built in dongles in the headset.  USB device ID is identical, and you can either pair a tracker to it or another wireless controller.  You can also use it without any HMD attached to the computer.  

By the way, when you say "non-VR controller", are you referring to just using the regular controller without Valve's software, or something else?

P.S.  Awesome link on reversing the protocols, I haven't come across that one yet.

P.P.S.  Feel free to jump in on our discord channel.  
https://discordapp.com/invite/7QbCAGS

TCL987, 2017-06-22:
You can flash Steam Controller dongles with Vive controller dongle firmware. The only issue is that the firmware included with SteamVR is out of date so SteamVR will prompt you to update but doing so will revert the dongle back to a Steam Controller dongle.
https://steamcommunity.com/app/507090/discussions/0/312265473878773762/

Also I think the firmware file included with SteamVR has moved to "SteamVR\tools\lighthouse\firmware\vr_controller\archive\htc_vrc_dongle_1461100729_2016_04_19.bin".

closed 2020-03-04 by jdavidberger