collabora/libsurvive #111: New SVD linmath tools

cnlohr, 2018-03-12:
As far as I understand, some folks need some SVD and other decomposition algorithms.  I wanted to float the idea to see if Dave's stuff works.

[bidiagonalize.txt](https://github.com/cnlohr/libsurvive/files/1801389/bidiagonalize.txt)
[givens.txt](https://github.com/cnlohr/libsurvive/files/1801390/givens.txt)
[svd.txt](https://github.com/cnlohr/libsurvive/files/1801391/svd.txt)

jdavidberger, 2018-03-13:
Maybe I'm not running it right, but I'm getting ~2.5 seconds to run a 4x20 and about ~12 seconds to run a 22x15. 

For comparison the same matrix in opencv isn't really measurable at those sizes. 40x200 takes 0.25s or so. 
[svd.c.txt](https://github.com/cnlohr/libsurvive/files/1805356/svd.c.txt)
is the file with the modification to run opencv. 

```gcc svd.c -lm -O3 -o svd -I/usr/include/opencv -lopencv_core```

to build it. Pass in an argument to run opencv version.

I don't want to undercredit the accomplishment of a self contained small SVD c file -- that math is hard and I honestly don't understand most of it. For small matrices it seems like it runs slightly faster than opencv (hard to say though -- one is static vs dynamic allocation), and it does seem to optimize for symmetrical. 

But the bottom line for me is that as soon as we need anything medium and above its immediately going to stall the pipeline. Not to mention we'd have to write adapters for it and any other library we want, like SBA. I really think lapack/clapack is the way to go here.

ultramn, 2018-03-13:
Hey, I wrote the code and I think the housholder (svd c) is not supposed to be run on it's own but after you precondition it with the bidiagonalization.  Obviously I need to tie the programs together into one to have a competitive solution, but I just haven't had time.

The givens is the simpler algorithm.  It's an eigenvector algorithm though, so you need to take a covariance matrix to use it.  Givens is slower than the housholder for large matrices, but for smaller ones it should be fine.

I will have time to work on this Wednesday

jdavidberger, 2018-03-16:
The full range of lapacke functionality we are likely to need long term is here:

https://github.com/balintfodor/sba/blob/master/sba_lapack.c
lines 50-83

Some of those could trivially be implemented with less specialized forms for a performance penalty, we can evaluate that in context. 

We also have a dependency on BLAS dgemm as well; but that is just a matrix multiply which you'd likely be hard pressed to do without to implement svd. 

We are currently using dgelss as well, but that can be rewritten in terms of dgesvd. I'm honestly not sure if there is a performance saving for dgelss or if its just a convenience.  

Optimally we'd have a lib with identical function signatures to lapack so that any low level library we might use down the line should just drop in.

ultramn, 2018-03-16:
Understood.  I'll continue to work on performance improvements.  But.... givens.c is already fast enough to be useful in some circumstances.

If you're only using the 20x12, then givens.c may be more than fast enough.  In testing the performace of givens.c, it solves the 12x12 eigenvector problem in 0.0003s and the 15x15 eigenvector om 0.002s.  Which means that you can sove the Mx12 svd in this same amount of time.

givens.c also outperforms OpenCV for the Mx40 matrix, because it only takes 0.5s to solve the 40x40 eigenvector problem

In order to scale to larger matrices, however, I will need to implement a better method once the matrix is in tridiagonal form.  I'm working on implementing the "divide-and-conquer" eigenvalue method, because this runs in quadratic time so most of the time will be spent converting the matrix to tridiagonal in that case.

   Regars

jdavidberger, 2018-03-16:
Is there a repo location with a callable givens.c? Specifically I'd just wrap this: http://www.netlib.org/lapack/explore-html/d0/dee/lapacke__dgesvd_8c_af31b3cb47f7cc3b9f6541303a2968c9f.html#af31b3cb47f7cc3b9f6541303a2968c9f declaration around it. (We only need a row major impl; we can assert if its given column)

I could then give you performance numbers in use. 

If you are doing all your work in place for SVD -- which I sort of expect -- be sure to compare it against opencv with CV_SVD_MODIFY_A turned on. Also possibly run it twice and use the second timings -- the first pass likely sets up a thread local work buffer dynamically and keeps it around. I'm probably telling you stuff you already are checking though.

Would it be difficult / feasible to make all your diag matrices output 1xMIN(N, M)?

ultramn, 2018-03-16:
Do you use a c 2D array or are you calling malloc?

cnlohr, 2018-03-16:
Just FYI to any reading this thread, it was taken to the Discord channel.  We will report back with decision and close this thread here.  @ultramn and @jdavidberger please update and close this thread.

jdavidberger, 2018-03-19:
For SBA, we need any _one_ of the methods in sba to be available. LU looks like it will work if I can can it with a dynamic array. 

I would like cholesky decomposition at some point - my understanding is that it should be somewhat faster for sba.

Input into SBA is an NxN matrix where N is the number of parameters we mean to solve for; currently this is 7. At some point we'll try to tackle the calibration issue and we'll likely see something like... 17x17. 

We will need a SVD. It is running a 12x12 for epnp. For rotation matching stuff, I think we'll need ~22x22 still though. 

The matrix multiplies probably matters performance wise a lot. We calculate A^T*A with a 12x24. Ideally that transpose is a flag we can pass in. 

So that list is:
- cblas_dgemm
- dgetrf - LU factorization
- dgetri - Use LU factorization from dgetrf, invert A
- dgetrs - Solve Ax=b with LU factorization from dgetrf
- dgelss - Minimize Ax-b with SVD
- dgesvd - SVD
- dpotrs (optional) -  Solve Ax=b with Cholesky factorization

closed 2020-08-17 by jdavidberger