collabora/libsurvive #128: compilation on Windows 10

fcampos, 2018-04-14:
Hello, 
thanks for making this great work available to the community.
I am trying to use libsurvive on windows, but I am afraid I may lack some expertize in doing this. 
Running build_tcc.bat I got the following error:

C:\Users\FranciscoMateus\Documents\libsurvive-master\winbuild>C:\tcc\tcc.exe -v -DNOZLIB -DTCC -DWINDOWS -DHIDAPI -DWIN32 -DRUNTIME_SYMNUM -O0 -g -rdynamic -I..\redist -I..\include\libsurvive -I..\src -I. ..\src\ootx_decoder.c ..\src\poser_charlesslow.c ..\src\poser_daveortho.c ..\src\poser_dummy.c ..\src\survive.c ..\src\survive_cal.c ..\src\survive_config.c ..\src\survive_data.c ..\src\survive_driverman.c ..\src\survive_process.c ..\src\survive_vive.c ..\redist\crc32.c ..\redist\linmath.c ..\redist\puff.c ..\redist\jsmn.c ..\redist\json_helpers.c  ..\redist\symbol_enumerator.c ..\calibrate.c ..\redist\CNFGWinDriver.c ..\redist\os_generic.c ..\redist\CNFGFunctions.c -lkernel32 -lgdi32 -luser32 -lsetupapi -ldbghelp tcc_stubs.c ..\redist\hid-windows.c -o calibrate.exe
tcc version 0.9.27 (i386 Windows)
-> ..\src\ootx_decoder.c
-> ..\src\poser_charlesslow.c
In file included from ../src/poser_charlesslow.c:1:
../redist/linmath.h:75: error: declaration for parameter 'cross3d' but no such parameter

Do you have any suggestion to overcome this error?

Thank you very much

cnlohr, 2018-04-14:
I regret that the tcc compile is a few months old. I'll have to update it... but I don't normally have windows so you could update the .bat based on the makefile. The specific bug you're running into is quite odd though. I'll try compiling against tcc on Linux today. What version of tcc are you using?

fcampos, 2018-04-14:
Thanks for your reply!
I am using tcc-0.9.27-win32.
How about the libsurvive.sln solution? I also tried compiling this, with visual studio 2013, but got a few errors.

jdavidberger, 2018-04-14:
The sln should work, what errors are you getting?

fcampos, 2018-04-14:
I am getting errors in each instance of the keyword 'inline'. I was able to replace it with '__inline', which seemed to solve it.
Then I got errors in survive_vive.c in lines such as 'static uint8_t vive_magic_enable_lighthouse2[5] = { 0x07, 0x02 };'
Something to do with declaring a variable deep inside the function?

jdavidberger, 2018-04-14:
I think 2013 is the issue. We commonly run it on 2015. The SLN has v141 toolset as its default; and I don't think we want to change that. The inline stuff we can IFDEF to a version of VC++; but if there are a lot of other issues I'm not sure it is worth it since the SLN is 2015 specific already. 

I don't think we want a separate 2013 solution since we already have enough issues keeping all the different project configurations up to date; and I'm not even sure 2013 supports all the C99 constructs we use

fcampos, 2018-04-14:
That's fine, I am glad to know what the issue is. I will upgrade to a newer version of visual studio. If any further issues come up, I will post again. Thanks a lot for your help!

jdavidberger, 2018-12-04:
We now have cmake support which targets 2013; but if 2013 can't handle the c99 constructs it can't be supported.

closed 2018-12-04 by jdavidberger