collabora/libsurvive #131: Vive VR Controllers - BLE capable?

mungewell, 2018-05-17:
So it appears that the Steam Controller (non-VR gamepad) can support BLE connections to use with Steam's new Android Steam Play app.
https://support.steampowered.com/kb_article.php?ref=7728-QESJ-4420

Interesting.... since the VR Controllers use the same dongle hardware (different f/w on it though, see bug #81), does this raise the possibility that they (VR controllers) could also use BLE in the future?

mungewell, 2018-05-23:
hmmm..... seems like the Lighthouses are BLE already.
https://github.com/nairol/LighthouseRedox/blob/master/docs/Base%20Station.md#bluetooth-le-communications

dan2wik, 2018-05-24:
The chips that the controllers use are capable of BLE, they use nRF51 chips, I'm not certain if it is the nRF51822 like the steam controller or not. It talks nRF24 by the looks of it so It would be a firmware limitation preventing use of BLE but its certainly possible to write your own.

closed 2020-08-17 by jdavidberger