collabora/libsurvive #134: information on how to force an orange pi to display to the vive hmd?

Lightsteed, 2018-06-10:
Hi, cool project you have here! Im currently trying to use a mini computer ( orange/raspberry pi etc) to do wireless vr streaming from steam using riftcat/moonlight/ALVR or similar) i see that you are able to display to the vive from an orange pi and wondering if you can share with me what custom resolution you had yo do to acheive that, and some steps on how to do that myself. I can get my way around rasbian mainly due to the huge amount of info there is on the platform, but finding it hard to get info on the orange pi debian jessie. I have an orange pi lite2. Thank you!

closed 2018-06-12 by Lightsteed

cnlohr, 2018-06-18:
Hey! Just FYI - I found out you can use the newest Raspberry Pi 3 B+ to do it!

cnlohr, 2018-06-18:
See description at the bottom of this gist:

https://gist.github.com/cnlohr/152d4b7f35a7a76e0688ee4305d81a0b

https://gist.github.com/cnlohr/152d4b7f35a7a76e0688ee4305d81a0b#gistcomment-2621874
and the comment immediately above.