collabora/libsurvive #137: Unable to complete calibration

ortasa, 2018-08-11:
First thank you. It is a great project that open so much possibilities.

I am unable to complete the calibration script.  I can not see in the servive debug gui screen the location points. [image]

I am using 1 lighthouse and 1 tracker. 
Bellow is the ./calibrate output and the config.json file. 

Thanks in advance, 
Ortal

**./calibrate output:** 
Info: Available Posers:
Info:      Dummy
Info:      IMU
Info:     *CharlesRefine
Info:      EPNP
Info:      SBA
Info: Totals 5 Posers.
Info: Using 'CharlesRefine' for defaultposer
Info: Available Disambiguators:
Info:     *Charles
Info:      Turvey
Info:      StateBased
Info: Totals 3 Disambiguators.
Info: Using 'Charles' for disambiguator
Info: Loading driver DriverRegPlayback (0x7f8c5ff38b50) (1)
Info: Driver DriverRegPlayback reports status 0
Info: Loading driver DriverRegHTCVive (0x7f8c5ff5b6f0) (2)
Info: Blacklisting -
Info: Vive starting in libusb mode.
Info: Did not find device HMD (0bb4:2c87.0)
Info: Did not find device HMD IMU & LH (28de:2000.0)
Info: Did not find device Watchman 1 (28de:2101.0)
Info: Did not find device Watchman 2 (28de:2101.1)
Info: Successfully enumerated Tracker 0 (6, 3)
Info: Did not find device Tracker 1 (28de:2300.0)
Info: Did not find device Wired Watchman 1 (28de:2012.0)
Info: All enumerated devices attached.
Info: Powered unit on.
Info: Got config data length 1865
Info: Loading config: 7604
Info: Driver DriverRegHTCVive reports status 0
Info: Loading driver DriverRegDummy (0x7f8c5ff60c80) (3)
Info: Driver DriverRegDummy reports status 0
Info: Loading driver DriverRegUDP (0x7f8c5ff61010) (4)
Info: Driver DriverRegUDP reports status 0
Info: Uncalibrated configuration detected. Attaching calibration. Please don't move tracked objects for the duration of calibration. Pass '--disable-calibrate' to skip calibration
Info: Calibration is using TR0

**config.json file**

"lighthousecount":"1",
"defaultposer":"CharlesRefine",
"disambiguator":"Charles",
"RequiredTrackersForCal":"TR0",
"AllowAllTrackersForCal":"0",
"ConfigPoser":"CharlesRefine",
"TurveyToriDebug":"0"
"lighthouse0":{
"index":"0",
"id":"0",
"mode":"0",
"pose":["0.000000","0.000000","0.000000","0.000000","0.000000","0.000000","0.000000"],
"fcalphase":["0.000000","0.000000"],
"fcaltilt":["0.000000","0.000000"],
"fcalcurve":["0.000000","0.000000"],
"fcalgibpha":["0.000000","0.000000"],
"fcalgibmag":["0.000000","0.000000"],
"PositionSet":"0"
}
"lighthouse1":{
"index":"-1"
}

cnlohr, 2018-08-13:
I don't actually know how to calibrate with just a tracker as I only have the vive and controllers, but I think I remember @jdavidberger saying you could do it with something like...

```./calibrate --requiredtrackersforcal TR0 --allowalltrackersforcal 1```

Hopefully he can provide some input.

jdavidberger, 2018-08-13:
Is there no further output from the calibration run? 

The flags @cnlohr mentioned used to be necessary but it now will default into whatever device is plugged in; and the log output correctly seems to be using TR0. 

The first gen trackers have a harder time correctly getting OOTX data then any other device. It seems to see all the pulses, but the pulse timings are sometimes off enough to not be able to pass checksums. However, if that is what you were seeing, I'd expect to see a bunch of error messages from OOTX parsing. This output looks like it doesn't see any light data at all. 

What does the output look like if you just run 

```
./data_recorder
``` 
It should flood you with IMU and lightcap data; if it doesn't there is something goofy with the USB. I'll check my TR0 to make sure its working later; possibly we broke it and didn't notice. 

If you are seeing light data, possibly try something like:

```
./calibrate --disambiguator StateBased --ootx-ignore-sync-error
```

The first option there uses a different disambiguator, which might be more resiliant to TR0 noise. The second flag ignores sync errors on the OOTX stream, which allows certain sync bits to not derail OOTX. It still does a checksum so if OOTX finishes, even with a bad sync bit somewhere, its very likely the results are correct.

ortasa, 2018-08-16:
Thank you for all your help.
The problem was because the lighthouse was on mode C. When we change the lighthouse mode to A every thing work OK.
Thanks,
Ortal

closed 2018-08-16 by ortasa

cnlohr, 2018-08-17:
@jdavidberger I didn't know the stuff worked with just one lighthouse? Neat?