collabora/libsurvive #141: button_process: SurviveObject is 0 when pressing the grip button or when (first) touching the touchpad

mortaas, 2019-02-06:
Hello, I currently have problems with implementing my own button process in libsurvive. I am currently using the so->user_ptr as a device index, and my program crashed whenever I pressed the grip button or touched the touchpad. I found that the so pointer was 0 under these conditions, and implemented a simple check to avoid using the SurviveObject pointer whenever it was 0.

This is the current code for my button process:

```
static void button_process(SurviveObject *so, uint8_t eventType, uint8_t buttonId, uint8_t axis1Id, uint16_t axis1Val, uint8_t axis2Id, uint16_t axis2Val)
{
    survive_default_button_process(so, eventType, buttonId, axis1Id, axis1Val, axis2Id, axis2Val);

    void *ptr = so;
    // printf("%x \n", ptr);

    if (ptr != nullptr) {
        struct SurviveSimpleContext *actx = (struct SurviveSimpleContext *) so->ctx->user_ptr;
        intptr_t i = (intptr_t) so->user_ptr;

        OGLockMutex(actx->poll_mutex);

        switch(eventType) {
            case BUTTON_EVENT_BUTTON_DOWN :
                controller_device_index.push(i);
                controller_event_type.push(200); // ButtonPress
                break;
            case BUTTON_EVENT_BUTTON_UP :
                controller_device_index.push(i);
                controller_event_type.push(201); // ButtonUnpress
                break;
            case BUTTON_EVENT_AXIS_CHANGED :
                if (axis1Val + axis2Val != 0.) {
                    controller_device_index.push(i);
                    controller_event_type.push(202); // ButtonTouch
                } else {
                    controller_device_index.push(i);
                    controller_event_type.push(203); // ButtonUntouch
                }
                break;
            case BUTTON_EVENT_BUTTON_NONE :
                break;
        }

        OGUnlockMutex(actx->poll_mutex);
    }
```

cnlohr, 2019-02-07:
Your printf that's commented out is 0?!?  Can you share more code? Honestly I don't know for sure we've ever used it this way so that might be untested but still want to verify before digging.

mortaas, 2019-02-07:
The steps to reproduce this error is basically:

1. Install button process function
2. Access any member of so (SurviveObject*) in the callback
3. Press the trigger button, or touch the touchpad
4. CRASH

I could add that it behaves the same with my code in the callback commented, just access any member of so.

The printf prints 0 in two very specific cases:
1. Pressing the grip button
2. When you first touch the touchpad for a single callback

You could use this button process function to reproduce this behaviour:
```
static void button_process(SurviveObject *so, uint8_t eventType, uint8_t buttonId, uint8_t axis1Id, uint16_t axis1Val, uint8_t axis2Id, uint16_t axis2Val)
{
    survive_default_button_process(so, eventType, buttonId, axis1Id, axis1Val, axis2Id, axis2Val);

    void *ptr = so;
    printf("%x \n", ptr);
}
```

jdavidberger, 2019-02-21:
The issue should be fixed now; please let me know if it still happens and thanks for filing the issue!

commit 4089dc1f (2019-02-21): Added button output to CLI tool; fixed so == null issue in button process -- fixes #141
closed 2019-02-21 by jdavidberger