collabora/libsurvive #146: ./calibrate: calibrate: src/survive_disambiguator.c:23: handle_lightcap: Assertion `le->sensor_id < 32' failed.

HWiese1980, 2019-07-04:
Hi, 

I'm trying to get this running on my Ubuntu 16.04 x86_64. It compiles, but upon `./calibration` it crashes after the assertion `calibrate: src/survive_disambiguator.c:23: handle_lightcap: Assertion `le->sensor_id < 32' failed.` failed. 

My setup only consists of two Vive Index base stations and a single 2018 Vive Tracker. No HMD, no Controllers, just that. Is that a valid and supported setup?

Thank you

HWiese1980, 2019-07-04:
Here's the output: 

```
Info: Successfully enumerated Tracker (2018) (3) 28de:2300
Info: Successfully enumerated Tracker (2018) 28de:2300
Info: Adding tracked object T20 from HTC
Info: Attaching IMU(0x81) for T20
Info: Attaching Lightcap(0x83) for T20
Info: Attaching Buttons(0x84) for T20
Info: All enumerated devices attached.
Info: Powered unit on.
Info: Got config data length 2211
Info: Loaded drivers: HTCVive
Info: Lighthouse 0 position is unset
Info: Lighthouse 1 position is unset
Info: Uncalibrated configuration detected. Attaching calibration. Please don't move tracked objects for the duration of calibration. Pass '--disable-calibrate' to skip calibration
Info: Calibration is using T20
calibrate: src/survive_disambiguator.c:23: handle_lightcap: Assertion `le->sensor_id < 32' failed.
Abgebrochen (Speicherabzug geschrieben)
```

jdavidberger, 2019-07-05:
The issue is that you are using the index lighthouses -- currently libsurvive doesn't support the lighthouse v2 (index) technology stack.

Hopefully this changes soon -- I've gotten a few lighthouses and am looking at the data streams. 

I've checked in code that detects this and issues an error unless you explicitly opt in to seeing debug data for the lighthouse version 2 details.

closed 2019-08-10 by jdavidberger