collabora/libsurvive #148: can libsurvive be used under 2nd lighthouse basestation?

chopin1998, 2019-08-14:
it show me 

`Warning: Could not find disambiguator state for WW0
Warning: Could not find disambiguator state for WW0
Warning: Could not find disambiguator state for WW0
Warning: Could not find disambiguator state for WW0
Warning: Could not find disambiguator state for WW0
Warning: Could not find disambiguator state for WW0
Warning: Could not find disambiguator state for WW0
Warning: Could not find disambiguator state for WW0
Warning: Could not find disambiguator state for WW0
Warning: Could not find disambiguator state for WW0
Warning: Could not find disambiguator state for WW0
Warning: Could not find disambiguator state for WW0
Warning: Could not find disambiguator state for WW0
Warning: Could not find disambiguator state for WW0
Warning: Could not find disambiguator state for WW0
Warning: Could not find disambiguator state for WW0
Warning: Could not find disambiguator state for WW0
Warning: Could not find disambiguator state for WW0
Warning: Could not find disambiguator state for WW0
Warning: Could not find disambiguator state for WW0
Warning: Could not find disambiguator state for WW0
Warning: Could not find disambiguator state for WW0
Warning: Could not find disambiguator state for WW0
Warning: Could not find disambiguator state for WW0
Warning: Could not find disambiguator state for WW0
Warning: Could not find disambiguator state for WW0
`

jdavidberger, 2019-08-14:
Should be fixed in latest master; please try again and see if the issue is fixed.

closed 2020-03-04 by jdavidberger

mossti, 2022-06-06:
Hi! I wasn't sure if I should start a new issue or to reopen this one. I'm having the same problem as described by the original poster. I am on Ubuntu 20.04, and have two Gen 1 Vive lighthouses and 1 Gen 1 Vive controller connected via USB to my machine.

[image]

The above screenshot is what happens when I used the command `./bin/survive-cli`; I see the same behavior when including any combination of the following flags: `--force-calibrate --lighthousecount 2 --lighthouse-gen 0 --disambiguator StateBased`.  When trying to force calibration, I have been deleting the `config.json` file under the `~/.config/libsurvive/` directory; every time that the has file has been rebuilt the contents are as follows:

[image]