collabora/libsurvive #150: Degenerate poses and noisy data

asfluido, 2019-09-06:
Good day. I am an old (almost ancient!) programmer working at the university of Bern, Switzerland. I am toying with HTC Vive hardware that had been previously purchased (no idea when) with the grand plan to set up an experiment where subjects will be located on a 6-degrees-of-freedom platform, wearing the headset.

First of all, **lots of compliments** to those who are behind the development of this library. It is exactly what I was looking for. I will deal with creating the openGL graphics (that's my speciality...), but libsurvive takes care of the **very** complex part of extracting the pose data from the system - a task about which I would not have known where to start. 

I set up a test environment by positioning two lighthouses, and then laying a headset and a wired watchman on the seat that is mounted upon the platform. This image should provide an idea of the setup:

[image]

The tentative procedure I follow to initialize the library from a clean setup is as follows:

- I remove all json files plus the calinfo directory from my base directory.
- I power the lighthouses.
- I plug in the USB cables of the headset and the wired watchman.
- I run the *simple_pose_test* program you provide until it starts producing pose data for the two objects
- After that, I execute my code, that will make use of the calibration data previously obtained.

My code initializes libsurvive with these lines:

```c
  s->ctx=survive_init(0,NULL);
  if(!s->ctx)
    rb_raise(rb_eArgError,"%s: cannot open libsurvive",__func__);
  
  survive_install_log_fn(s->ctx,log_fn);
  survive_install_pose_fn(s->ctx,pose_fn);
```
(the code is called from the Ruby interpreter). `log_fn` is very simple:

```c
static void log_fn(SurviveContext *ctx,SurviveLogLevel log_level,const char *msg)
{
  lg("Survive (%d): %s",log_level,msg);
}
```
(lg is my standard logging subroutine that tags messages with time and date). `pose_fn` is a bit more complex:

```c
static void pose_fn(SurviveObject *so,uint32_t tc,SurvivePose *pose)
{
  struct timespec now;

  clock_gettime(CLOCK_MONOTONIC,&now);
  write(spub->log_unit,&now,sizeof(struct timespec));
  
  survive_default_pose_process(so,tc,pose);

  dev_stc *dev=find_device(spub,so->serial_number);

  if(spub->last_prog<dev->prog)
  {
    write(spub->log_unit,"S",1);
    uint16_t slen=strlen(dev->srl);
    write(spub->log_unit,&slen,2);
    write(spub->log_unit,dev->srl,slen);
    spub->last_prog=dev->prog;
  }
  write(spub->log_unit,&dev->prog,sizeof(uint16_t));
  write(spub->log_unit,&pose->Pos,sizeof(FLT)*3);
  write(spub->log_unit,&pose->Rot,sizeof(FLT)*4);
}
```

What it does is just to save the pose data. Everything works OK. 

The file attached below contains the log messages generated by libsurvive and captured by `log_fn` during a session where I asked the platform to move a bit along its six degrees of freedom, in sequence (the headset and watchman moving together with it):

[log.txt](https://github.com/cnlohr/libsurvive/files/3583326/log.txt)

The archive file attached next contains four graphs, two for the headset and two for the watchman. They compare the linear and angular positions of the platform and of the two objects. I obtain yaw, pitch and roll from the quaternion, and I am sure I will find the right way to map the two sets of motion data in a coherent way. The returned motion of the Vive objects is consistent with that of the platform, and that is a very good starting point.

[graphs.zip](https://github.com/cnlohr/libsurvive/files/3583343/graphs.zip)

I would like to find the answer to the following questions, and will appreciate any help that you will provide:

- I receive data from the headset constantly, but data from the watchman are missing during most of the period. Is this to be expected? What could be the reason?
- The pose data I receive is noisy. Is this to be expected? I certainly can smooth out the data, but I am wondering whether there is some sort of well-known trick in the set-up or the interface to the library that I am omitting or doing wrong. 
- I receive lots of these `pose is degenerate` and `MPFIT failure` messages. Is there something I can do about them?

Thanks in advance.

Carlo

jdavidberger, 2019-09-07:
It sounds like one or some combination of the following:

- Bad initial calibration results
- Dropped USB packets for the WW devices
- The solvers aren't running fast enough and so it is effectively dropping data. 

Dropped USB packets would happen if the USB hub / host was congested -- possibly if you had a USB2 hub and not a USB3 hub. Worth mentioning; but probably the least likely of the possible issues but the easiest to fix. 

From the setup, a bad initial calibration would only happen if there was a bug -- it seems like ideal conditions; no giant mirrors etc. I would verify it visually in the viz tool or some other way to plot the lighthouse positions.

If the process is pegging the CPU at 75-100% it might be just running too slow. Try '--syncs-per-run 4' to solve once every 4 sync frames (default its once per sync frame), which might make it less likely to drop data. 

It'd help me to know more of what is going on with it if you send in a few recording files -- one which is just everything standing still, and good for calibration, and another which is with some or all of the stuff in motion. To generate a recording, just pass in '--record <log-name>.rec.gz --defaultposer Dummy --configposer Dummy' as an argument to either survive-cli or any other application that initializes survive with its command line parameters. 

For the calibration one, start it up and give it 30 seconds to a minute. It just has to be enough to get OOTX + enough data to solve the initial position for. Let me know if you have any questions!

jdavidberger, 2019-09-07:
Oh -- another important point -- using the callbacks like that is fine, but do execute on the thread that the solvers and USB read works on. If you are slow through the callbacks, it will cause tracking issues. I'd suggest using the 'simple API' interface if you are going to do heavy lifting with the poses. That API will let you safely get updates / poses / button presses in a separate thread.

asfluido, 2019-09-09:
Thanks a lot for your reply. 

First of all, the absence of data for the watchman was due to it being connected to a USB2 port. I connected it to USB3 (the second and last port available on my motherboard) and I had data all the time. The unpleasant fact is that the data for the watchman was noisier than that for the headset. 

For the moment, since the first concrete project I'll have to work on only requires data from the headset, I just disconnected the watchman.

I have dumped log data as per your request. You can find two files in this archive:

[recs.zip](https://github.com/cnlohr/libsurvive/files/3590272/recs.zip)

- File `spt.rec` includes the log file generated while running the calibration phase (by running program `simple_pose_test`). I let the program run for a few seconds after the `Stage 4 succeeded` message was generated, without moving the headset.
- File `.rec` includes the log file generated while running my code, during a set of platform motions equal to the one that I used the other day.

I also attach the list of log messages generated during that run:

[log.txt](https://github.com/cnlohr/libsurvive/files/3590274/log.txt)

I add two more graphs, to show that the noise is comparable with what I sent previously:

[graphs.zip](https://github.com/cnlohr/libsurvive/files/3590276/graphs.zip)

What is the theoretical meaning of the "MPFIT failure" and "pose is degenerate" messages? I see that 'MPFIT' is the poser. Could it be adviceable to try other posers?

I run the libsurvive task in its own thread. I verified that the thread (which just loops calling `survive_poll`) consumes only 35% of one thread (this machine has an Intel Core i7-3770 CPU @ 3.40GHz). I believe this to be acceptable.

What is the 'viz tool' that you mention? Is it something I can download and install under Linux?

I am available to run any test.

Thanks again.
Carlo

jdavidberger, 2019-09-10:
Did you generate those recordings off of the USB2 port or the USB3 port? From the recordings, it looks like there is still data being dropped. 

Look here: https://github.com/cnlohr/libsurvive#visualization for info on how to set up the viz tool. 

I made a recording of the calibration run: https://www.youtube.com/watch?v=cwGwPkRfNYQ

Where you see the blinking indicators on the HMD / in the 2D view shows that its losing data on that sensor. You'd only expect a sensor to go dark when its no longer in the field of view of the lighthouse. 

I suspect it isn't just dropping data; but its getting confused on syncs every so often and you get the horizontal time for where the vertical time should be. I'll see if I can detect / reject these cases; it's probably what is causing really large jumps. But you'll want to solve the data rate issue for any real usage of this data even with that fixed. 

This will happen with every poser; as the issue is upstream of the solvers in general.

Degenerate poses are detected by being way outside of the observable space, or by the optimization pass failing to minimize the reprojection error to an acceptable amount. It almost always means some form of bad data; whether it be confused syncs or something mirror like in the room.

jdavidberger, 2019-09-10:
The large error spikes were due to IMU windup -- it'd go too long without an observation, and the IMU error would just blow up. I've fixed it to stop integrating the IMU after 1/20th of a second -- you should get 2 full XY pairs in that time. 

The light data is really noisy here too -- I'm seeing a lot of small pulses that are out of place. Could be due to reflections in the space; but I've seen older hardware have the same kind of characteristics. Might add some more outlier rejection logic; but if I filter purely based on light length any more than it already does now, it'll drop valid data. 

Is this on some sort of CNC thing? Here is the movement I saw:

[image]

Still noisy due to some bad light data; but looks far too regular to be done by hand.

asfluido, 2019-09-10:
Thanks again. 

First some data to fill you up. The 6-degrees-of-freedom platform looks like this:

[image]

The top platform can move in any direction (three linear and three angular degrees of freedom). Over it we have a structure that holds a seat, where persons can sit, while the platform under them moves. To make things clearer, I work for the institute of psychology, and the researchers study the perception of motion. 

The visor will have to be worn by subjects for experiments. Currently I am laying it on the seat, while my program moves the platform along the six axes of freedom in turn.  You can follow the motions in the graphs. The script I use for testing always performs the same motions.

Your fixes made the 'pose is degenerate' messages disappear! that's a **great improvement**. 

The devices I am using are certainly old. I will be able to ask precise details about when they were bought in about an hour's time, but they are most probably as old as they can be. 

The data is now much smoother. A new pack of graphs follows:

[graphs.zip](https://github.com/cnlohr/libsurvive/files/3594292/graphs.zip)

The run with time `090122` is from a run where I was not collecting log files (using the parameters you suggested), and the other run (`091815`) is from a run when I was saving the log, which you can find here:

[rec.zip](https://github.com/cnlohr/libsurvive/files/3594307/rec.zip)

Here you can find again my log extract, where you can see when the `MPFIT Failure` messages are generated:

[log.txt](https://github.com/cnlohr/libsurvive/files/3594312/log.txt)

At this point, with some smoothing, data will look great, but if you wish to experiment further, I thank you in advance.

Carlo