collabora/libsurvive #151: Building libsurvive (Windows 10/VS 2017)

mgavornik, 2019-09-08:
Hi, first of all thank you for the effort to create such a library. I would like to use it to extract raw tracking and imu data to test my own Kalman filter implementation but i am stuck at building the VS2017 solution under Windows 10. 

Am i missing something obvious ? Any help is appreciated :)

CMake config: 

[image]

Building the solution gives me the following errors: 
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK1104	cannot open file 'Debug\survive.lib'	poser_epnp	I:\Libraries\libsurvive\build\LINK	1	
Error	C1083	Cannot open include file: 'zlib.h': No such file or directory	survive	I:\Libraries\libsurvive\src\survive_optimizer.c	5	
Error	C1083	Cannot open include file: 'zlib.h': No such file or directory	survive	I:\Libraries\libsurvive\src\survive_playback.c	11	
Error	C1083	Cannot open include file: 'alloca.h': No such file or directory	survive	I:\Libraries\libsurvive\src\lfsr_lh2.c	2	
Error	C1083	Cannot open include file: 'pcap/usb.h': No such file or directory	driver_usbmon	I:\Libraries\libsurvive\src\driver_usbmon.c	13	
Error	C1083	Cannot open include file: 'openvr/openvr_driver.h': No such file or directory	driver_survive	I:\Libraries\libsurvive\tools\openvr_driver\openvr_driver.cc	1	
Error	C2059	syntax error: '}'	minimal_opencvtest	I:\Libraries\libsurvive\redist\minimal_opencvtest.c	50	
Error	C2059	syntax error: '}'	minimal_opencvtest	I:\Libraries\libsurvive\redist\minimal_opencvtest.c	64	
Error	C2059	syntax error: '}'	minimal_opencvtest	I:\Libraries\libsurvive\redist\minimal_opencvtest.c	80	
Error	C2059	syntax error: '}'	minimal_opencvtest	I:\Libraries\libsurvive\redist\minimal_opencvtest.c	83	
Error	C2059	syntax error: '}'	minimal_opencvtest	I:\Libraries\libsurvive\redist\minimal_opencvtest.c	86	
Error	C2059	syntax error: '}'	lintest	I:\Libraries\libsurvive\redist\lintest.c	147	
Error	C2059	syntax error: '}'	survive	I:\Libraries\libsurvive\src\poser.c	156	
Error	C2059	syntax error: '}'	survive	I:\Libraries\libsurvive\src\poser.c	204	
Error	C2059	syntax error: '}'	survive	I:\Libraries\libsurvive\src\survive_process.c	222	
Error	C2059	syntax error: ';'	driver_dummy	I:\Libraries\libsurvive\redist\os_generic.h	95	
Error	C2059	syntax error: '<parameter-list>'	driver_dummy	I:\Libraries\libsurvive\redist\os_generic.h	95	
Error	C2059	syntax error: ';'	driver_simulator	I:\Libraries\libsurvive\redist\os_generic.h	95	
Error	C2059	syntax error: '<parameter-list>'	driver_simulator	I:\Libraries\libsurvive\redist\os_generic.h	95	
Error	C2059	syntax error: ';'	driver_vive	I:\Libraries\libsurvive\redist\os_generic.h	95	
Error	C2059	syntax error: '<parameter-list>'	driver_vive	I:\Libraries\libsurvive\redist\os_generic.h	95	
Error	C2059	syntax error: '}'	driver_vive	I:\Libraries\libsurvive\src\driver_vive.c	2068	
Error	C2059	syntax error: '}'	driver_vive	I:\Libraries\libsurvive\src\driver_vive.c	2096	
Error	C2059	syntax error: ';'	driver_usbmon	I:\Libraries\libsurvive\redist\os_generic.h	95	
Error	C2059	syntax error: '<parameter-list>'	driver_usbmon	I:\Libraries\libsurvive\redist\os_generic.h	95	
Error	C2059	syntax error: '}'	poser_mpfit	I:\Libraries\libsurvive\src\poser_mpfit.c	189	
Error	C2059	syntax error: '}'	poser_mpfit	I:\Libraries\libsurvive\src\poser_mpfit.c	317	
Error	C2059	syntax error: ';'	test_replays	I:\Libraries\libsurvive\redist\os_generic.h	95	
Error	C2059	syntax error: '<parameter-list>'	test_replays	I:\Libraries\libsurvive\redist\os_generic.h	95	
Error	C2059	syntax error: ';'	test_replays	I:\Libraries\libsurvive\src\test_cases\test_replays.c	11	
Error	C2059	syntax error: '<parameter-list>'	test_replays	I:\Libraries\libsurvive\src\test_cases\test_replays.c	11	
Error	C2059	syntax error: '}'	test_replays	I:\Libraries\libsurvive\src\test_cases\test_replays.c	22	
Error	C2059	syntax error: '}'	test_replays	I:\Libraries\libsurvive\src\test_cases\test_replays.c	45	
Error	C2059	syntax error: ';'	survive_tests	I:\Libraries\libsurvive\redist\os_generic.h	95	
Error	C2059	syntax error: '<parameter-list>'	survive_tests	I:\Libraries\libsurvive\redist\os_generic.h	95	
Error	C2059	syntax error: '}'	survive_tests	I:\Libraries\libsurvive\src\driver_vive.c	2068	
Error	C2059	syntax error: '}'	survive_tests	I:\Libraries\libsurvive\src\driver_vive.c	2096	
Error	C2059	syntax error: '}'	api_example	I:\Libraries\libsurvive\api_example.c	33	
Error	C2061	syntax error: identifier 'OGGetAbsoluteTimeMS'	driver_dummy	I:\Libraries\libsurvive\redist\os_generic.h	95	
Error	C2061	syntax error: identifier 'OGGetAbsoluteTimeMS'	driver_simulator	I:\Libraries\libsurvive\redist\os_generic.h	95	
Error	C2061	syntax error: identifier 'OGGetAbsoluteTimeMS'	driver_vive	I:\Libraries\libsurvive\redist\os_generic.h	95	
Error	C2061	syntax error: identifier 'OGGetAbsoluteTimeMS'	driver_usbmon	I:\Libraries\libsurvive\redist\os_generic.h	95	
Error	C2061	syntax error: identifier 'OGGetAbsoluteTimeMS'	test_replays	I:\Libraries\libsurvive\redist\os_generic.h	95	
Error	C2061	syntax error: identifier 'diff'	test_replays	I:\Libraries\libsurvive\src\test_cases\test_replays.c	11	
Error	C2061	syntax error: identifier 'OGGetAbsoluteTimeMS'	survive_tests	I:\Libraries\libsurvive\redist\os_generic.h	95	
Error	C2065	'M_PI': undeclared identifier	survive	I:\Libraries\libsurvive\src\survive_reproject_gen2.c	73	
Error	C2065	'M_PI': undeclared identifier	survive	I:\Libraries\libsurvive\src\survive_reproject_gen2.c	99	
Error	C2065	'masks': undeclared identifier	driver_vive	I:\Libraries\libsurvive\src\driver_vive.c	2090	
Error	C2065	'times': undeclared identifier	driver_vive	I:\Libraries\libsurvive\src\driver_vive.c	2091	
Error	C2065	'masks': undeclared identifier	driver_vive	I:\Libraries\libsurvive\src\driver_vive.c	2097	
Error	C2065	'times': undeclared identifier	driver_vive	I:\Libraries\libsurvive\src\driver_vive.c	2097	
Error	C2065	'err': undeclared identifier	test_replays	I:\Libraries\libsurvive\src\test_cases\test_replays.c	80	
Error	C2065	'err': undeclared identifier	test_replays	I:\Libraries\libsurvive\src\test_cases\test_replays.c	83	
Error	C2065	'err': undeclared identifier	test_replays	I:\Libraries\libsurvive\src\test_cases\test_replays.c	84	
Error	C2065	'err': undeclared identifier	test_replays	I:\Libraries\libsurvive\src\test_cases\test_replays.c	86	
Error	C2065	'err': undeclared identifier	test_replays	I:\Libraries\libsurvive\src\test_cases\test_replays.c	97	
Error	C2065	'err': undeclared identifier	test_replays	I:\Libraries\libsurvive\src\test_cases\test_replays.c	99	
Error	C2065	'err': undeclared identifier	test_replays	I:\Libraries\libsurvive\src\test_cases\test_replays.c	101	
Error	C2065	'err': undeclared identifier	test_replays	I:\Libraries\libsurvive\src\test_cases\test_replays.c	102	
Error	C2065	'M_PI': undeclared identifier	survive_tests	I:\Libraries\libsurvive\src\test_cases\rotate_angvel.c	68	
[... 2 lines cut ...]
Error	C2065	'masks': undeclared identifier	survive_tests	I:\Libraries\libsurvive\src\driver_vive.c	2097	
Error	C2065	'times': undeclared identifier	survive_tests	I:\Libraries\libsurvive\src\driver_vive.c	2097	
Error	C2065	'NAN': undeclared identifier	sensors-readout	I:\Libraries\libsurvive\sensors-readout.c	151	
Error	C2109	subscript requires array or pointer type	driver_vive	I:\Libraries\libsurvive\src\driver_vive.c	2090	
Error	C2109	subscript requires array or pointer type	driver_vive	I:\Libraries\libsurvive\src\driver_vive.c	2091	
Error	C2109	subscript requires array or pointer type	survive_tests	I:\Libraries\libsurvive\src\driver_vive.c	2090	
Error	C2109	subscript requires array or pointer type	survive_tests	I:\Libraries\libsurvive\src\driver_vive.c	2091	
Error	C2146	syntax error: missing ';' before identifier 'err'	test_replays	I:\Libraries\libsurvive\src\test_cases\test_replays.c	80	
Error	C2146	syntax error: missing ';' before identifier 'err'	test_replays	I:\Libraries\libsurvive\src\test_cases\test_replays.c	97	
Error	C2440	'function': cannot convert from 'int' to '_Fcomplex'	test_replays	I:\Libraries\libsurvive\src\test_cases\test_replays.c	83	
Error	C2440	'function': cannot convert from 'int' to '_Fcomplex'	test_replays	I:\Libraries\libsurvive\src\test_cases\test_replays.c	84	
Error	C2440	'function': cannot convert from 'int' to '_Fcomplex'	test_replays	I:\Libraries\libsurvive\src\test_cases\test_replays.c	86	
Error	C2440	'function': cannot convert from 'int' to '_Fcomplex'	test_replays	I:\Libraries\libsurvive\src\test_cases\test_replays.c	99	
[... cut ...]
Error	LNK1104	cannot open file 'Debug\survive.lib'	poser_imu	I:\Libraries\libsurvive\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	disambiguator_statebased	I:\Libraries\libsurvive\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	disambiguator_charles	I:\Libraries\libsurvive\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	simple_pose_test	I:\Libraries\libsurvive\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	data_recorder	I:\Libraries\libsurvive\build\LINK	1

jdavidberger, 2019-09-09:
I think I fixed all the windows build issues on latest master; try again and let me know if you see any issues

mgavornik, 2019-09-10:
Thank you,  i still have a few errors: 

Severity	Code	Description	Project	File	Line	Suppression State
Error	C2059	syntax error: ';'	driver_vive	I:\Libraries\libsurvive\src\driver_vive.c	1172	
Error	C2120	'void' illegal with all types	survive	I:\Libraries\libsurvive\src\survive_playback.c	683	
Error	LNK1104	cannot open file 'Debug\survive.lib'	disambiguator_turvey	I:\Libraries\libsurvive\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	disambiguator_statebased	I:\Libraries\libsurvive\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	driver_dummy	I:\Libraries\libsurvive\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	driver_simulator	I:\Libraries\libsurvive\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	poser_imu	I:\Libraries\libsurvive\build\LINK	1	
Error	C1083	Cannot open include file: 'pcap/usb.h': No such file or directory	driver_usbmon	I:\Libraries\libsurvive\src\driver_usbmon.c	13	
Error	C2061	syntax error: identifier 'read_light_data'	driver_vive	I:\Libraries\libsurvive\src\driver_vive.c	1172	
Error	C2059	syntax error: '<parameter-list>'	driver_vive	I:\Libraries\libsurvive\src\driver_vive.c	1172	
Error	C2065	'ssize_t': undeclared identifier	driver_vive	I:\Libraries\libsurvive\src\driver_vive.c	1682	
Error	C2146	syntax error: missing ';' before identifier 'cnt'	driver_vive	I:\Libraries\libsurvive\src\driver_vive.c	1682	
Error	C2065	'cnt': undeclared identifier	driver_vive	I:\Libraries\libsurvive\src\driver_vive.c	1682	
Error	C2065	'cnt': undeclared identifier	driver_vive	I:\Libraries\libsurvive\src\driver_vive.c	1684	
Error	C2065	'cnt': undeclared identifier	driver_vive	I:\Libraries\libsurvive\src\driver_vive.c	1686	
Error	C2065	'cnt': undeclared identifier	driver_vive	I:\Libraries\libsurvive\src\driver_vive.c	1697	
Error	LNK1104	cannot open file 'Debug\survive.lib'	disambiguator_charles	I:\Libraries\libsurvive\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	poser_dummy	I:\Libraries\libsurvive\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	poser_charlesrefine	I:\Libraries\libsurvive\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	poser_barycentric_svd	I:\Libraries\libsurvive\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	poser_epnp	I:\Libraries\libsurvive\build\LINK	1	
Error	C1083	Cannot open include file: 'openvr/openvr_driver.h': No such file or directory	driver_survive	I:\Libraries\libsurvive\tools\openvr_driver\openvr_driver.cc	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	poser_mpfit	I:\Libraries\libsurvive\build\LINK	1	
Error	LNK1104	cannot open file '..\..\Debug\survive.lib'	test_replays	I:\Libraries\libsurvive\build\src\test_cases\LINK	1	
Error	C2061	syntax error: identifier 'read_light_data'	survive_tests	I:\Libraries\libsurvive\src\driver_vive.c	1172	
Error	C2059	syntax error: ';'	survive_tests	I:\Libraries\libsurvive\src\driver_vive.c	1172	
Error	C2059	syntax error: '<parameter-list>'	survive_tests	I:\Libraries\libsurvive\src\driver_vive.c	1172	
Error	C2065	'ssize_t': undeclared identifier	survive_tests	I:\Libraries\libsurvive\src\driver_vive.c	1682	
Error	C2146	syntax error: missing ';' before identifier 'cnt'	survive_tests	I:\Libraries\libsurvive\src\driver_vive.c	1682	
Error	C2065	'cnt': undeclared identifier	survive_tests	I:\Libraries\libsurvive\src\driver_vive.c	1682	
Error	C2065	'cnt': undeclared identifier	survive_tests	I:\Libraries\libsurvive\src\driver_vive.c	1684	
Error	C2065	'cnt': undeclared identifier	survive_tests	I:\Libraries\libsurvive\src\driver_vive.c	1686	
Error	C2065	'cnt': undeclared identifier	survive_tests	I:\Libraries\libsurvive\src\driver_vive.c	1697	
Error	LNK1104	cannot open file 'Debug\survive.lib'	data_recorder	I:\Libraries\libsurvive\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	api_example	I:\Libraries\libsurvive\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	survive-cli	I:\Libraries\libsurvive\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	sensors-readout	I:\Libraries\libsurvive\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	simple_pose_test	I:\Libraries\libsurvive\build\LINK	1

jdavidberger, 2019-09-20:
Apparently some versions of vs have ssize_t and some don't. Should be fixed now.

mgavornik, 2019-09-21:
Hi ty, i still have the following errors: 

Error (active)	E0029	expected an expression	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	56	
Error (active)	E1696	cannot open source file "pcap/usb.h"	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	13	
Error (active)	E1696	cannot open source file "pcap.h"	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	14	
Error (active)	E1696	cannot open source file "libusb-1.0/libusb.h"	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	18	
Error (active)	E1696	cannot open source file "pcap/usb.h"	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	19	
Error (active)	E1696	cannot open source file "zlib.h"	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	20	
Error (active)	E0020	identifier "pcap_t" is undefined	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	62	
Error (active)	E0020	identifier "pcap_dumper_t" is undefined	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	64	
Error (active)	E0020	identifier "PCAP_ERRBUF_SIZE" is undefined	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	66	
Error (active)	E0393	pointer to incomplete class type is not allowed	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	168	
Error (active)	E0393	pointer to incomplete class type is not allowed	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	168	
Error (active)	E0393	pointer to incomplete class type is not allowed	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	168	
Error (active)	E0393	pointer to incomplete class type is not allowed	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	169	
Error (active)	E0393	pointer to incomplete class type is not allowed	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	173	
Error (active)	E0393	pointer to incomplete class type is not allowed	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	173	
Error (active)	E0393	pointer to incomplete class type is not allowed	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	173	
Error (active)	E0393	pointer to incomplete class type is not allowed	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	174	
Error (active)	E0070	incomplete type is not allowed	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	226	
Error (active)	E0070	incomplete type is not allowed	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	229	
Error (active)	E0020	identifier "libusb_context" is undefined	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	255	
Error (active)	E0020	identifier "context" is undefined	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	255	
Error (active)	E0020	identifier "libusb_device" is undefined	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	256	
Error (active)	E0020	identifier "list" is undefined	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	256	
Error (active)	E0020	identifier "ssize_t" is undefined	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	258	
Error (active)	E0020	identifier "device" is undefined	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	268	
Error (active)	E0070	incomplete type is not allowed	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	269	
Error (active)	E0070	incomplete type is not allowed	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	275	
Error (active)	E0070	incomplete type is not allowed	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	276	
Error (active)	E0028	expression must have a constant value	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	318	
Error (active)	E0029	expected an expression	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	358	
Error (active)	E0020	identifier "pcap_usb_header_mmapped" is undefined	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	372	
Error (active)	E0084	invalid combination of type specifiers	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	447	
Error (active)	E0110	expected either a definition or a tag name	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	447	
Error (active)	E0029	expected an expression	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	448	
Error (active)	E0393	pointer to incomplete class type is not allowed	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	463	
Error (active)	E0029	expected an expression	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	465	
Error (active)	E0020	identifier "PCAP_ERROR" is undefined	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	482	
Error (active)	E0020	identifier "PCAP_ERROR_BREAK" is undefined	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	484	
Error (active)	E0020	identifier "PCAP_ERRBUF_SIZE" is undefined	driver_usbmon	i:\Libraries\libsurvive\src\driver_usbmon.c	541	
Error	C1083	Cannot open include file: 'pcap/usb.h': No such file or directory	driver_usbmon	I:\Libraries\libsurvive\src\driver_usbmon.c	13	
Error	C1083	Cannot open include file: 'openvr/openvr_driver.h': No such file or directory	driver_survive	I:\Libraries\libsurvive\tools\openvr_driver\openvr_driver.cc	1	

After copying the missing header files to the redist folder and applying some windows related fixes to driver_usbmon.c, i still have the following errros:

Error	C2466	cannot allocate an array of constant size 0	driver_usbmon	I:\Libraries\libsurvive\src\driver_usbmon.c	317	
Error	C2059	syntax error: '}'	driver_usbmon	I:\Libraries\libsurvive\src\driver_usbmon.c	55	
[... cut ...]
Error	C2059	syntax error: '}'	driver_usbmon	I:\Libraries\libsurvive\src\driver_usbmon.c	357	
Error	LNK1104	cannot open file 'lapacke.lib'	driver_survive	I:\Libraries\libsurvive\tools\openvr_driver\LINK	1	

After manually fixing these errors and adding the missing lib files to the VS solution, i got some unresolved external symbol errors. 

Unfortunately my C/C++ expertise is only rudimentary. It looks to me like the make file is missing the configuration settings for the missing header files and libraries.

jdavidberger, 2019-09-21:
Do you have libpcap installed somewhere? It should only be trying to build that if you do. 

Comment out lines 134-140 in CMakeLists.txt, reconfigure and rebuild. Let me know if that fixes it. I've never tried to build with pcap support in windows because I wasn't entirely clear how to install it.

mgavornik, 2019-09-21:
I have downloaded Npcap SDK 1.03 

> https://nmap.org/npcap/

which contains prebuild lib and header files. I have linked the lib file through cmake and copied the header files to \redist.

[image]

I removed the lines from CMakeLists.txt reconfigured and rebuilt. But the error is still there i think there is something wrong the the linker input

[image]

If i correct 

- empty initializers struct `pcap_stat stats = {};` with `struct pcap_stat stats =  {.ps_recv=0,.ps_drop=0,.ps_ifdrop=0,.ps_capt=0,.ps_sent=0};`
- replace `int device_cnts[DEVICES_CNT];` with `int device_cnts[6];`
- lapacke.lib to ..\..\packages\lapacke.0.1.0\lib\native\lib\x64\liblapacke.lib
- lapack.lib to packages\lapacke.0.1.0\lib\native\lib\x64\liblapack.lib
- and delete cblas.lib

i get unresolved external symbols: 

> Error	LNK1120	18 unresolved externals	driver_usbmon	I:\Libraries\libsurvive\Debug\plugins\driver_usbmon.dll	1	
> Error	LNK2019	unresolved external symbol _pcap_open_live referenced in function _DriverRegUSBMon	driver_usbmon	I:\Libraries\libsurvive\driver_usbmon.obj	1	
> Error	LNK2019	unresolved external symbol _pcap_hopen_offline referenced in function _DriverRegUSBMon	driver_usbmon	I:\Libraries\libsurvive\driver_usbmon.obj	1	
> Error	LNK2019	unresolved external symbol _pcap_close referenced in function _usbmon_close	driver_usbmon	I:\Libraries\libsurvive\driver_usbmon.obj	1	
> Error	LNK2019	unresolved external symbol _pcap_next_ex referenced in function _pcap_thread_fn	driver_usbmon	I:\Libraries\libsurvive\driver_usbmon.obj	1	
> Error	LNK2019	unresolved external symbol _pcap_breakloop referenced in function _usbmon_close	driver_usbmon	I:\Libraries\libsurvive\driver_usbmon.obj	1	
> Error	LNK2019	unresolved external symbol _pcap_stats referenced in function _usbmon_close	driver_usbmon	I:\Libraries\libsurvive\driver_usbmon.obj	1	
> Error	LNK2019	unresolved external symbol _pcap_geterr referenced in function _pcap_thread_fn	driver_usbmon	I:\Libraries\libsurvive\driver_usbmon.obj	1	
> Error	LNK2019	unresolved external symbol _pcap_dump_hopen referenced in function _DriverRegUSBMon	driver_usbmon	I:\Libraries\libsurvive\driver_usbmon.obj	1	
> Error	LNK2019	unresolved external symbol _pcap_dump_close referenced in function _usbmon_close	driver_usbmon	I:\Libraries\libsurvive\driver_usbmon.obj	1	
> Error	LNK2019	unresolved external symbol _pcap_dump referenced in function _pcap_thread_fn	driver_usbmon	I:\Libraries\libsurvive\driver_usbmon.obj	1	
> Error	LNK2019	unresolved external symbol _survive_data_cb referenced in function _pcap_thread_fn	driver_usbmon	I:\Libraries\libsurvive\driver_usbmon.obj	1	
> Error	LNK2019	unresolved external symbol _libusb_init@4 referenced in function _get_usb_info_from_libusb	driver_usbmon	I:\Libraries\libsurvive\driver_usbmon.obj	1	
> Error	LNK2019	unresolved external symbol _libusb_exit@4 referenced in function _get_usb_info_from_libusb	driver_usbmon	I:\Libraries\libsurvive\driver_usbmon.obj	1	
> Error	LNK2019	unresolved external symbol _libusb_get_device_list@8 referenced in function _get_usb_info_from_libusb	driver_usbmon	I:\Libraries\libsurvive\driver_usbmon.obj	1	
> Error	LNK2019	unresolved external symbol _libusb_free_device_list@8 referenced in function _get_usb_info_from_libusb	driver_usbmon	I:\Libraries\libsurvive\driver_usbmon.obj	1	
> Error	LNK2019	unresolved external symbol _libusb_get_device_descriptor@8 referenced in function _get_usb_info_from_libusb	driver_usbmon	I:\Libraries\libsurvive\driver_usbmon.obj	1	
> Error	LNK2019	unresolved external symbol _libusb_get_bus_number@4 referenced in function _get_usb_info_from_libusb	driver_usbmon	I:\Libraries\libsurvive\driver_usbmon.obj	1	
> Error	LNK2019	unresolved external symbol _libusb_get_device_address@4 referenced in function _get_usb_info_from_libusb	driver_usbmon	I:\Libraries\libsurvive\driver_usbmon.obj	1

jdavidberger, 2019-09-21:
Sorry you went through this trouble -- libpcap is used only for usbmon; which lets it process the same data that openvr sees and is mostly just a testing tool to make sure that we can process the USB data. I'm guessing you just want to run the tool normally; where it talks directly to the USB devices? 

I don't think driver_usbmon will work in general on windows -- I might be wrong but I think it doesn't expose usb events in general in windows.

mgavornik, 2019-09-21:
Well np :) thanks for you help! 

Honestly i just wanted to run libsurvive and play around with it, i don't understand yet how it works. 

What i actually need is to record imu and optical tracking data to my test my own kalman filter implementation. Is that possible without usbmon ? 

I moved to linux by now, on ubuntu i can build libsurvive but running any of the cli apps is throwing errors :)

I will open up a separate issue for that.

fcampos, 2020-01-27:
Hello,
Did anyone find a solution to build libsurvive on windows? Using cmake+Visual Studio 2017I also get a bunch of errors:

Severity	Code	Description	Project	File	Line	Suppression State
Error	MSB6006	"cmd.exe" exited with code 9009.	nuget-restore	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	209	
Error	C2059	syntax error: '}'	survive	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\src\survive_optimizer.c	313	
Error	C2059	syntax error: '}'	survive	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\src\survive_optimizer.c	317	
Error	C2059	syntax error: '}'	survive	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\src\survive_optimizer.c	334	
Error	C2065	'LINE_MAX': undeclared identifier	survive	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\src\survive_optimizer.c	425	
Error	C2057	expected constant expression	survive	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\src\survive_optimizer.c	425	
Error	C2466	cannot allocate an array of constant size 0	survive	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\src\survive_optimizer.c	425	
Error	C2059	syntax error: '}'	survive	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\src\survive_optimizer.c	425	
Error	C2065	'survive_reproject_gen2_model': undeclared identifier	survive	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\src\survive_optimizer.c	431	
Error	C2065	'LINE_MAX': undeclared identifier	survive	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\src\survive_optimizer.c	438	
Error	C2059	syntax error: '}'	survive	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\src\survive_process.c	204	
Error	C2059	syntax error: '}'	survive	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\src\survive_str.c	26	
Error	C2059	syntax error: '}'	survive	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\src\test_cases\str.c	5	
Error	LNK1104	cannot open file 'Debug\survive.lib'	disambiguator_turvey	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	poser_sba	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	disambiguator_charles	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	poser_imu	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	poser_epnp	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	poser_mpfit	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	poser_dummy	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	poser_charlesrefine	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	driver_dummy	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	poser_barycentric_svd	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	disambiguator_statebased	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\build\LINK	1	
Error	C2236	unexpected token 'struct'. Did you forget a ';'?	driver_vive	c:\users\franciscomateus\desktop\libsurvive-master\libsurvive-master\src\driver_vive.h	56	
Error	C2332	'struct': missing tag name	driver_vive	c:\users\franciscomateus\desktop\libsurvive-master\libsurvive-master\src\driver_vive.h	56	
Error	C2236	unexpected token 'struct'. Did you forget a ';'?	driver_vive	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\src\driver_vive.c	291	
Error	C2332	'struct': missing tag name	driver_vive	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\src\driver_vive.c	291	
Error	C2055	expected formal parameter list, not a type list	driver_vive	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\src\driver_vive.c	291	
Error	C2332	'struct': missing tag name	driver_vive	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\src\driver_vive.c	292	
Error	C2059	syntax error: 'type'	driver_vive	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\src\driver_vive.c	292	
Error	C2046	illegal case	driver_vive	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\src\driver_vive.c	295	
Error	C2046	illegal case	driver_vive	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\src\driver_vive.c	297	
Error	C2046	illegal case	driver_vive	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\src\driver_vive.c	299	
Error	C2046	illegal case	driver_vive	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\src\driver_vive.c	301	
Error	C2046	illegal case	driver_vive	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\src\driver_vive.c	303	
[... cut ...]
Error	LNK1104	cannot open file 'Debug\survive.lib'	data_recorder	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	api_example	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	survive-cli	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\survive.lib'	simple_pose_test	C:\Users\FranciscoMateus\Desktop\libsurvive-master\libsurvive-master\build\LINK	1	

Many thanks

mgavornik, 2020-01-29:
Yes, i was able to build it but it didn't run properly under windows. 

I was getting OOTX errors. 

I can check on the weekend what the changes where and get back to you.

fcampos, 2020-01-29:
So, did you eventually get it to run on Windows?
Thanks for the help.

mgavornik, 2020-01-29:
Yes it ran, but did not work. I moved for data gathering to linux, running playbacks on windows worked.

mgavornik, 2020-02-01:
It compiles and runs now under windows. I can't push them to a branch because of insufficient rights so here is my git diff for you to make it work : 

diff --git a/sensors-readout.c b/sensors-readout.c
index cb25250..78578a1 100644
--- a/sensors-readout.c
+++ b/sensors-readout.c
@@ -36,8 +36,8 @@ struct sensor_time_stats {
 	size_t hz_count;
 	survive_timecode hz_start;
 };
-struct sensor_stats stats[32][NUM_GEN2_LIGHTHOUSES][SENSORS_PER_OBJECT][2] = {};
-struct sensor_time_stats time_stats[32][SENSORS_PER_OBJECT] = {};
+struct sensor_stats stats[32][NUM_GEN2_LIGHTHOUSES][SENSORS_PER_OBJECT][2] = {0};
+struct sensor_time_stats time_stats[32][SENSORS_PER_OBJECT] = {0};
 
 void process_reading(int i, int lh, int sensor, int axis, FLT angle) {
 	struct sensor_stats *s = &stats[i][lh][sensor][axis];
@@ -118,7 +118,7 @@ char *new_str(const char *s) {
 	return rtn;
 }
 
-char *lines[10] = {};
+char *lines[10];
 size_t lines_idx = 0;
 
 int window_rows = -1, window_cols = -1;
diff --git a/src/driver_simulator.c b/src/driver_simulator.c
index d89e1b5..37cf071 100644
--- a/src/driver_simulator.c
+++ b/src/driver_simulator.c
@@ -1,12 +1,13 @@
 // All MIT/x11 Licensed Code in this file may be relicensed freely under the GPL
 // or LGPL licenses.
 
-#include "math.h"
+//#include "math.h"
 #include "os_generic.h"
 #include "survive_config.h"
 #include "survive_default_devices.h"
 #include "survive_reproject_gen2.h"
 #include "survive_str.h"
+#include "survive_str.c"
 #include <assert.h>
 #include <json_helpers.h>
 #include <math.h>
@@ -16,6 +17,7 @@
 #include <survive.h>
 #include <survive_reproject.h>
 
+
 STATIC_CONFIG_ITEM(Simulator_DRIVER_ENABLE, "simulator", 'i', "Load a Simulator driver for testing.", 0);
 STATIC_CONFIG_ITEM(Simulator_TIME, "simulator-time", 'f', "Seconds to run simulator for.", 0.0);
 
@@ -272,8 +274,8 @@ int DriverRegSimulator(SurviveContext *ctx) {
 	sp->velocity.AxisAngleRot[1] = .5;
 	sp->velocity.AxisAngleRot[2] = .5;
 
-	cstring cfg = {};
-	cstring loc = {}, nor_buf = {};
+	cstring cfg = {0,0,0};
+	cstring loc = {0, 0, 0}, nor_buf = {0, 0, 0};
 
 	FLT r = .1;
 	srand(42);
diff --git a/src/driver_vive.c b/src/driver_vive.c
index 214a566..02eb92b 100755
--- a/src/driver_vive.c
+++ b/src/driver_vive.c
@@ -288,8 +288,8 @@ const struct DeviceInfo KnownDeviceTypes[] = {
 typedef struct SurviveUSBInterface SurviveUSBInterface;
[... 116 lines cut ...]
--- a/src/survive_str.c
+++ b/src/survive_str.c
@@ -23,7 +23,8 @@ char *str_increase_by(cstring *str, size_t len) {
 }
 void str_free(cstring *str) {
 	free(str->d);
-	*str = (cstring){};
+	cstring newStr = {0,0,0};
+	*str = newStr;
 }
 void str_append(cstring *str, const char *add) {
 	size_t add_len = strlen(add);
diff --git a/src/test_cases/str.c b/src/test_cases/str.c
index 37ae4a7..44f302f 100644
--- a/src/test_cases/str.c
+++ b/src/test_cases/str.c
@@ -2,7 +2,7 @@
 #include "test_case.h"
 
 TEST(SurviveUtils, Str) {
-	cstring test = {};
+	cstring test = {0,0,0};
 	str_append(&test, "012345");
 	ASSERT_EQ(test.d[test.length], 0);
 	ASSERT_EQ(test.length, 6);

fcampos, 2020-02-02:
Many thanks, I will give it a try!

BenWoodford, 2020-02-04:
@mgavornik if you fork the repo, you can push your changes to your own branch and submit a PR with your fixes.

mgavornik, 2020-02-04:
@BenWoodford  Thanks for the info, in the case i will do that.

jdavidberger, 2020-02-12:
I've fixed the windows build and also added it to ci so it won't stay broken again. Sorry about the inconvenience!

closed 2020-02-12 by jdavidberger