collabora/libsurvive #154: segfaults while initializing survive

sharif1093, 2019-10-05:
I am experiencing a segfault after running any of the executable commands of the `libsurvive`. The logs indicate a missing config/plugin; however, I am not sure what I am exactly missing.

**System**

OS: `Ubuntu 16.04`
NVIDIA Driver: `384.130`

**Steps followed**

```bash
# `udev` rule is added.
# `AllowHDM` is set to `yes` for the Xorg.

git clone https://github.com/cnlohr/libsurvive.git
cd libsurvive
cmake .
make

./survive-cli
```

**`Screen Output`**

```bash
Error loading lib/plugins/driver_usbmon.so: lib/plugins/driver_usbmon.so: undefined symbol: survive_data_cb
Info: Successfully enumerated HMD (1) 0bb4:2c87
Info: Successfully enumerated HMD 0bb4:2c87
Info: Successfully enumerated HMD IMU & LH (2) 28de:2000
Info: Successfully enumerated HMD IMU & LH 28de:2000
Info: Successfully enumerated Watchman (1) 28de:2101
Info: Successfully enumerated Watchman 28de:2101
Info: Successfully enumerated Watchman (1) 28de:2101
Info: Successfully enumerated Watchman 28de:2101
Info: Successfully enumerated Watchman (1) 28de:2101
Info: Successfully enumerated Watchman 28de:2101
Info: Successfully enumerated Watchman (1) 28de:2101
Info: Successfully enumerated Watchman 28de:2101
Info: Adding tracked object HMD from HTC
Info: Adding tracked object WM0 from HTC
Info: Adding tracked object WM1 from HTC
Info: Adding tracked object WM2 from HTC
Info: Adding tracked object WM3 from HTC
Info: All enumerated devices attached.
Info: Powered unit on.
Info: Got config data length 3496
[1]    5611 segmentation fault (core dumped)  ./simple_pose_test
```

**`gdb backtrace` results**

```bash
#0  __GI__IO_fwrite (buf=buf@entry=0x7ffffffec4a0, size=size@entry=16643, 
    count=count@entry=1, fp=fp@entry=0x0) at iofwrite.c:37
#1  0x00007fffdf801073 in survive_get_config (
    config=config@entry=0x7fffffffc560, usbInfo=usbInfo@entry=0x617990, 
    send_extra_magic=<optimized out>, sv=0x617520, iface=0)
    at src/driver_vive.c:962
#2  0x00007fffdf80269b in LoadConfig (iface=0, sv=0x617520, usbInfo=0x617990)
    at src/driver_vive.c:2320
#3  DriverRegHTCVive (ctx=0x613370) at src/driver_vive.c:2379
#4  0x00007ffff7ba4936 in callDriver (
    buffer=0x7fffffffcb10 "Loaded drivers: ", 
    DriverName=0x7ffff7bcbe1e "DriverRegHTCVive", ctx=0x613370)
    at src/survive.c:460
#5  survive_startup (ctx=ctx@entry=0x613370) at src/survive.c:515
#6  0x00007ffff7ba6568 in survive_poll (ctx=0x613370) at src/survive.c:723
#7  0x00000000004017a8 in main (argc=<optimized out>, argv=<optimized out>)
    at simple_pose_test.c:229
```

**`valgrind` output**
```bash
==8142== Memcheck, a memory error detector
==8142== Copyright (C) 2002-2015, and GNU GPL'd, by Julian Seward et al.
==8142== Using Valgrind-3.11.0 and LibVEX; rerun with -h for copyright info
==8142== Command: ./survive-cli
==8142== 
==8142== Warning: ignored attempt to set SIGKILL handler in sigaction();
==8142==          the SIGKILL signal is uncatchable
Error loading /home/sharif/libsurvive/plugins/driver_usbmon.so: /home/sharif/libsurvive/plugins/driver_usbmon.so: undefined symbol: survive_data_cb
Info: Successfully enumerated HMD (1) 0bb4:2c87
Info: Successfully enumerated HMD 0bb4:2c87
Info: Successfully enumerated HMD IMU & LH (2) 28de:2000
Info: Successfully enumerated HMD IMU & LH 28de:2000
Info: Successfully enumerated Watchman (1) 28de:2101
Info: Successfully enumerated Watchman 28de:2101
Info: Successfully enumerated Watchman (1) 28de:2101
Info: Successfully enumerated Watchman 28de:2101
Info: Successfully enumerated Watchman (1) 28de:2101
Info: Successfully enumerated Watchman 28de:2101
Info: Successfully enumerated Watchman (1) 28de:2101
Info: Successfully enumerated Watchman 28de:2101
Info: Adding tracked object HMD from HTC
Info: Adding tracked object WM0 from HTC
Info: Adding tracked object WM1 from HTC
Info: Adding tracked object WM2 from HTC
Info: Adding tracked object WM3 from HTC
Info: All enumerated devices attached.
Info: Powered unit on.
Info: Got config data length 3496
==8142== Invalid read of size 4
==8142==    at 0x53236F8: fwrite (iofwrite.c:37)
==8142==    by 0x1CCB93EB: survive_get_config (in /home/sharif/libsurvive/plugins/driver_vive.so)
==8142==    by 0x1CCBCC77: LoadConfig (in /home/sharif/libsurvive/plugins/driver_vive.so)
==8142==    by 0x1CCBD066: DriverRegHTCVive (in /home/sharif/libsurvive/plugins/driver_vive.so)
==8142==    by 0x4E59C6F: callDriver (in /home/sharif/libsurvive/libsurvive.so)
==8142==    by 0x4E5A03B: survive_startup (in /home/sharif/libsurvive/libsurvive.so)
==8142==    by 0x400E45: main (in /home/sharif/libsurvive/survive-cli)
==8142==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==8142== 
==8142== 
==8142== Process terminating with default action of signal 11 (SIGSEGV)
==8142==  Access not within mapped region at address 0x0
==8142==    at 0x53236F8: fwrite (iofwrite.c:37)
==8142==    by 0x1CCB93EB: survive_get_config (in /home/sharif/libsurvive/plugins/driver_vive.so)
==8142==    by 0x1CCBCC77: LoadConfig (in /home/sharif/libsurvive/plugins/driver_vive.so)
==8142==    by 0x1CCBD066: DriverRegHTCVive (in /home/sharif/libsurvive/plugins/driver_vive.so)
==8142==    by 0x4E59C6F: callDriver (in /home/sharif/libsurvive/libsurvive.so)
==8142==    by 0x4E5A03B: survive_startup (in /home/sharif/libsurvive/libsurvive.so)
==8142==    by 0x400E45: main (in /home/sharif/libsurvive/survive-cli)
==8142==  If you believe this happened as a result of a stack
==8142==  overflow in your program's main thread (unlikely but
==8142==  possible), you can try to increase the size of the
==8142==  main thread stack using the --main-stacksize= flag.
==8142==  The main thread stack size used in this run was 8388608.
==8142== 
==8142== HEAP SUMMARY:
==8142==     in use at exit: 205,586 bytes in 290 blocks
==8142==   total heap usage: 5,136 allocs, 4,846 frees, 3,190,983 bytes allocated
==8142== 
==8142== LEAK SUMMARY:
==8142==    definitely lost: 0 bytes in 0 blocks
==8142==    indirectly lost: 0 bytes in 0 blocks
==8142==      possibly lost: 12,125 bytes in 41 blocks
==8142==    still reachable: 193,461 bytes in 249 blocks
==8142==         suppressed: 0 bytes in 0 blocks
==8142== Rerun with --leak-check=full to see details of leaked memory
==8142== 
==8142== For counts of detected and suppressed errors, rerun with: -v
==8142== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
[1]    8142 killed     valgrind ./survive-cli
```

sharif1093, 2019-10-05:
I tried with `sudo` as in: `sudo ./survive-cli` and the segfault issue was resolved.

Now I have another issue, I get the following warnings and no data in practice (screen output is only partially shown):

```bash
Error loading /home/sharif/libsurvive/plugins/driver_usbmon.so: /home/sharif/libsurvive/plugins/driver_usbmon.so: undefined symbol: survive_data_cb
Info: Successfully enumerated HMD (1) 0bb4:2c87
Info: Successfully enumerated HMD 0bb4:2c87
Info: Successfully enumerated HMD IMU & LH (2) 28de:2000
Info: Successfully enumerated HMD IMU & LH 28de:2000
Info: Successfully enumerated Watchman (1) 28de:2101
Info: Successfully enumerated Watchman 28de:2101
Info: Successfully enumerated Watchman (1) 28de:2101
Info: Successfully enumerated Watchman 28de:2101
Info: Successfully enumerated Watchman (1) 28de:2101
Info: Successfully enumerated Watchman 28de:2101
Info: Successfully enumerated Watchman (1) 28de:2101
Info: Successfully enumerated Watchman 28de:2101
Info: Adding tracked object HMD from HTC
Info: Adding tracked object WM0 from HTC
Info: Adding tracked object WM1 from HTC
Info: Adding tracked object WM2 from HTC
Info: Adding tracked object WM3 from HTC
Info: All enumerated devices attached.
Info: Powered unit on.
Info: Got config data length 3496
Info: HMD is treated as HMD device
Info: Read config for HMD
Info: Got config data length 2207
Info: Read config for WM0
Info: Got config data length 2211
Info: Read config for WM1
Info: Got config data length 1938
Info: Read config for WM2
Info: Got config data length 1933
Info: Read config for WM3
Info: Attaching Mainboard(0x81) for HMD
Info: Attaching IMU(0x81) for HMD
Info: Attaching Lightcap(0x82) for HMD
Info: Attaching IMU/Lightcap/Buttons(0x81) for WM0
Info: Attaching IMU/Lightcap/Buttons(0x81) for WM1
Info: Attaching IMU/Lightcap/Buttons(0x81) for WM2
Info: Attaching IMU/Lightcap/Buttons(0x81) for WM3
Info: Loaded drivers: HTCVive
Info: Detected LH gen 2 system.
Warning: Unknown gen two event 0x9C [Time:018E] [Payload: 2f d8 a8 92 ] <<ABORT FURTHER READ>>
Warning: Unknown gen two event 0x20 [Time:0194] [Payload: a8 d8 1f 97 45 0f e6 05 f3 9d 06 c5 02 e5 01 c4 06 ab 1b dc 96 ] <<ABORT FURTHER READ>>
Warning: Invalid sensor 255 detected hit (27)
Warning: Unknown gen two event 0x20 [Time:9A3D] [Payload: 48 ed 1f 90 45 0f e9 05 ef 9f 06 c3 02 d7 01 cc 06 a5 88 cc 40 ] <<ABORT FURTHER READ>>
Warning: Invalid sensor 255 detected hit (29)
Warning: Invalid sensor 255 detected hit (25)
Warning: Invalid sensor 255 detected hit (25)
Warning: Unknown gen two event 0x38 [Time:9AA6] [Payload: 9c 51 41 9e 85 04 99 13 84 a6 8a 00 01 a2 2f e5 6a 07 95 03 39 d7 92 ] <<ABORT FURTHER READ>>
Warning: Unknown status event 0x4A [Time:44F4] [Payload: 45 25 81 a4 a9 f7 12 b1 91 99 99 8e ef 03 b9 02 a9 e1 e6 06 6e 0f f1 ] <<ABORT FURTHER READ>>
Warning: Read light data error -1   [Time:44AE] [Payload: 20 80 50 8b 06 c8 0c e3 04 e1 10 a7 05 86 1d 10 cd 2e e4 05 64 ]
Warning: Read light data error -1   [Time:44B4] [Payload: 00 87 05 f0 63 0c e3 2a 80 1e 6a ]
Warning: Unknown gen two event 0xA0 [Time:44F4] [Payload: e3 0b 9b 8d 0d 35 16 f1 ] <<ABORT FURTHER READ>>
Warning: Unknown gen two event 0x20 [Time:9A54] [Payload: 48 87 18 ed 4c 0f f0 05 ea 9c 06 c5 02 e4 01 c7 06 a9 cc 36 59 ] <<ABORT FURTHER READ>>
Warning: Read light data error -1   [Time:44BD] [Payload: b1 51 08 f2 26 ef 72 0a ed 04 a2 02 c6 03 95 78 0b 81 23 a8 09 52 ]
Info: WM2 button event type: 3 id: 0 axis1id: 2 axis1val 16432 axis2id: 3 axis2val 32800
Warning: Unknown gen two event 0x08 [Time:9ABE] [Payload: a8 30 80 04 cd 01 cc 02 95 0e ae 02 f2 4f 0e dc 1b 0d 0f af ] <<ABORT FURTHER READ>>
Warning: Unknown gen two event 0xA2 [Time:450B] [Payload: 48 cf 03 f2 07 a6 93 90 0d 3c 80 09 ] <<ABORT FURTHER READ>>
Info: WM3 button event type: 3 id: 0 axis1id: 2 axis1val 2176 axis2id: 3 axis2val 43328
Warning: Read light data error -5   [Time:44C6] [Payload: 17 f1 00 01 f4 26 df 77 0a 9e 01 a2 04 aa 0d 66 67 ]
Warning: Read light data error -1   [Time:44CB] [Payload: 48 b0 08 e5 2a c0 14 06 b6 05 e9 0c df 05 ef 0c e5 04 f0 62 6c ]
Warning: Unknown status event 0x4A [Time:9A69] [Payload: 5a 33 41 48 ac 33 d6 4e 08 a3 03 b1 d9 f0 02 e2 f7 88 02 e9 ce 69 ] <<ABORT FURTHER READ>>
Warning: Read light data error -1   [Time:4511] [Payload: 80 51 49 eb 05 94 85 06 d3 03 81 05 98 12 e4 05 f3 22 0e ]
Warning: Unknown gen two event 0x91 [Time:44DA] [Payload: 06 1a c4 84 ] <<ABORT FURTHER READ>>
Warning: Unknown gen two event 0x20 [Time:9A72] [Payload: 38 de 1f 97 45 0f f1 05 e8 a0 06 c2 02 e6 01 c8 06 a6 08 a1 71 ] <<ABORT FURTHER READ>>
Warning: Invalid sensor 255 detected hit (27)
Warning: Read light data error -1   [Time:9AD2] [Payload: af f7 01 e3 01 c2 02 31 bb c3 ]
Info: WM2 button event type: 1 id: 24 axis1id: 0 axis1val 0 axis2id: 0 axis2val 0
Info: WM2 button event type: 1 id: 3 axis1id: 0 axis1val 0 axis2id: 0 axis2val 0
Info: WM2 button event type: 1 id: 7 axis1id: 0 axis1val 0 axis2id: 0 axis2val 0
Warning: Unknown gen two event 0xA9 [Time:9ADB] [Payload: 99 08 cd 03 c8 0a d4 02 8a 02 ad 01 92 50 0d b9 27 fd 99 cd ] <<ABORT FURTHER READ>>
Warning: Read light data error -1   [Time:4528] [Payload: 51 49 b1 89 95 0a bb 04 9d ad 09 e8 05 97 85 06 d3 03 80 05 f5 80 26 ]
Warning: Light contains probable non-light data : 0xE7 [Time:4528] [Payload: e7 05 f2 8c 26 ]
Warning: Read light data error -1   [Time:4528] [Payload: e7 05 f2 8c 26 ]
Warning: Invalid sensor 255 detected hit (23)

.....
```

sharif1093, 2019-10-08:
I could solve linking problem against `driver_usbmon.so` by directly modifying the `Makefile`. So no more complaints about loading `driver_usbmon` plugin. However, the issue of unknown codes and no position reading still persists.

Any suggestions?

jdavidberger, 2020-03-04:
Those messages should be parsable now. 

I also fixed the issue with unwriteable config files; it now issues a warning.