collabora/libsurvive #155: reserved identifier violation

elfring, 2019-10-17:
I would like to point out that identifiers like “[`_SURVIVE_TYPES_H`](https://github.com/cnlohr/libsurvive/blob/1f81d56549a6076402a9ed9ae226426e5cbd59ea/include/libsurvive/survive_types.h#L1 "Update candidate")” and “[`__TORI_INCLUDES_H`](https://github.com/cnlohr/libsurvive/blob/e68b69a9a14aeb70988c52666e60502fe6f049ab/tools/lighthousefind_tori/tori_includes.h#L1 "Another update candidate")” [do not fit](https://www.securecoding.cert.org/confluence/display/c/DCL37-C.+Do+not+declare+or+define+a+reserved+identifier#DCL37-C.Donotdeclareordefineareservedidentifier-NoncompliantCodeExample%28IncludeGuard%29 "Do not use identifiers which are reserved for the compiler implementation.") to the expected naming convention of the C language standard.
Would you like to adjust your selection for unique names?

jdavidberger, 2019-10-17:
I'd be more inclined to just swap everything out with '#pragma once' since TCC ow supports that construct. 

Without a concrete build system where this is actually an issue; I don't think it's worth anyones time to go in and change at this moment. 

Possibly if whatever bot goes through and creates the issue just made a PR I'd merge that PR if it identified all similar constructs and did a reasonable job at using safe macro name.

closed 2019-10-17 by jdavidberger

jdavidberger, 2019-10-17:
I differentiate heavily between undefined runtime behavior and compile time behavior; especially if the compile time behavior is unlikely to the point of statistically impossible. 

I'm curious -- why do you have a bot that checks random repos for such an incredibly pedantic issue? I'd be shocked if this check, even running against all of github, would find an actual problem on any compiler configuration.

elfring, 2019-10-17:
I prefer defined software behaviour generally.
Would you like to reduce the tampering with the reserved name space?

cnlohr, 2019-10-22:
This is the first I have never heard of anyone specifically taking reservation with guarding header files with `_SURVIVE_TYPES_H` etc... I think this is the first time I have ever seen a website which complained about leading `_`'s I question if this is just someone who's gone off the deep end.

Not that I'm complaining @jdavidberger do as you see fit!