collabora/libsurvive #158: Consider IRC over Discord

ddevault, 2019-10-31:
Here with a heartfelt plea to ditch Discord for development chat, or at least set up a relay to IRC. IRC is where most other open source software development is going on: OpenXR/Monado, Mesa/OpenGL/DRI, Vulkan, OpenHMD... please join us! We want to talk with you!

jdavidberger, 2019-11-06:
I'm sorta confused. I'm in the Monado discord. Is there a big split between the two. 

I don't think it's advisable for us to ditch discord since it's what everyone is on currently.

ddevault, 2019-11-06:
I'm not there, and neither is @emersion, another Monado contributor. I don't know about anyone else. Here's the active list for #monado on freenode right now:

[image]

Don't you think it's bad that an open source project depends on not only a proprietary chat service, but one which requires proprietary chat clients?

jdavidberger, 2019-11-06:
> Don't you think it's bad that an open source project depends on not only a proprietary chat service, but one which requires proprietary chat clients?

Frankly, I don't see anything bad about using a proprietary tool as support infrastructure for an open source project. 

More important from my perspective is allowing for one unambiguous place to discuss libsurvive details. To that end there is no pragmatic difference between the two, and switching seems unnecessary. 

I hope that doesn't keep you from coming into the channel with questions or anything though!

closed 2019-11-06 by jdavidberger

ddevault, 2019-11-06:
>I hope that doesn't keep you from coming into the channel with questions or anything though!

It does. This is the pragmatic difference. There is no chance whatsoever that I will be joining the Discord channel, not out of spite, but because I simply do not use proprietary software. The choice to use Discord keeps people away from your channel.

emersion, 2019-11-06:
Hmm, as @ddevault said I'd like to discuss but I don't want a Discord account. If you really want to keep the Discord channel, I wonder if it would be possible to setup a bridge?

jdavidberger, 2019-11-06:
Is GitHub not a proprietary site? I don't understand the principled difference between GitHub and, say, the discord web client. 

Is there a bridge setup monando uses?

ddevault, 2019-11-06:
I'm not especially pleased about using GitHub, but at least it doesn't require proprietary client software to connect to it and it speaks protocols (HTTP, git) which are implemented by open source tools. Monado is hosted on freedesktop.org's GitLab instance.

Monado does not bridge Discord to IRC, but has a separate IRC channel. I don't know what the activity is like in Discord but the IRC channel sees plenty of development chat and discussion.

eric-schleicher, 2019-11-07:
FWIW, I think attempting to move the conversation away from discord would negatively impact participation and significantly affect visibility/discovery in an ongoing basis.

I'm a super nerd as well, but almost **never** use IRC proper because it's a pain in the ass and invariably ends up sounding like crickets.   Like it or not, the network effect of slack and discord is a net positive for people to find and most critically *STAY ENGAGED* these types of projects.

ddevault, 2019-11-07:
```
$ pwd
/var/lib/znc/.../log/freenode/#monado
$ wc -l 2019-11-07.log
215
$ date
Thu Nov  7 18:22:16 UTC 2019
```

ChristophHaag, 2019-11-07:
Discord is well usable with just the webinterface on their website, at least with desktop browsers.

issues with discord:

* no mobile webinterface, desktop webinterface does not really work on mobile browsers
* no open source android application
* does not allow open source clients (third party) to connect to their network
* text chat is not public to people until they make an account and join a discord "server"

eric-schleicher, 2019-11-07:
These are fair gripes.  But keeping the discussion between Monado and Libsurvive separate here, the critical mass is already established and there seems to be very little interest present to uproot.  

Aren't thing like matrix better for this on a forward-looking basis? they seem **cool** and easy to use; engage like slack/discord but check the open source boxes?

ddevault, 2019-11-07:
Matrix isn't really loved by most IRC users. IRC is where Linux graphics development takes place, that's just how it is. If not ditching Discord, at least a bridge would be greatly appreciated.

eric-schleicher, 2019-11-07:
disclosure:  I'm not in a position to do anything about it.   But which connector is the best one?  

I found:

- https://github.com/qaisjp/go-discord-irc
- https://github.com/zachbr/Dis4IRC
- https://github.com/reactiflux/discord-irc

Matrix looks like good bridging capabilities FWIW.
https://matrix.org/bridges/

ddevault, 2019-11-07:
Matrix can connect to IRC, but not vice-versa.

jdavidberger, 2019-11-07:
> IRC is where Linux graphics development takes place, that's just how it is. If not ditching Discord, at least a bridge would be greatly appreciated.

Which is great but libsurvive isn't a graphics development component. 

I find it hard to find motivation to set up a bridge for what seems like a very small audience which hasn't found the motivation to do the same in their community. 

I appreciate that discord isn't without its drawbacks. I think we can all agree no communications solution is. But switching is essentially not viable for the reasons @eric-schleicher gave re: an existing community. 

If you have something you'd like to discuss with the wider libsurvive audience, fire up the discord web client -- it's https and webrtc, as open as GitHub is in those terms. If that is really intolerable to discuss specific I think GitHub issues work well for that.

ddevault, 2019-11-07:
libinput is on IRC, too. We don't have the motivation to extend a bridge to a proprietary, centralized chat platform, no, and you won't see us adopt it soon. No two ways about it, Discord was a mistake. A bridge is a compromise to correct that mistake. IRC was not a mistake for us, so we don't need a bridge.

jdavidberger, 2019-11-07:
I think this discussion has run it's useful course. I'm going to go ahead and lock the thread. If anyone has technical issues or topics of discussion please don't hesitate to open a separate issue.