collabora/libsurvive #161: Semaphore deadlock in survive_data_cb

mhofmann85, 2019-11-25:
Hi,

I'm just experiencing a deadlock with the semaphores. I'm using Windows 10 with visual studio 2017 x64. But from source-code the problem seems to also exist in linus. (All semaphores are initialized with 0, which is the obvious intention.)

The Problem is the newly introduced locking in survive_data_cb from commit 46f681357049b98ef509f04d33866cd09a96c8b5. It seems like the semaphore is never unlocked.

I think, the intention is, to unlock the semaphore in survive.c line 745 and allow to execute all callbacks and stuff while sleeping. It seems, that there is now a new execution path through survive.c line 737 calling driverpolls, leading to the call to survive_data_cb.

Possible fix shift the release call from survive.c line 745 to line 735. At least it is working, please check if this screws up any other race condition prevention.

greeting
Michael

jdavidberger, 2019-11-25:
I just pushed something to master -- can you see if this fixes the issue?

I hadn't fixed HIDAPI to use the lock properly; and windows requires it. I don't have time this morning to check the windows build; but the latest work on HIDAPI under linux.

mhofmann85, 2019-11-26:
Thanks,
Tracking with USB- connected tracker is now working.

closed 2019-11-26 by mhofmann85