collabora/libsurvive #162: Tracker is not found using Dongle with Windows 10

mhofmann85, 2019-11-26:
Hi, again

I'm using Windows 10 with visual studio 2017 x64.  
My setup is a HTC-Vive - 1xTracker, 1x Lighthouse and 1x Dongle.  

Somehow I do net get the Tracker work with only dongle. Using USB-connection is working perfectly. This is the console-output I get.

```
$ simple_pose_test.exe -l 1 --v
Info: Available Posers:
Info:    BaryCentricSVD
Info:    CharlesRefine
Info:    Dummy
Info:    EPNP
Info:    IMU
Info:   *MPFIT
Info: Totals 6 Posers.
Info: Using 'MPFIT' for poser
Info: Available Disambiguators:
Info:    Charles
Info:    StateBased
Info:   *Turvey
Info: Totals 3 Disambiguators.
Info: Using 'Turvey' for disambiguator
Info: Successfully enumerated Watchman 28de:2101
Info: Adding tracked object WM0 from HTC
Info: Available Posers:
Info:    BaryCentricSVD
Info:    CharlesRefine
Info:    Dummy
Info:    EPNP
Info:    IMU
Info:   *MPFIT
Info: Totals 6 Posers.
Info: Using 'MPFIT' for poser
Info: All enumerated devices attached.
Warning: Could not turn on WM0(2) (-1/5 - )
Warning: Could not turn on WM0(2) (-1/5 - )
Info: Powered unit on.
Info: WM0 couldn't configure; probably turned off
Info: Removing tracked object WM0 from HTC
Info: Loaded drivers: HTCVive
Info: Position found for LH 0(f62a56f1)
```

mhofmann85, 2019-11-26:
possibly hat to pair device with steamVR.
is it somehow possible to pair new controllers with libSurvive?

jdavidberger, 2020-08-17:
Currently pairing new controllers isn't something libsurvive does. It is possible we might add it in the future but it'd require some data in terms of the USB protocol uses; and using steamvr right now seems like a decent work around.

closed 2020-08-17 by jdavidberger