collabora/libsurvive #163: MPFIT Errors

hamster1147, 2019-11-27:
Hey guys, trying out the repository for 
robotics. Trying to get things working and I haven't had much luck with MPFIT. BaryCentricSVD seems to work, but is extremely jittery.

Setup:
Operating System: Raspbian GNU/Linux 10 (buster)
Kernel: Linux 4.19.75-v7+
Architecture: arm
2x Lighthouse V2.0
2x Valve Index controllers wired
1x Vive Tracker V2.0

[libsurviveLog.txt](https://github.com/cnlohr/libsurvive/files/3898125/libsurviveLog.txt) - This is what my console outputs.
[config.txt](https://github.com/cnlohr/libsurvive/files/3898133/config.txt)

I got the websocket running with data_recorder and was able to see what the data looks like. All the devices seem to start where they are supposed to but then slowly vibrate in place until they violently fly off into space.

goldarte, 2019-11-27:
Hi! Try to use SBA poser, it's quite good. You can check our tryings here: https://github.com/sfalexrog/roslibsurvive

hamster1147, 2019-11-27:
I tried again with just the tracker and it did a better job. I feel like I'm stumbling through this but I'm getting somewhere.

jdavidberger, 2019-11-27:
SBA might not work so well with gen2 stuff. In general MPFIT is the thing that'll work best eventually; but gen2 stuff is a work in progress.

I just pushed a branch called gen2-wip. It's extremely experimental; but contains support for USB and RF gen2 devices. It tracks fairly well with knuckles but my tracker is sorta jittery with it right now. Still trying to fix that; multiple LH help a lot though.

jdavidberger, 2020-08-17:
Kalman changes should fix this

closed 2020-08-17 by jdavidberger