collabora/libsurvive #164: Custom tracker support?

linfan68, 2019-12-05:
Not sure if this is the right place to ask...
I'm looking for a solution that we can build custom tracker hardware (like a full body tracking suit).
I found HiveTracker working with LH1.0. 
Another project here using TS4231: https://hackaday.io/project/19570-htc-vive-lighthouse-custom-tracking
And Lots of LH2.0 protocol hacking working from here: https://github.com/cnlohr/esptracker

So I'm not sure where to start... Does this project suppose to support 3rd-party trackers in the future? Is this a good place to start if I want to build a 3rd party tracker working with LH2.0?

Thanks

jdavidberger, 2019-12-08:
In short, yes. The longer answer is there is a plugin system and a custom tracker would be exposed as a driver plugin for the project. 

If it's custom hardware -- trackers and firmware and all -- you'd need to write that driver. If you use one of the vive HDK's, it should work with the vive driver. Those drivers tend to be as complicated or as simple as whatever the hardware is doing. I'd look at driver_vive.c as a reference implementation, but the basics of it are that you'll call `handle_lightcap` with light data and it takes it from there. You can also call the `raw_imuproc` callback to propagate IMU messages.

linfan68, 2019-12-11:
> In short, yes. The longer answer is there is a plugin system and a custom tracker would be exposed as a driver plugin for the project.
>
> If it's custom hardware -- trackers and firmware and all -- [...]

Thank for the reply and those pointers. I'll start from there!

closed 2019-12-11 by linfan68