collabora/libsurvive #171: Compiling libsurvive for Android

BenWoodford, 2020-02-02:
Has anyone done it successfully in the past? Before I go down what is potentially a deep rabbit hole and also potentially a wild goose chase (honk)

My intent is to build something I can pop into  Unity as an Android plugin and wrap around. Which is probably an Android cross-compiled libsurvive.so file

Initial avenues to go down welcome, it’s been a long, long time since I played with CMake but initial findings tell me I need to set some environment variables to set it to Android as the target, give it the NDK, etc...

BenWoodford, 2020-05-11:
It may be possible for me to get a bounty put together for this if anyone can manage it. Ideally need it with a Unity wrapper though.

sparklightlabs, 2020-07-29:
Is there still interest in this?

BenWoodford, 2020-07-29:
I definitely have an interest in a functional Android build for libsurvive, would be good to get it as part of the CI output

cnlohr, 2020-08-29:
You should be able to use rawdrawandroid for this. https://github.com/cnlohr/rawdrawandroid - would also provide a GUI.

There are examples of connecting to USB devices in there as well.

BenWoodford, 2020-08-29:
Strictly speaking it’s not the test applications that need building for Android but the library itself into something that projects can use - given the nature of Android it’s most likely only going to be used in library form not as a background service (I mean... it could but that’d probably be an uncommon use case) 

My particular use case needs it as a Unity plugin, making the hooks in C# is simple enough as it’s the same as other platforms (DllImport I believe, even though it’s not a DLL) but haven’t a clue where to start with building the Android modules

jdavidberger, 2020-08-29:
I think the raw draw Android repo builds shared library files and does some usb stuff which is a large part of the problem.

The main things that make this hard: I really don't want to write another usb implemention. It does look like libusb sorta works and hidapi definitely does? But the how to build is also an issue. There are a bunch of hardware targets and platform versions to try / verify / build against. To be honest I keep looking at it, getting to a page about how to build for jni and my eyes start to bleed.

BenWoodford, 2020-08-29:
Yeah the information available on building native Android libraries isn’t terribly abundant...

jdavidberger, 2020-08-29:
I've got the main core library building; but libusb / hidapi seems trickier to build than previously thought. This also includes a build from scratch blas/lapack package. Probably worth focusing on building unity plugins first; none of this is very well suited to a CI build.

BenWoodford, 2020-08-29:
AFAIK the Unity plugin would just be a C# wrapper around a Native Android library that externs the relevant methods for C# to call. This is all a bit out of my area of expertise though... 

Great that the core library builds, surprised by hidapi though as I would have thought it’d be a fairly common Android library?

cnlohr, 2020-08-30:
Yeah, that's one rough thing - so, I show how to do it without libusb... It does seem as though libusb may have gotten somewhat functional again.  And it's probably not worth the effort to port over to the native Android USB mechanism.

jdavidberger, 2020-08-31:
I really do hate the idea of adding yet another USB ecosystem in; in theory having HID and libusb is already redundant but in practice it's necessary. As far as I can tell; if you don't partially go through the android USB mechanism you need to have a rooted phone to work though; which is less than ideal too. Will keep poking around to see what other options there might be.

BenWoodford, 2020-08-31:
Just throwing this out there that if it requires a rooted device that throws all use on Android HMDs out the window as none of them have root

sparklightlabs, 2020-09-02:
@jdavidberger, you have the library compiled for android?

jdavidberger, 2020-09-02:
I have the core library; but that doesn't include the USB portions which are pretty important for working with current commercially available gear.

sparklightlabs, 2020-09-02:
Is that on a certain branch? I'd be interested in playing around with it.

jdavidberger, 2020-09-02:
It's actually checked in for the most part; you just invoke cmake with the android cmake toolchain; ie:

`cmake CMAKE_TOOLCHAIN_FILE=~/Android/Sdk/ndk/21.3.6528147/build/cmake/android.toolchain.cmake ..`

You'll also need binary build of lapack; I used clapack to not have to deal with fortran nonsense.

sparklightlabs, 2020-09-02:
Did you roll that clapack binary yourself? Did you have to do anything special to get cmake to find it?

jdavidberger, 2020-09-02:
Yeah; basically this: https://github.com/jdavidberger/clapack

You have to give it `CMAKE_INSTALL_PREFIX` and then install it to a system root; then give the libsurvive cmake process that path to look in for finding stuff. It isn't quite turnkey. In theory you can use opencv too but that is difficult too.

BenWoodford, 2021-08-24:
So a few interesting things have popped up:

- https://github.com/libusb/libusb/pull/874
- This appears to have been updated: https://github.com/libusb/libusb/wiki/Android

Couldn’t tell you what devices have USB OTG support, other than I know the Pico Neo 3 has it and if it was necessary I could perhaps hook someone up with a Neo 3 to help progress things on this front? I also have one, so either myself or one of my team can do some playing around on our end too.