collabora/libsurvive #174: commit "Hot plug functionality is in" breaks OG Vive HMD support

ChristophHaag, 2020-02-13:
Commit: cecf67e1282c4cb03fd17700858a3f6316273f72

starting `./survive-cli` or `./survive-cli --requiredtrackersforcal HMD --allowalltrackersforcal 0`

```
Thread 1 "survive-cli" received signal SIGSEGV, Segmentation fault.
0x00007ffff7f074d0 in survive_data_cb_locked (si=0x55555556dfc8) at ../src/driver_vive.c:2493
2493            if (obj->conf == 0) {
(gdb) bt
#0  0x00007ffff7f074d0 in survive_data_cb_locked (si=0x55555556dfc8) at ../src/driver_vive.c:2493
#1  0x00007ffff7f003cf in survive_data_cb (si=0x55555556dfc8) at ../src/driver_vive.c:508
#2  0x00007ffff7eff7ee in handle_transfer (transfer=0x5555555a7690) at ../src/driver_vive.libusb.h:174
#3  0x00007ffff7af70f5 in ?? () from /usr/lib/libusb-1.0.so.0
#4  0x00007ffff7afbae7 in ?? () from /usr/lib/libusb-1.0.so.0
#5  0x00007ffff7afd3e0 in ?? () from /usr/lib/libusb-1.0.so.0
#6  0x00007ffff7af6b22 in ?? () from /usr/lib/libusb-1.0.so.0
#7  0x00007ffff7af7c0d in libusb_handle_events_timeout_completed () from /usr/lib/libusb-1.0.so.0
#8  0x00007ffff7af7cb1 in libusb_handle_events_completed () from /usr/lib/libusb-1.0.so.0
#9  0x00007ffff7af8609 in ?? () from /usr/lib/libusb-1.0.so.0
#10 0x00007ffff7af88e3 in libusb_control_transfer () from /usr/lib/libusb-1.0.so.0
#11 0x00007ffff7efef20 in getupdate_feature_report (dev=0x555555593460, interface=0, data=0x7ffffffeaa80 "\020", datalen=256) at ../src/driver_vive.libusb.h:52
#12 0x00007ffff7f008ba in hid_get_feature_report_timeout (device=0x555555593460, iface=0, buf=0x7ffffffeaa80 "\020", len=256) at ../src/driver_vive.c:579
#13 0x00007ffff7f0093d in get_feature_report_timeout_locked (ctx=0x555555569040, device=0x555555593460, iface=0, buf=0x7ffffffeaa80 "\020", len=256) at ../src/driver_vive.c:591
#14 0x00007ffff7f02694 in survive_get_config (config=0x7fffffffcfc8, sv=0x55555556d580, usbInfo=0x55555556d590, iface=0, send_extra_magic=1) at ../src/driver_vive.c:939
#15 0x00007ffff7f086d4 in LoadConfig (sv=0x55555556d580, usbInfo=0x55555556d590, iface=0) at ../src/driver_vive.c:2818
#16 0x00007ffff7f01e9e in survive_vive_usb_poll (ctx=0x555555569040, v=0x55555556d580) at ../src/driver_vive.c:840
#17 0x00007ffff7f8768a in survive_poll (ctx=0x555555569040) at ../src/survive.c:773
#18 0x00005555555554bd in main (argc=5, argv=0x7fffffffda88) at ../survive-cli.c:41
```

Bailing from that function with `if (obj == 0) return` makes controllers connected with rf to the HMD work, but the HMD is not tracked.

ChristophHaag, 2020-02-24:
Tracking down why the OG Vive doesn't work was complicated because it was actually two (four) issues.

1. the crash that indeed has an easy workaround: 3c0c726c6f71a8cb96468dc70cc23c6b8bb24a5f
2. the Vive config is not automatically loaded. As a workaround I can load it manually: 00adbed4bd2fd75fb92723e2e0565e8be5987240
3. It fails to autodetect lighthouse gen 1: `Info: Detected LH gen 2 system.`. It requires the parameter `--lighthouse-gen 1`
4. due to #175 it requires the parameter `--lighthousecount 2` to trigger calibration

jdavidberger, 2020-03-02:
Should be fixed in master

ChristophHaag, 2020-03-03:
Seems to work fine now, thanks.

closed 2020-03-03 by ChristophHaag