collabora/libsurvive #175: Valve Index HMD doesn't calibrate with lighthouse v1 on first startup

ChristophHaag, 2020-02-13:
This only happens with v1 basestations. With v2 basestations the calibration works on the first startup.

Running libsurvive without a config present stalls after getting OOTX packets (probably because of `Warning: Could not find disambiguator state for T20`?)
```
./simple_pose_test
Info: Adding tracked object KN0 from HTC
Info: Adding tracked object KN1 from HTC
Info: Adding tracked object T20 from HTC
Info: All enumerated devices attached.
Info: Powered unit on.
Info: Loaded drivers: HTCVive
Info: Knuckles couldn't configure; probably turned off -1 LIBUSB_ERROR_IO
Info: Knuckles couldn't configure; probably turned off -1 LIBUSB_ERROR_IO
Info: T20 is treated as HMD device
Info: Detected LH gen 1 system.
Warning: Could not find disambiguator state for T20
Info: Locked onto state 10(11, 1220083) at 449173309 for T20
Info: OOTX not set for LH 1; attaching ootx decoder using device T20
Info: OOTX not set for LH 0; attaching ootx decoder using device T20
Info: (1) Preamble found
Info: (0) Preamble found
Info: Got OOTX packet 1
Info: Got OOTX packet 0
```

Killing this process and starting libsurvive again (after the first run wrote an initial configuration to disk) runs the calibration and works fine:
```
./simple_pose_test
Info: Adding tracked object KN0 from HTC
Info: Adding tracked object KN1 from HTC
Info: Adding tracked object T20 from HTC
Info: All enumerated devices attached.
Info: Powered unit on.
Info: Loaded drivers: HTCVive
Info: Knuckles couldn't configure; probably turned off -1 LIBUSB_ERROR_IO
Info: Knuckles couldn't configure; probably turned off -1 LIBUSB_ERROR_IO
Info: T20 is treated as HMD device
Info: Detected LH gen 1 system.
Info: Lighthouse 0 position is unset
Info: Lighthouse 1 position is unset
Info: Uncalibrated configuration detected. Attaching calibration. Please don't move tracked objects for the duration of calibration. Pass '--disable-calibrate' to skip calibration
Info: Calibration is using T20
Info: Locked onto state 8(9, 939268) at 2301575901 for T20
Info: Stage 2 good - continuing. 32 1 0
Info: Stage 2 good - continuing. 32 1 1
Info: Stage 2 good - continuing. 32 1 2
Info: Stage 2 moving to stage 3. 32 1 3
Info: Can't solve for cameras with just 0 measurements
Info: Stage 4 succeeded.
Info: Attempting to solve for 0 with 20 meas (T20)
Info: Attempting to solve for 1 with 41 meas (T20)
Info: Solved for 0 with error of 0.000740/0.000077
Info: Solved for 1 with error of 0.000740/0.000077
Info: Using LH 0 (084071d2) as reference lighthouse
Info: Position found for LH 0(084071d2)
Info: Position found for LH 1(925017e7)
Pose: [2559583009][T20][-0.07520859, 0.04444248,-0.03473917] [ang:-0000.54 -0000.51 00000.46 00000.48]
Pose: [2559486806][T20][-0.07520859, 0.04444248,-0.03473917] [ang:-0000.54 -0000.51 00000.46 00000.48]
```

ChristophHaag, 2020-02-24:
Actually it also happens with the OG Vive. bisected to be333de0b41b1adc690a254175a74d53879947c0

ChristophHaag, 2020-02-24:
The fault seems to be the autodetection, using `--lighthousecount 1` or `--lighthousecount 2` calibrates fine, only when omitting the parameter does it stall on the first run.

ChristophHaag, 2020-02-24:
Actually it subsequently started happening with the lighthousecount parameter too with 592e5337af1a44272861d3dd6d953d0e14fcc341

jdavidberger, 2020-03-02:
Should be fixed in master

ChristophHaag, 2020-03-03:
This seems to be fixed, thanks.

closed 2020-03-03 by ChristophHaag