collabora/libsurvive #176: Adding libsurvive as external library to another cmake project

M-J-Murray, 2020-02-27:
I'm attempting to add libsurvive to a project of mine on Linux, however I cannot get it working.
I have run `make` and `make install`, but when I attempt to call `add_package(survive REQUIRED)` it cannot find this library.

As a workaround i found that `find_library(SURVIVE_LIBRARY survive)` picked up the library path. However, I still need to link the survive libraries to my project, because I'm currently getting the error:
```C++
/usr/local/include/libsurvive/survive_types.h:4:10: fatal error: linmath.h: No such file or directory
 #include "linmath.h"
```

Ideally I would like there to be a `Findsurvive.cmake` in `/usr/share/cmake-X.X/Modules`, however that doesn't seem to be the case in this project.
Does anyone know what the proper way to link this library is?

ChristophHaag, 2020-02-28:
The cmake install target does not install the linmath.h header at all here, so I doubt that currently a libsurvive installation done with cmake can be used like that.

For my libsurvive driver in monado I build libsurvive with `add_subdirectory()` and set a bunch of include directories manually.

It would be nice to add a pkg-config file at least, and verify that the installation can be used.

jdavidberger, 2020-02-28:
I don't quite have time right now to get a find cmake file together; but any PR for it would be very welcome. 

The redist stuff should probably be added to the cmake install. 

I typically have used either add_subdirectory in that same way with libsurvive as a submodule. This is probably more appropriate for people working on libsurvive than pure consumers of the library; so a valid make install would be good.

M-J-Murray, 2020-02-28:
@ChristophHaag when you say "set a bunch of include directories manually", could you specify exactly which directories you set? I had a look at your monado survive branch, but I couldn't figure out where you are setting these directories.

I've also attempted to change the CMakeLists.txt to also install redist, however the pathing doesn't seem to be correct. I used `install(DIRECTORY redist/ DESTINATION include/libsurvive)`. What is redist? Why isn't it separated into headers and sources? I'm not really sure what the easiest solution is here.

ChristophHaag, 2020-02-28:
The relevant lines should be something like this
```cmake
add_subdirectory(src/external/libsurvive)

target_include_directories(your_target SYSTEM PRIVATE
	${PROJECT_SOURCE_DIR}/src/external/libsurvive/include
	${PROJECT_SOURCE_DIR}/src/external/libsurvive/include/libsurvive
	${PROJECT_SOURCE_DIR}/src/external/libsurvive/redist)

target_link_libraries(your_target survive)
```

M-J-Murray, 2020-03-02:
Thank you, I managed to get it working.

closed 2020-03-02 by M-J-Murray