collabora/libsurvive #177: Install pose as member function

M-J-Murray, 2020-02-28:
I'm attempting to have `survive_install_pose_fn` called on a member function, however I can't seem to get it working. I've tried an approach using `std::bind(&MyClass::update_pose, this, std::placeholders::_1, std::placeholders::_2, std::placeholders::_3)` but it doesn't work. I've tried a lamda `[this](SurviveObject *so, survive_timecode timecode, SurvivePose *pose){}`, but that doesn't work either.

Has anyone done this successfully? Or do we need to add another version of `survive_install_pose_fn` which takes a friend function.

jdavidberger, 2020-02-28:
Depending on your use case; you might prefer to use the simplified API -- it's more stable, and it runs in it's own thread so you don't have to worry about taking too long in your pose function. It also has python bindings.

In terms of hooking the various callbacks into a C++ class; you  have to keep in mind that the base library is C. You probably want something like this (note that I didn't compile this; so it' probably has some typos or something but the basic pattern is what you want): 

```
class MyPoseClass {
        void update_pose(SurviveObject *so, uint32_t timecode, SurvivePose *pose) {
                /*** Custom Logic ***/ 
        }
public:

        static void _update_pose(SurviveObject *so, uint32_t timecode, SurvivePose *pose) {
                survive_default_pose_process(so, timecode, pose);
                auto _this = static_cast<MyPoseClass>(so->ctx->user_ptr);
                _this->update_pose(so, timecode, pose); 
        }

};

int main(int argc, char** argv) {
        MyPoseClass instance;
        auto ctx = survive_init(argc, argv);
        ctx->user = &instance;

	survive_startup(ctx);

        survive_install_pose_fn(ctx, &MyPoseClass::_update_pose);
	while (survive_poll(ctx) == 0) {
	}
	survive_close(ctx);
        return 0;
}
```

M-J-Murray, 2020-03-02:
Thanks you for that. The user pointer solved my problems.
I ended up using a friend function for the update_pose. Works perfectly now, thank you!

closed 2020-03-02 by M-J-Murray