collabora/libsurvive #180: Missing Button events on Index Controller

ChristophHaag, 2020-03-03:
I get button events for these inputs:

* System click
* A click
* B click
* Trigger click
* Trigger value
* Thumbstick x, y
* Thumbstick click
* Trackpad x, y
* Trackpad touch

I do not get any input events for:

* System touch
* A touch
* B touch
* Squeeze value
* Squeeze force
* Trigger touch
* Thumbstick touch
* Trackpad force

These are all the necessary inputs to implement the `Valve Index Controller` interaction profile [from the OpenXR specification](https://www.khronos.org/registry/OpenXR/specs/1.0/html/xrspec.html#semantic-path-interaction-profiles).

Also missing and nice to have:

* Finger tracking

I have seen code in libsurvive to read finger distance from the handle, but this code has never been executed with my Index controllers.

jdavidberger, 2020-03-03:
Can you see if you can run with `--v 150` and see if it gives you debug output when you do these things? If it does, I can do the mapping if you tell me what it says for each button. If not, I can see about getting it to debug output in the right spot.

ChristophHaag, 2020-03-06:
Unfortunately it's hard to say because with `--v 150` there is a lot of output generated. I'll see if there's a way to comb through it.

jdavidberger, 2020-08-29:
Finally got around to mapping these out; there is a new utility to test it out with: `survive-buttons`. Also mapped some index events -- IPD, head proximity, etc. 

Let me know if anything is missing.

closed 2020-08-29 by jdavidberger