collabora/libsurvive #181: complication failure on ubuntu 18.04 with latest liboopencv-dev installed

eric-schleicher, 2020-03-04:
Have verified this behavior on a number of Ubuntu 18.04 systems.   If they by chance update to latest versions of CV related libraries in Ubuntu compilation of libsurvive fails.

[image]

chasing this down i identified that the the dependency it's referencing is in the livopencv-dev packages.

full log below

these system all have ROS installed, so not sure if there is a relation where.

```
rosuser@vega8:~$ cd libsurvive/
rosuser@vega8:~/libsurvive$ ls
api_example.c  CMakeLists.txt   include   README.md          setup.py            survive_autocomplete.sh  travis-exclusion.lst
attic          data_recorder.c  LICENSE   redist             simple_pose_test.c  survive-cli.c            useful_files
bindings       docs             Makefile  sensors-readout.c  src                 tools                    winbuild
rosuser@vega8:~/libsurvive$ cmake .
-- The C compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Looking for lapacke.h
-- Looking for lapacke.h - not found
-- Found OpenCV: /usr (found version "3.2.0") 
-- Looking for fopencookie
-- Looking for fopencookie - found
-- Looking for funopen
-- Looking for funopen - not found
-- Looking for X11/Xlib.h
-- Looking for X11/Xlib.h - found
Can't build usbmon plugin -- pcap library was not found
-- Using CATKIN_DEVEL_PREFIX: /home/rosuser/libsurvive/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/melodic
-- This workspace overlays: /opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/rosuser/libsurvive/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- The CXX compiler identification is GNU 7.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.20
-- BUILD_SHARED_LIBS is on
-- Configuring done
-- Generating done
-- Build files have been written to: /home/rosuser/libsurvive
rosuser@vega8:~/libsurvive$ make
Scanning dependencies of target minimal_opencv
[  1%] Linking C static library libminimal_opencv.a
[  1%] Built target minimal_opencv
Scanning dependencies of target survive
[  2%] Building C object CMakeFiles/survive.dir/redist/crc32.c.o
[  3%] Building C object CMakeFiles/survive.dir/redist/glutil.c.o
[  5%] Building C object CMakeFiles/survive.dir/redist/jsmn.c.o
[  6%] Building C object CMakeFiles/survive.dir/redist/json_helpers.c.o
[  7%] Building C object CMakeFiles/survive.dir/redist/linmath.c.o
In file included from /home/rosuser/libsurvive/redist/minimal_opencv.h:5:0,
                 from /home/rosuser/libsurvive/redist/linmath.c:11:
/usr/include/opencv2/core/fast_math.hpp:93:1: error: static declaration of ‘cvRound’ follows non-static declaration
 cvRound( double value )
 ^~~~~~~
In file included from /usr/include/opencv2/core/core_c.h:48:0,
                 from /home/rosuser/libsurvive/redist/minimal_opencv.h:4,
                 from /home/rosuser/libsurvive/redist/linmath.c:11:
/usr/include/opencv2/core/types_c.h:929:13: note: previous implicit declaration of ‘cvRound’ was here
     ipt.x = cvRound(point.x);
             ^~~~~~~
CMakeFiles/survive.dir/build.make:158: recipe for target 'CMakeFiles/survive.dir/redist/linmath.c.o' failed
make[2]: *** [CMakeFiles/survive.dir/redist/linmath.c.o] Error 1
CMakeFiles/Makefile2:1565: recipe for target 'CMakeFiles/survive.dir/all' failed
make[1]: *** [CMakeFiles/survive.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

```

jdavidberger, 2020-03-05:
The main issue here is that opencv deprecated their C api; so they aren't keeping it stable from the standpoint of header order / where things live / etc etc. Pretty annoying. 

The opencv library is a fallback though; the main supported package is lapack and friends:
`sudo apt-get install liblapacke-dev libopenblas-dev libatlas-base-dev`

If those are available for your system, I'd install them and redo the cmake process. It should choose that backend instead, and sidestep these issues.

Long term I think the answer is to rename the cv call prefix to be `survive` and write the opencv backend to use the opencv C++ api, which is maintained as stable.

eric-schleicher, 2020-03-05:
Confirmed FIxed

closed 2020-03-05 by eric-schleicher