collabora/libsurvive #182: Receiving Incorrect z velocity from tracker

M-J-Murray, 2020-03-05:
I'm attempting to use a vive tracker on a drone so that I can do flight tests inside. To achieve this I need to feed in odometry data from the tracker into my drone. I have everything setup, and feeding the data into the drone. However, when I was about to do a flight test I was reviewing the data and noticed the z velocity was offset by `0.06`. So when it was still, the library thought the drone was moving up at 6cm/s. I'm using the gen 1 light houses and tracker. I have the code setup using the `survive_install_pose_fn` and `survive_install_velocity_fn` to retrieve the odometry data. Does anyone know what might be causing this?

Also I've noticed that there are two other functions `survive_install_external_pose_fn` and `survive_install_external_velocity_fn`, what is the difference with these two?

jdavidberger, 2020-03-05:
Do you get the same behavior with the IMU turned off -- pass `--no-use-imu` to survive_cli / libsurvive to see. 

The external functions just expose a way to pass in poses and velocities from outside libsurvive into the system. This is useful for comparing how libsurvive is doing against other tracking approaches; but isn't super useful outside of that.

M-J-Murray, 2020-03-05:
Disabling the imu with `--use-imu 0` fixes the issue. In your experience is disabling the IMU a bad idea? I'm going to request the IMU data separately and check if it is generating extraneous data. How do you guys fuse the data? EKF?

M-J-Murray, 2020-03-05:
I just ran it again and plugged into the `survive_install_raw_imu_fn` function. My accelerometer z reading is completely wrong. Is this likely a hardware fault?
```
ACC:{55, 29, -3900}, Gyro:{-5, 8, -3}
ACC:{67, 18, -3910}, Gyro:{-6, 9, -5}
ACC:{74, 31, -3906}, Gyro:{-4, 6, -5}
ACC:{79, 12, -3912}, Gyro:{-8, 8, -6}
ACC:{54, 9, -3918}, Gyro:{-5, 9, -4}
ACC:{70, 19, -3922}, Gyro:{-6, 9, -6}
```

jdavidberger, 2020-03-05:
There is an EKF; but from some of the latest reports I've seen I'm wondering if we are properly aligning sensors to the IMU orientation.

The scale on ACC data for that device is almost certainly 4096 = 1g = 9.81ms^-2. So it isn't crazy off, but the bias is real bad -- 5% or so. The real way to handle this in actual hardware is to recalculate scale and bias at runtime, but that hasn't made it's way in yet -- I think most devices are better calibrated than this but maybe not. 

In terms of disabling IMU; it's typically a pretty useful diagnostic step, and depending on what you need it might be a good idea. I don't know how well the IMU on any given device will handle the vibrations from a drone; I know in the past haptics have really screwed them up. I also imagine there is a better IMU available on the drone. 

The filtering is necessary when the device is moving at moderate to fast speeds; otherwise there are kind of spikes in the positioning as the sensor sweeps aren't taken from the same position. If you plug the non-imued position into an external EKF with an IMU you trust more, you probably get pretty good positioning. 

I need to read up on how to do online calibration. Namely how to properly tune scale and bias from an object standing still. I think it should be relatively straight forward; just hasn't happened yet. 

The filtering also boosts the pose/sec rate; although this is less true with generation 2 lighthouse stuff.

M-J-Murray, 2020-03-05:
There is a lot more to this than I first thought. Thank you for looking into it for me. Let me know if you make any progress on the IMU calibration, as I would be happy to test any fixes you come up with.
I'll disable the tracker IMU for now and just use my drone's IMU. Hopefully, that will be enough. The more inputs sources the better though, so it would be nice to eventually have the tracker IMU enabled.

jdavidberger, 2020-08-17:
The filtering should be giving a much more accurate velocity now; even with noisy IMU data.

closed 2020-08-17 by jdavidberger