collabora/libsurvive #183: Converting angular velocity to euler rad/s

M-J-Murray, 2020-03-06:
I'm testing the `survive_install_velocity_fn` function, but I want to convert the `SurviveVelocity` angular velocity into euler radians per second. I've tried to use `quatfromaxisanglemag`, `quattoeuler` but I'm getting the incorrect euler angles. Does anyone have a recommendation for how this should be done?

jdavidberger, 2020-03-06:
The angular velocity is stored as axis angle, not quaternions. It really
can't be a quaternion since you can have something rotating faster than
360/sec.

Euler has a few conventions on axis ordering, so it'll depend on which you
want. I don't think there are premade lib functions for it, but there are
sources online to do what you want.

M-J-Murray, 2020-03-10:
I solved it by just applying a z rotation to the axis angles.
```
Eigen::Vector3f axis_rot{velocity->AxisAngleRot[0], velocity->AxisAngleRot[1], velocity->AxisAngleRot[2]};

LinmathEulerAngle e;
LinmathQuat q{pose->Rot[0], pose->Rot[1], pose->Rot[2], pose->Rot[3]};
quattoeuler(e, q);
Eigen::Quaternionf q;
q = Eigen::AngleAxisf(e[2], Eigen::Vector3f::UnitZ());
Eigen::Vector3f euler_rad_s = q*axis_rot;
```

closed 2020-03-10 by M-J-Murray