collabora/libsurvive #186: survive-cli: redist/minimal_opencv.c:386: cvSVD: Assertion `inf == 0' failed.

jb55, 2020-03-29:
trying this lib out for the first time...

```
[nix-shell:~/dev/github/cntools/libsurvive]$ ./survive-cli 
Info: Adding tracked object WM0 from HTC
Info: Adding tracked object WM1 from HTC
Info: Adding tracked object HMD from HTC
Info: All enumerated devices attached.
Info: Powered unit on.
Info: Loaded drivers: HTCVive
Info: Detected LH gen 1 system.
Info: Lighthouse 0 position is unset
Info: Lighthouse 1 position is unset
Info: Uncalibrated configuration detected. Attaching calibration. Please don't move tracked objects for the duration of calibration. Pass '--disable-calibrate' to skip calibration
Info: Calibration is using WM0
Info: Locked onto state 5(6, 586021) at 2937566014 for WM0
Info: Locked onto state 1(2, 19838) at 3028827796 for WM1
Info: HMD is treated as HMD device
Warning: Could not find disambiguator state for HMD
Warning: Could not find disambiguator state for HMD
Info: Stage 2 good - continuing. 32 1 0
Warning: Could not find disambiguator state for HMD
Warning: Could not find disambiguator state for HMD
Warning: Could not find disambiguator state for HMD
Warning: Could not find disambiguator state for HMD
Warning: Could not find disambiguator state for HMD
Info: Locked onto state 5(6, 571688) at 4110315645 for HMD
Info: Stage 2 good - continuing. 32 1 1
Info: Stage 2 good - continuing. 32 1 2
Info: Stage 2 moving to stage 3. 32 1 3
Info: Stage 4 succeeded.
 ** On entry to DGESVD parameter number  3 had an illegal value
survive-cli: redist/minimal_opencv.c:386: cvSVD: Assertion `inf == 0' failed.
Aborted (core dumped)
```

jdavidberger, 2020-08-17:
This should be fixed now

closed 2020-08-17 by jdavidberger

jb55, 2020-08-17:
awesome, thanks!