collabora/libsurvive #190: Valve Index kit, cannot pass calibration stage

kwahoo2, 2020-05-11:
I use full Valve Index kit: HMD, 2x knuckles and 2 x LH 2 gen. Light houses are about 3 meters one from another, 2 m above the ground. I also tried many other configuration, including two near LH on the same plane, single LH setup, HMD-only calibration etc. 

simple_pose_test is stuck 99% times at 
```
Info: (0) Preamble found
Info: (1) Preamble found
Info: Got OOTX packet 0
Info: Got OOTX packet 1
``` 

I recorder some data using simple_pose_test --record data.log --record-angle, see attached log. 

[data.log](https://github.com/cntools/libsurvive/files/4611371/data.log)

```
$ ./simple_pose_test --record data.log --record-angle
Info: Recording to 'data.log' Compression: 0
Info: Adding tracked object KN0 from HTC
Info: Adding tracked object KN1 from HTC
Info: Adding tracked object T20 from HTC
Info: All enumerated devices attached.
Info: Powered unit on.
Info: Loaded drivers: HTCVive
Warning: Unknown status event 0xE2 [Time:644C] [Payload: 01 ] <<ABORT FURTHER READ>>
Warning: Read event failed; full payload: e2 01 
Warning: Unknown status event 0xE2 [Time:644C] [Payload: 01 ] <<ABORT FURTHER READ>>
Warning: Read event failed; full payload: e2 01 
Warning: Unknown status event 0xE2 [Time:644F] [Payload: 01 ] <<ABORT FURTHER READ>>
Warning: Read event failed; full payload: e2 01 
Warning: Unknown status event 0xE2 [Time:644F] [Payload: 01 ] <<ABORT FURTHER READ>>
Warning: Read event failed; full payload: e2 01 
Warning: Unknown status event 0xE2 [Time:6452] [Payload: 01 ] <<ABORT FURTHER READ>>
Warning: Read event failed; full payload: e2 01 
Warning: Unknown status event 0xE2 [Time:6452] [Payload: 01 ] <<ABORT FURTHER READ>>
Warning: Read event failed; full payload: e2 01 
Warning: Unknown status event 0xE2 [Time:6452] [Payload: 01 ] <<ABORT FURTHER READ>>
Warning: Read event failed; full payload: e2 01 
Warning: Read light data error -5   [Time:645A] [Payload: d2 e6 c5 0e 34 04 c5 fb 08 00 09 00 02 00 30 00 00 98 fe 02 8d 8e e0 ]
Info: Detected LH gen 2 system.
Warning: Invalid sensor 255 detected hit (0)
Warning: Invalid sensor 255 detected hit (0)
Info: LightcapMode (KN0) 1 -> 2
Info: LightcapMode (KN1) 1 -> 2
Warning: Unknown gen two event 0x64 0b01100100 [Payload: ] <<ABORT FURTHER READ>>
e2 01                                                                                                             |    ..                                                                                                        
Info: T20 is treated as HMD device
Info: LightcapMode (T20) 1 -> 2
Info: Adding lighthouse ch 1 (idx: 0, cnt: 1)
Info: OOTX not set for LH in channel 1; attaching ootx decoder using device KN1
Info: Adding lighthouse ch 0 (idx: 1, cnt: 2)
Info: OOTX not set for LH in channel 0; attaching ootx decoder using device KN1
Info: (0) Preamble found
Info: (1) Preamble found
Info: Got OOTX packet 0
Info: Got OOTX packet 1
```

jdavidberger, 2020-05-12:
The noise on the IMU is making it think the headset is moving a little bit. I've adjusted the threshold in the master branch. If you have more issues with it, adjust it with the command line flag: 

`--move-threshold-gyro 1.0`

and let me know. It's hard to get good heuristics from a limited number of test devices but I raised the limit from .05 to .075 (yours was reporting ~ .06) so that should cover you.

closed 2020-05-12 by jdavidberger

kwahoo2, 2020-05-12:
It did the trick! Thank you.