collabora/libsurvive #191: Playing Nice with SteamVR on Windows

BenWoodford, 2020-05-11:
Right now if you have SteamVR installed on Windows, libsurvive... doesn’t work all that much. It loads up and starts seeing a tracker but does all sorts of weird things and doesn’t appear to calculate a pose (or at least the web visualiser doesn’t show any pose)

Frankly I’m surprised it sees them at all, would have thought it’d need the windows equivalent of usbmon 

According to discussions with @jdavidberger a while back, SteamVR knocks it into another mode?

Annoyingly I don’t have the logs handy right now but it’s very easy to reproduce with a 2.0 tracker and bases. I believe it actually fires an assert error because the byte length matches a different mode entirely

jdavidberger, 2020-05-12:
This is only when steamvr is actively running right? 

It's sorta annoying to test since some stuff auto-launches steamvr, but I'd expect it to be fine without steamvr running?

BenWoodford, 2020-05-12:
Hard to say, as closing SteamVR also turns everything off. Might be possible to force a null driver to test it with perhaps...

Realistically though, there’s a lot more use cases that would open up if it worked in tandem with SteamVR. Not necessarily providing tracking data to it but just being able to get libsurvive data into Unity (as an example) would be awesome. Why? All roads lead back to the 4 basestation limit for me...

jdavidberger, 2020-08-18:
There is an openvr driver using libsurvive. Would just using that hit your use case? It would have to be built out a bit more; but it might be the quicker option

jdavidberger, 2020-10-22:
See: https://github.com/cntools/libsurvive/commit/5ae26e88d87a4e58f73efe5a626944ef1a725f3a

Some work to be done but the concept works. I'm not sure when I'm going to have time to circle back to this; it also requires windows to either use libusb or for the usbmon driver to ingest vendor/product IDs directly. Both will probably be done eventually even outside this issue.

joejarvis64, 2020-11-24:
@jdavidberger  - is there a working / semi-working openvr driver for libsurvive?  I saw the driver_openvr.cc but it did not look like it implemented all the necessary components to work as a driver within SteamVR.

jdavidberger, 2020-11-24:
@joejarvis64 Depends on what you need. `driver_openvr.cc` actually works in the other direction -- it takes SteamVR data and plops it into the libsurvive framework. `openvr_driver.cc` -- and sorry for the naming clash there; it's not great -- in the tools directory exposes libsurvive output to steamvr. 

Right now it currently doesn't handle button stuff; just tracking. It also doesn't handle visual details; such as driving the HMD and all. The button stuff will be done at some point likely, the graphics pipeline probably never will be done -- it's outside the scope of the project. 

For that there is a library -- https://monado.freedesktop.org/ -- which can be configured to use libsurvive and integrate into steamvr in a more holistic way.

joejarvis64, 2020-11-24:
Thanks -- Actually I was looking for something that would replace driver_ligthouse.dll so sounds like driver_openvr is not quite what I needed.  I will check out monado to see if how that works.