collabora/libsurvive #195: Test are failing when built with different versions/implementation of BLAS/LAPACK

im-0, 2020-06-23:
I discovered this when trying to build and run latest master libsurvive (7660270a9174247dac9685da6924ee97e88081f7) on Fedora 32.

* **multi_cal** and **wm1_wand_test_cal_usb** are failing consistently when using blas/lapack 3.9.0: https://gist.github.com/im-0/4c735c003a53af9ccadbc51ff50bcd23. cmake command:

  ```
  CFLAGS="-I/usr/include/cblas -I/usr/include/lapacke -O2" cmake -DUSE_HIDAPI=1 -DENABLE_TESTS=1 ..
  ```

* **lh2_test_cal_usb** is failing consistently when using openblas 0.3.9: https://gist.github.com/im-0/551ac71e596ad974817d6134bf9770bd. cmake command:

  ```
  CFLAGS="-I/usr/include/openblas -O2" cmake -DUSE_HIDAPI=1 -DENABLE_TESTS=1 ..
  ```

Are these failures expected? Or is this a sign of numerical instability of some algorithms in the library? Travis build is not failing, which means that libsurvive passes all tests with blas/lapack 3.6.0 from Ubuntu 16.04.

P.S. Small patch required to build with openblas:
```
diff --git a/redist/CMakeLists.txt b/redist/CMakeLists.txt
index 4b19b0926b70..38dc79ac0ef6 100644
--- a/redist/CMakeLists.txt
+++ b/redist/CMakeLists.txt
@@ -9,7 +9,7 @@ if(USE_LAPACKE)
        set_target_properties(sba PROPERTIES FOLDER "libraries")
        
        IF(NOT WIN32)
-               target_link_libraries(sba lapacke lapack cblas )
+               target_link_libraries(sba openblas )
        ENDIF()
 
 ENDIF()
@@ -61,7 +61,7 @@ ENDIF()
 
 IF(USE_LAPACKE OR WIN32) # NOT USE_OPENCV -- lapack / blas
        IF(UNIX)
-         target_link_libraries(minimal_opencv cblas lapacke)
+         target_link_libraries(minimal_opencv openblas)
        else()
                if (CMAKE_SIZEOF_VOID_P MATCHES 8)
                        set(WIN_PLATFORM "x64")
```

jdavidberger, 2020-06-24:
It isn't my favorite thing that they are failing from that, but it isn't very surprising either -- the tolerances are somewhat tight and the system is somewhat chaotic -- small numerical issues in fixing the position of lighthouses can cause much larger ones in the overall positioning. I'd guess it's the rotational check that is mostly breaking.

How big of a difference is there from what that comes up with and what is expected? It should be pretty close, and if so I think it'd make sense to just bump the tolerance accordingly. 

I'll see if I can make it do a search for openblas vs cblas and act accordingly. Curious why you went with openblas; was it an availability thing or is there some other reason to prefer that library?

im-0, 2020-06-25:
> How big of a difference is there from what that comes up with and what is expected? It should be pretty close, and if so I think it'd make sense to just bump the tolerance accordingly.

Full output of both ctest runs (which shows the difference, if I understand the output right):
* https://gist.github.com/im-0/4c735c003a53af9ccadbc51ff50bcd23
* https://gist.github.com/im-0/551ac71e596ad974817d6134bf9770bd

Not tried to use libsurvive with actual hardware yet, so do not know whether this causes any visible issues or not.

> I'll see if I can make it do a search for openblas vs cblas and act accordingly. Curious why you went with openblas; was it an availability thing or is there some other reason to prefer that library?

Actually, I'm planning to package libsurvive for Fedora. To begin, I figured out -devel dependencies and tried to build and run tests included in libsurvive.

First, I tried with implementation from http://www.netlib.org/ (same that used in Ubuntu 16.04 image in Travis, just newer version). It failed. Then I tried with OpenBLAS, just to check whether changing implementation of math lib will make any difference or not. And with OpenBLAS it also failed, but in different test case.

Also, it looks like OpenBLAS is preferred in Fedora: https://fedoraproject.org/wiki/Changes/OpenBLAS_as_default_BLAS. So, most probably I'll proceed with OpenBLAS if results are the same.

By the way, I'm curious: have you tried to run tests on newer version of Ubuntu? Which contains v3.9.0 of implementation from http://www.netlib.org/ and newer gcc.

jdavidberger, 2020-07-27:
I think the latest on Ubuntu I've tried is the 3.8 stuff. 

Now if you specify -DUSE_OPENBLAS=On it'll use it. It still uses lapacke; but it'll swap out the cblas backend which I think is what you want. On my setup at least; it still must link to LAPACKE but let me know if this breaks on your end.

closed 2020-08-17 by jdavidberger