collabora/libsurvive #196: Can't connect second lighthouse and diverging data

hortejak, 2020-07-08:
Hi,
I really love your work here, what you have done here is amazing. I have an issue (I am sure it is something rather easy to solve I just can't see), I need help with connecting second lighthouse. I have two version 2 lighthouses and 1 tracker. I simply need to track, where the tracker is in the space. I am testing it via simple_pose_test.c, which sends few warnings and then works for a little while. after 2 sec of working data start running to negative 20 (sometimes x, other times y or z). The tracker does not move at all. I was wondering if this is caused by the lack of second lighthouse or do you please have another idea what is wrong with it (on my side) and how to fix it? Thanks for every idea.
This is the warning:
Warning: MPFIT failure T20 2.503089/2.146718 (33 measurements, 2 result, 1 lighthouses, 0 canSolveLHs)

hortejak

jdavidberger, 2020-07-13:
It sounds like the calibration didn't work well or that there are reflections in the room that caused it to miscalculate position. You can reset calibration by deleting the config.json file.

That particular warning indicates that the error is pretty high; you'd expect that 1.66 number to be more like .001. Can you do a recording -- just add the argument `--record myplayback.rec.gz` and attach it?

hortejak, 2020-07-14:
Hi,
yea you are absolutely right, the problem is in calibration, I think we can close this and resolve the calibration itself first. With well calibrated 1 LH and poser MPFIT this happens like once or twice during movement, which is surely due to reflections and other unwanted effects. Thanks for the help and I hope we can resolve the issue of calibrating multiple LHs.
hortejak

closed 2020-07-14 by hortejak