collabora/libsurvive #197: Potential USB Claim Interface Problem

webgeek1234, 2020-07-08:
https://github.com/cntools/libsurvive/blob/0b4f49e1958057b08533ee43f3610b48bf060c64/src/driver_vive.libusb.h#L140

When some work with libsurvive for the original Vive on Android, I was getting a lot of log spam from the watchman interfaces. That libusb couldn't claim the interface, then a bunch of submiturb errors trying to talk to them anyways. Which lead me to the above line, where the claim_interface call gets returned to a rescoped local 'ret' variable instead of the function level one. Is this intentional, that claim interface isn't fatal for a given object?

jdavidberger, 2020-07-13:
Yes, that rescoping is a problem. Fixed in latest master. 

It'll now probably not work at all for you if it can't claim that interface; was the original usb permissions or something?

webgeek1234, 2020-07-13:
Thanks.

I'm still trying to figure it out. The two watchman interfaces are getting error -22 or EINVAL from disconnect and claim in the kernel. As far as I can tell, the only way that can happen is if usb_ifnum_to_if fails (see link below). And that's doing something comparing the interface num to the altsetting[0]. I'm not familiar enough with low level usb stuff to know what's really going on. Right now I'm trying to get libusb to print the altsettings and see why that lookup would fail.

https://gitlab.incom.co/CM-Shield/android_kernel_nvidia_linux-4.9_kernel_kernel-4.9/-/blob/lineage-16.0/drivers/usb/core/devio.c#L2311

jdavidberger, 2020-07-13:
You've probably already found this, but there are a few lines in that driver_vive.libusb.h that set either call libusb_set_debug or libusb_set_option. You can set that to higher debug levels which might help.

Not super familiar with android USB details to know what might be doing that though.

webgeek1234, 2020-07-13:
No I hadn't seen that, actually. Thanks, flipping that on for my next build.

At this level, libusb should act the same on Android and Linux. I did have permission issues early on because all the usb nodes are own by group 'usb' and the service I was running wasn't a member. But even with seemingly proper permissions, the watchman nodes aren't working. The main HMD node seems fine, though.

I added conf->interface[j].altsetting[0].bInterfaceNumber to the claim interface error print in hopes of seeing what the kernel usb_ifnum_to_if function would be comparing against and it said '0' as expected. I'm pretty confused here. Maybe turning up libusb logging will show something.

webgeek1234, 2020-07-14:
Update: I found the issue. Turns out there's a bug in the xusb firmware on the device I'm using. That caused the controller to think it ran out of resources when the watchman interfaces were enumerating. They're coming up fine now.

closed 2020-07-14 by webgeek1234