collabora/libsurvive #199: Calibration

hortejak, 2020-07-13:
Hey guys,
I have a problem with calibration of my gen2 LHs. When I run for example api_example, there is no information about lighthouses in config.json without me manually typing it ... and I have no idea what values for pose and fcalphase and so on I have to type. Do you have any idea how to calibrate them properly? Thanks guys.
hortejak

jdavidberger, 2020-07-13:
You should not have to manually fill those in. Those should auto-fill in if you let it calibrate on the first run of api_example. 

If that isn't working, please do something like:

`./api_example --v 100`

and paste the input in the issue.

hortejak, 2020-07-14:
Hi,
I have tried it with just one LH and it works just fine, calibrates itself and works fine, problem comes, when I add second LH ... with it config file has only info about generation and disambiguator. I add the output of the command you suggested, hope we can find a solution to this. No matter what I want to thank you for caring about this issue.
hortejak
`SimpleApi: Available Posers:

SimpleApi: 	*MPFIT
SimpleApi: 	 Dummy
SimpleApi: 	 IMU
SimpleApi: 	 SBA
SimpleApi: 	 CharlesRefine
SimpleApi: 	 BaryCentricSVD
SimpleApi: 	 EPNP
SimpleApi: Totals 7 Posers.
SimpleApi: Using 'MPFIT' for poser
SimpleApi: Available Disambiguators:
SimpleApi: 	 Charles
SimpleApi: 	 Turvey
SimpleApi: 	*StateBased
SimpleApi: Totals 3 Disambiguators.
SimpleApi: Using 'StateBased' for disambiguator
SimpleApi: Successfully enumerated Watchman (1) 28de:2101
SimpleApi: Successfully enumerated Watchman 28de:2101
SimpleApi: Adding tracked object WM0 from HTC
SimpleApi: Available Posers:
SimpleApi: 	*MPFIT
SimpleApi: 	 Dummy
SimpleApi: 	 IMU
SimpleApi: 	 SBA
SimpleApi: 	 CharlesRefine
SimpleApi: 	 BaryCentricSVD
SimpleApi: 	 EPNP
SimpleApi: Totals 7 Posers.
SimpleApi: Using 'MPFIT' for poser
SimpleApi: Attaching IMU/Lightcap/Buttons(0x81) for WM0
SimpleApi: All enumerated devices attached.
SimpleApi: Powered unit on.
SimpleApi: Loaded drivers: HTCVive
Found 'WM0'
SimpleApi: Got config data length 2194 for Watchman:0
SimpleApi: Read config for WM0
SimpleApi: Read light data error -5   [Time:4B6F] [Payload: 90 2a 7c 00 02 00 f8 ef 13 00 03 00 df ff 08 20 99 02 e8 01 b9 03 4e be 1f ]
SimpleApi: Unknown status event 0xE2 [Time:4B6F] [Payload: 01 ] <<ABORT FURTHER READ>>
SimpleApi: Read event failed; full payload: e2 01 
SimpleApi: Unknown status event 0xE2 [Time:4B72] [Payload: 01 ] <<ABORT FURTHER READ>>
SimpleApi: Read event failed; full payload: e2 01 
SimpleApi: Unknown status event 0xE2 [Time:4B72] [Payload: 01 ] <<ABORT FURTHER READ>>
SimpleApi: Read event failed; full payload: e2 01 
SimpleApi: Unknown status event 0xE2 [Time:4B72] [Payload: 01 ] <<ABORT FURTHER READ>>
SimpleApi: Read event failed; full payload: e2 01 
SimpleApi: Unknown status event 0xE2 [Time:4B75] [Payload: 01 ] <<ABORT FURTHER READ>>
SimpleApi: Read event failed; full payload: e2 01 
SimpleApi: Unknown status event 0xE2 [Time:4B75] [Payload: 01 ] <<ABORT FURTHER READ>>
SimpleApi: Read event failed; full payload: e2 01 
SimpleApi: Unknown status event 0xE2 [Time:4B77] [Payload: 01 ] <<ABORT FURTHER READ>>
SimpleApi: Read event failed; full payload: e2 01 
SimpleApi: Unknown status event 0xE2 [Time:4B77] [Payload: 01 ] <<ABORT FURTHER READ>>
SimpleApi: Read event failed; full payload: e2 01 
SimpleApi: Read light data error -5   [Time:4B77] [Payload: d3 f5 71 00 09 00 df ef 12 00 05 00 e1 ff 70 08 00 00 a8 bd 03 19 bf 1f ]
SimpleApi: Gen2 reason: WM0 Lightcap length >= 0x8000
SimpleApi: Detected LH gen 2 system.
SimpleApi: Invalid sensor 255 detected hit (28)
SimpleApi: Invalid sensor 255 detected hit (26)
SimpleApi: LightcapMode (WM0) 1 -> 2
SimpleApi: WM0 Unknown flag 0x40 byte 30 00
SimpleApi: WM0 Unknown flag 0x40 byte 30 00
SimpleApi: WM0 Unknown flag 0x40 byte 30 00
SimpleApi: WM0 Unknown flag 0x40 byte 30 00
SimpleApi: WM0 Unknown flag 0x40 byte 70 0a
SimpleApi: WM0 Unknown flag 0x40 byte 20 01
SimpleApi: Adding lighthouse ch 0 (idx: 0, cnt: 1)
SimpleApi: OOTX not set for LH in channel 0; attaching ootx decoder using device WM0
SimpleApi: WM0 Unknown flag 0x40 byte 20 01
SimpleApi: WM0 Unknown flag 0x40 byte 20 01
SimpleApi: WM0 Unknown flag 0x40 byte 20 01
SimpleApi: WM0 Unknown flag 0x40 byte 20 01
SimpleApi: WM0 Unknown flag 0x40 byte 20 01
SimpleApi: WM0 Unknown flag 0x40 byte 20 01
SimpleApi: WM0 Unknown flag 0x40 byte 20 01
SimpleApi: WM0 Unknown flag 0x40 byte 80 2c
SimpleApi: (0) Preamble found
SimpleApi: (0) OOTX Decoder: Bad sync bit
SimpleApi: WM0 Unknown flag 0x40 byte 80 2c
SimpleApi: WM0 Unknown flag 0x40 byte 80 2c
SimpleApi: (0) Preamble found
SimpleApi: (0) OOTX Decoder: Bad sync bit
SimpleApi: WM0 Unknown flag 0x40 byte 80 2c
SimpleApi: WM0 Unknown flag 0x40 byte 80 2c
SimpleApi: (0) Preamble found
SimpleApi: WM0 Unknown flag 0x40 byte 80 2c
SimpleApi: (0) OOTX Decoder: Bad sync bit
SimpleApi: WM0 Unknown flag 0x40 byte 80 2b
SimpleApi: (0) Preamble found
SimpleApi: (0) OOTX Decoder: Bad sync bit
SimpleApi: WM0 Unknown flag 0x40 byte 80 2b
SimpleApi: (0) Preamble found
SimpleApi: (0) OOTX Decoder: Bad sync bit
SimpleApi: (0) Preamble found
SimpleApi: (0) OOTX Decoder: Bad sync bit
SimpleApi: WM0 Unknown flag 0x40 byte 80 2b
SimpleApi: WM0 Unknown flag 0x40 byte 80 2b
`

ash891228, 2020-07-19:
I am running into similar issues. By default, does it require HMD or it can work with Tracker only? I am trying to use it without HMD.
I've tested both 1 and 2 LHs and it never had any staged calibration output and the config file generated is:
```
"v":"0",
"lighthouse-gen":"0",
"poser":"MPFIT",
"disambiguator":"StateBased"
```
and my output is:
```
Info: Available Posers:
Info: 	 IMU
Info: 	 Dummy
Info: 	 CharlesRefine
Info: 	 BaryCentricSVD
Info: 	*MPFIT
Info: 	 EPNP
Info: 	 SBA
Info: Totals 7 Posers.
Info: Using 'MPFIT' for poser
Info: Available Disambiguators:
Info: 	*StateBased
Info: 	 Turvey
Info: 	 Charles
Info: Totals 3 Disambiguators.
Info: Using 'StateBased' for disambiguator
Info: Successfully enumerated Watchman (1) 28de:2101
Info: Successfully enumerated Watchman 28de:2101
Info: Adding tracked object WM0 from HTC
Info: Available Posers:
Info: 	 IMU
Info: 	 Dummy
Info: 	 CharlesRefine
Info: 	 BaryCentricSVD
Info: 	*MPFIT
Info: 	 EPNP
Info: 	 SBA
Info: Totals 7 Posers.
Info: Using 'MPFIT' for poser
Info: Attaching IMU/Lightcap/Buttons(0x81) for WM0
Info: All enumerated devices attached.
Info: Powered unit on.
Info: Loaded drivers: HTCVive
Info: Force calibrate flag set -- clearing position on all lighthouses
Info: Got config data length 2194 for Watchman:0
Info: Read config for WM0
Warning: Read light data error -5   [Time:88EB] [Payload: 90 d0 79 00 5b 00 f6 ef de ff 0e 00 ee ff 48 48 97 33 af 68 16 b1 27 25 64 87 ]
Warning: Unknown status event 0xE2 [Time:88EE] [Payload: 01 ] <<ABORT FURTHER READ>>
Warning: Read event failed; full payload: e2 01 
Warning: Unknown status event 0xE2 [Time:88EE] [Payload: 01 ] <<ABORT FURTHER READ>>
Warning: Read event failed; full payload: e2 01 
Warning: Unknown status event 0xE2 [Time:88EE] [Payload: 01 ] <<ABORT FURTHER READ>>
Warning: Read event failed; full payload: e2 01 
Warning: Unknown status event 0xE2 [Time:88F1] [Payload: 01 ] <<ABORT FURTHER READ>>
Warning: Read event failed; full payload: e2 01 
Warning: Unknown status event 0xE2 [Time:88F1] [Payload: 01 ] <<ABORT FURTHER READ>>
Warning: Read event failed; full payload: e2 01 
Warning: Unknown status event 0xE2 [Time:88F4] [Payload: 01 ] <<ABORT FURTHER READ>>
Warning: Read event failed; full payload: e2 01 
Warning: Unknown status event 0xE2 [Time:88F4] [Payload: 01 ] <<ABORT FURTHER READ>>
Warning: Read event failed; full payload: e2 01 
Warning: Unknown status event 0xE2 [Time:88F4] [Payload: 01 ] <<ABORT FURTHER READ>>
Warning: Read event failed; full payload: e2 01 
Info: Gen2 reason: WM0 Lightcap length >= 0x8000
Info: Detected LH gen 2 system.
Info: LightcapMode (WM0) 1 -> 2
Info: WM0 Unknown flag 0x40 byte 30 00
Info: WM0 Unknown flag 0x40 byte 80 23
```

jdavidberger, 2020-07-27:
It should support just a tracker; but I think there is an issue right now with it figuring out it's Gen2 from just the tracker. I'll try to replicate the issue here tomorrow; but for now can you set "lighthouse-gen" to 2 to manually specify it's gen2 and see if that works?

jdavidberger, 2020-07-27:
I pushed a few fixes; it might report a few errors before it figures out what is going on but it should figure it out in a packet or two.

sparklightlabs, 2020-07-29:
Just a heads up, I'm getting pretty bad calibrations just using an original vive controller on lighthouse v1. 

This is the same data set run with a calibration from the HMD
[image]

And calibration/config file generated with the vive controller
[image]

sparklightlabs, 2020-07-29:
Because a solid calibration is paramount for tracking, I'm wondering if there ought to be some redundancy. I.E. If someone doesn't have a config.json already, run calibration twice and see if the estimated pose for the lighthouses come out the same.

jdavidberger, 2020-07-29:
It should have given you error numbers for the calibration, were they high? That looks quite bad; but the baseline on the HMD is much nicer for this than the controllers. From the screenshots it looks like it flipped the LH upside down which it can do if the sensors look planar to it. Was the controller hooked in via USB or RF? 

Possibly the thing to do is to do more of an on-line solve -- whenever the controller is stationary collection a snapshot of data, and if the confidence in the calibration is low enough resolve it with multiple snapshots as opposed to the singular one we do right now.

ash891228, 2020-08-01:
> It should support just a tracker; but I think there is an issue right now with it figuring out it's Gen2 from just the tracker. I'll try to replicate the issue here tomorrow; but for now can you set "lighthouse-gen" to 2 to manually specify it's gen2 and see if that works?

I pulled the changes but the config file generated is still the same. I manually changed it to 2 but it makes no difference.

However, with the changes I get the following output by running  
```
./survive-cli --force-calibrate -v 100
Info: Available Posers:
Info: 	 IMU
Info: 	 Dummy
Info: 	 CharlesRefine
Info: 	 BaryCentricSVD
Info: 	*MPFIT
Info: 	 EPNP
Info: 	 SBA
Info: Totals 7 Posers.
Info: Using 'MPFIT' for poser
Info: Available Disambiguators:
Info: 	*StateBased
Info: 	 Turvey
Info: 	 Charles
Info: Totals 3 Disambiguators.
Info: Using 'StateBased' for disambiguator
Info: Successfully enumerated Watchman (1) 28de:2101
Info: Successfully enumerated Watchman 28de:2101
Info: Adding tracked object WM0 from HTC
Info: Available Posers:
Info: 	 IMU
Info: 	 Dummy
Info: 	 CharlesRefine
Info: 	 BaryCentricSVD
Info: 	*MPFIT
Info: 	 EPNP
Info: 	 SBA
Info: Totals 7 Posers.
Info: Using 'MPFIT' for poser
Info: Attaching IMU/Lightcap/Buttons(0x81) for WM0
Info: All enumerated devices attached.
Info: Powered unit on.
Info: Loaded drivers: HTCVive
Info: Force calibrate flag set -- clearing position on all lighthouses
Warning: Watchman couldn't configure; probably turned off -1 LIBUSB_ERROR_IO
^CInfo: Libusb poll failed. -10 (LIBUSB_ERROR_INTERRUPTED)
Info: MPFIT stats for WM0:
Info: MPFIT overall stats:
Info: 	seed runs         0 / 0
Info: 	error failures    0
Info: Cleaning up interface on 3 IMU/Lightcap/Buttons
Info: Cleaning up transfer on 3 IMU/Lightcap/Buttons
```
Given it is only a warning, is this expected?

jdavidberger, 2020-08-01:
This is with a dongle correct? 

The "Probably turned off" warning isn't fatal since it's valid to track with the HMD and not the controllers. Is the light on the controller turned on? You might have to turn it on manually after you see that message?

ash891228, 2020-08-01:
> This is with a dongle correct?
> 
> The "Probably turned off" warning isn't fatal since it's valid to track with the HMD and not the controllers. Is the light on the controller turned on? You might have to turn it on manually after you see that message?

Yes it is with a dongle. I rerun again and it works now. It successfully generates WM0_config.json. However, I don't get the output like the example:
```
Info: Attempting to solve for 0 with 30 meas
Info: Attempting to solve for 1 with 21 meas
Info: Solved for 0 with error of 0.005446/0.000000
Info: Solved for 1 with error of 0.005446/0.000000
Info: Using LH 0 (d99e7eac) as reference lighthouse
Info: Position found for LH 0(d99e7eac)
Info: Position found for LH 1(fe0398ef)
```

The output I am getting is:
```
Info: Adding tracked object WM0 from HTC
Info: All enumerated devices attached.
Info: Powered unit on.
Info: Loaded drivers: HTCVive
Info: Force calibrate flag set -- clearing position on all lighthouses
Info: Detected LH gen 2 system.
Info: LightcapMode (WM0) 1 -> 2
```
Hangs here for like 5 minutes and after killing I am getting.
```
Info: MPFIT stats for WM0:
Info: 	seed runs         0 / 0
Info: 	error failures    0
Info: Cleaning up interface on 3 IMU/Lightcap/Buttons
```
Is this right?

jdavidberger, 2020-08-01:
No for whatever reason it's not trying to calibrate. Maybe the force flag isn't working? 

- try setting the position set fields in the config file to 0 and trying
- add "--v 100" to get more output, might have more clues

I used the force calibrate flag the other day, so it should work but maybe it's different for wireless for some reason.

ash891228, 2020-08-01:
> No for whatever reason it's not trying to calibrate. Maybe the force flag isn't working?
>
>     * try setting the position set fields in the config file to 0 and trying
>
>     * add "--v 100" to get [...]

The output is 
```
Info: Available Posers:
Info: 	 IMU
Info: 	 Dummy
Info: 	 CharlesRefine
Info: 	 BaryCentricSVD
Info: 	*MPFIT
Info: 	 EPNP
Info: 	 SBA
Info: Totals 7 Posers.
Info: Using 'MPFIT' for poser
Info: Available Disambiguators:
Info: 	*StateBased
Info: 	 Turvey
Info: 	 Charles
Info: Totals 3 Disambiguators.
Info: Using 'StateBased' for disambiguator
Info: Successfully enumerated Watchman (1) 28de:2101
Info: Successfully enumerated Watchman 28de:2101
Info: Adding tracked object WM0 from HTC
Info: Available Posers:
Info: 	 IMU
Info: 	 Dummy
Info: 	 CharlesRefine
Info: 	 BaryCentricSVD
Info: 	*MPFIT
Info: 	 EPNP
Info: 	 SBA
Info: Totals 7 Posers.
Info: Using 'MPFIT' for poser
Info: Attaching IMU/Lightcap/Buttons(0x81) for WM0
Info: All enumerated devices attached.
Info: Powered unit on.
Info: Loaded drivers: HTCVive
Info: Force calibrate flag set -- clearing position on all lighthouses
Info: Got config data length 2194 for Watchman:0
Info: Read config for WM0
Info: Gen2 reason: WM0 Lightcap length >= 0x8000
Info: Detected LH gen 2 system.
Info: LightcapMode (WM0) 1 -> 2
```
I don't see a position set field. My config file is:
```
"lighthouse-gen":"2",
"v":"0",
"poser":"MPFIT",
"disambiguator":"StateBased",
"ootx-ignore-sync-error":"0",
"requiredtrackersforcal":"",
"allowalltrackersforcal":"0"
```
For your test case, was is directly connecting the tracker to the PC?

jdavidberger, 2020-08-17:
The latest master has a lot of logic to make calibration against a tracker more reliable. Is it possible for you to retry with the latest and verify the fix?

hortejak, 2020-08-18:
Hi,
I do not know if it is just me, but with the latest master I cannot seem to get most of the test programs (simple_pose_test, survive-cli, ...) working. It seems to calibrate, but after that it does nothing. Here is log from calibration:
```
Info: Adding tracked object WM0 from HTC
Info: All enumerated devices attached.
Info: Powered unit on.
Info: Loaded drivers: HTCVive
Info: Detected LH gen 2 system.
Info: LightcapMode (WM0) 1 -> 2
Info: Adding lighthouse ch 0 (idx: 0, cnt: 1)
Info: OOTX not set for LH in channel 0; attaching ootx decoder using device WM0
Info: (0) Preamble found
Info: Got OOTX packet 0 b424bcda
```
This creates config.json:
```
"v":"0",
"lighthouse-gen":"0",
"poser":"MPFIT",
"disambiguator":"StateBased",
"ootx-ignore-sync-error":"0"
"lighthouse0":{
"index":"0",
"id":"-1272660774",
"mode":"0",
"pose":["0.000000","0.000000","0.000000","0.000000","0.000000","0.000000","0.000000"],
"fcalphase":["0.000000","-0.008163"],
"fcaltilt":["-0.054138","0.045044"],
"fcalcurve":["0.296631","0.196899"],
"fcalgibpha":["1.502930","1.514648"],
"fcalgibmag":["-0.005028","-0.006748"],
"fcalogeephase":["1.938477","1.269531"],
"fcalogeemag":["0.044800","0.159668"],
"OOTXSet":"1",
"PositionSet":"0"
}
```
which seems to be alright (worked like this in previous versions) but after this happens, nothing else does ...
What is odd, sensors-readout seems to work well.
If started with config.json file already created, it prints just this:
```
Info: Adding tracked object WM0 from HTC
Info: All enumerated devices attached.
Info: Powered unit on.
Info: Loaded drivers: HTCVive
Info: Detected LH gen 2 system.
Info: LightcapMode (WM0) 1 -> 2
```
Do you need some kind of data from sensors-readout? I will not get much from the others.
Thanks,
hortejak

jdavidberger, 2020-08-18:
Could you make a recording? https://github.com/cntools/libsurvive#data-recording

It looks like your base stations are on the same channel which would break using both at the same time. If you start steam vr with a headset it'll fix them. There is also a script in https://github.com/cntools/libsurvive/blob/master/tools/bsd_ctrl/bsd_ctrl.py that will change them for you if you have a Bluetooth adapter.

hortejak, 2020-08-18:
Yeah sure,
[logs.zip](https://github.com/cntools/libsurvive/files/5091076/logs.zip)
hope it helps. Well it is possible that they are on the same channel, I will try to look at that script, I unfortunately do not have a headset, so it is possible that they need some kind of initial settings. What concerns me, is that it does not work even with one Base station ... just the sensors-readout. The config file looks kind of incomplete while calibrating during recording.
Thanks for the help, hortejak.

hortejak, 2020-08-18:
Yeah sure,
[logs.zip](https://github.com/cntools/libsurvive/files/5091129/logs.zip)
hope it helps. Well it is possible that they are on the same channel, I will try to look at that script, I unfortunately do not have a headset, so it is possible that they need some kind of initial settings. What concerns me, is that it does not work even with one Base station ... just the sensors-readout.
Thanks for the help, hortejak.

jdavidberger, 2020-08-18:
Are there any reflective surfaces near the object? The light data is very noisy, which makes the object hold off on calibration since it really wants a stationary object for that. It seems like the noise is between two fixed positions though.

You can also reset the mode via the USB jack at the back. If the script doesn't work, I can put instructions up for that.

hortejak, 2020-08-18:
I have just switched channels and it finally finds both stations, yea it says that camera reprojection error was too high, I will move to a darker room right away. Thanks for the help, you are great.

hortejak, 2020-08-18:
Good now it "works", but with BaryCentricSVD poser it is extremely jittery and squelchy. With the MPFIT poser it does this:
```
Info: Attempting to solve for 0 with 22 meas from device WM0
Info: Attempting to solve for 1 with 20 meas from device WM0
Info: Assuming object position of +0.000000e+00   +0.000000e+00   +0.000000e+00	+1.000000e+00   +0.000000e+00   +0.000000e+00   +0.000000e+00
Warning: MPFIT failure WM0 0.393005/0.364696 (42 measurements, 1 result, 2 lighthouses, 1 canSolveLHs, 3 since success, run #71)
Warning: Seed poser failed for 1, not trying to solve LH system
Warning: Seed poser failed for 1, not trying to solve LH system
Warning: Seed poser failed for 1, not trying to solve LH system
Warning: Seed poser failed for 1, not trying to solve LH system
Warning: Seed poser failed for 1, not trying to solve LH system
Warning: Seed poser failed for 1, not trying to solve LH system
Warning: Seed poser failed for 1, not trying to solve LH system
Warning: Seed poser failed for 1, not trying to solve LH system
Warning: Seed poser failed for 1, not trying to solve LH system
Warning: Seed poser failed for 1, not trying to solve LH system
Warning: Seed poser failed for 1, not trying to solve LH system
```
The conditions in this room are quite nice, so it is unlikely it is caused purely due to reflections. Tomorrow I will try 4 base stations if it does get better. Thanks for all your help, really appreciate it.
hortejak

jdavidberger, 2020-08-18:
That looks better but not great. Maybe make a recording of that? 

Sunlight coming in typically will impact the range of the lighthouse; but if the lighthouses and the objects are far away from the window, you won't see too much reflection typically. 

It's weird that MPFIT bounces out like that. Even when the system is completely FUBARed it can usually improve on the initial guess somewhat. You can set `--max-cal-error 1` to allow that high of error; but it almost certainly isn't going to be correct. I bet the BaryCentricSVD is misplacing the lighthouses completely, which means MPFIT can't adjust them properly and that the pose output from BaryCentricSVD is just noise. You can use the [viz tool](https://github.com/cntools/libsurvive/#visualization) to see where it is placing the lighthouses; I bet it is way off. 

How do you have your lighthouses arranged? For best tracking, having them at 90 degree angles -- so that, say, one of them is a few meters in front and the other is a few meters to the right -- is ideal; and it might help to set them up like that. BUT if you have an arrangement that breaks the SVD solver; I'd still like a recording since it should still work; and with the data I probably could figure out why it doesn't.

hortejak, 2020-08-19:
Well, I tried it today (moved it just a few inches) and it works really really well. So the problem was in my choice of space ... I doubt that you want a recording of a fine working program, so I will send you that promised recording when I brake it again :D, thanks for the huge amount of help and keep up the good work.
hortejak

jdavidberger, 2020-08-19:
That's good to hear! I poked around at your original recording and had it filtered enough to calibrate but need to put more thought into how it should work and how much it should be allowed to impact runtime. It might work it's way into the code at some point but right now it's too risky since it might mis filter valid moving injected.

closed 2020-08-19 by jdavidberger

ash891228, 2020-08-25:
> The latest master has a lot of logic to make calibration against a tracker more reliable. Is it possible for you to retry with the latest and verify the fix?

Sorry for the late reply, I pull the updates and it still fails with a dongle but works when directly plugged into the PC. Is dongle currently not supported?

jdavidberger, 2020-08-25:
The dongle is supported; does it show any data going through?

ash891228, 2020-08-26:
> The dongle is supported; does it show any data going through?

I don't think so. The output is pasted below and I killed it after 3min:
```
./bin/survive-cli --force-calibration -v100
Info: Available Posers:
Info: 	 Dummy
Info: 	 BaryCentricSVD
Info: 	*MPFIT
Info: 	 KalmanOnly
Info: Totals 4 Posers.
Info: Using 'MPFIT' for poser
Info: Available Disambiguators:
Info: 	*StateBased
Info: Totals 1 Disambiguators.
Info: Using 'StateBased' for disambiguator
Info: Adding tracked object WM0 from HTC
Info: Available Posers:
Info: 	 Dummy
Info: 	 BaryCentricSVD
Info: 	*MPFIT
Info: 	 KalmanOnly
Info: Totals 4 Posers.
Info: Using 'MPFIT' for poser
Info: All enumerated devices attached.
Info: Powered unit on.
Info: Loaded drivers: HTCVive
Info: Detected LH gen 2 system.
Info: LightcapMode (WM0) 1 -> 2

^CInfo: Libusb poll failed. -10 (LIBUSB_ERROR_INTERRUPTED)
Info: MPFIT stats for WM0:
Info: MPFIT overall stats:
Info: 	meas failures     0
Info: 	total iterations  0
Info: 	avg iterations    0.000000
Info: 	total fevals      0
Info: 	avg fevals        0.000000
Info: 	total runs        0
Info: 	avg error         0.0000000000
Info: 	avg orig error    0.0000000000
Info: 	Status  MP_OK_CHI 0
Info: 	Status  MP_OK_PAR 0
Info: 	Status MP_OK_BOTH 0
Info: 	Status  MP_OK_DIR 0
Info: 	Status MP_MAXITER 0
Info: 	Status    MP_FTOL 0
Info: 	Status    MP_XTOL 0
Info: 	Status    MP_GTOL 0
Info: 	Status MP_OK_NORM 0
Info: 	seed runs         0 / 0
Info: 	error failures    0
Info: Cleaning up interface on 3 IMU/Lightcap/Buttons
```

jdavidberger, 2020-08-26:
What color is the LED on the tracker when you do this?

ash891228, 2020-08-26:
> What color is the LED on the tracker when you do this?

Green.

jdavidberger, 2020-08-26:
Try with the verbose flag at 500 and paste that output? Also watch out -- there should be a space between `-v` and `500

ash891228, 2020-08-26:
> Try with the verbose flag at 500 and paste that output? Also watch out -- there should be a space between `-v` and `500`

I got this, and it just repeats the last line.
```
 ./bin/survive-cli --force-calibration -v 500
Info: Config file is /home/shuo/.config/libsurvive/config.json
Info: Adding LH 0 mode: 2 id: 8644a1e9
Info: Adding LH 1 mode: 1 id: d06a444d
Info: Available Posers:
Info: 	 Dummy
Info: 	 BaryCentricSVD
Info: 	*MPFIT
Info: 	 KalmanOnly
Info: Totals 4 Posers.
Info: Using 'MPFIT' for poser
Info: Available Disambiguators:
Info: 	*StateBased
Info: Totals 1 Disambiguators.
Info: Using 'StateBased' for disambiguator
Info: 	usb-hz-output: -1
Info: Successfully enumerated Watchman (1) 28de:2101
Info: Successfully enumerated Watchman 28de:2101
Info: Initializing Filter:
Info: 	report-ignore-start: 0
Info: 	light-required-obs: 10
Info: 	use-adaptive-imu: 0
Info: 	use-adaptive-lightcap: 0
Info: 	use-adaptive-obs: 0
Info: 	min-report-time: +0.005000
Info: 	light-error-threshold: +0.100000
Info: 	light-ignore-threshold: +1.000000
Info: 	report-ignore-threshold: +1.000000
Info: 	imu-acc-variance: +0.000050
Info: 	imu-gyro-variance: +0.010000
Info: 	obs-pos-variance: +0.020000
Info: 	obs-rot-variance: +0.010000
Info: 	light-variance: +0.000001
Info: 	process-weight-acc: +10.000000
Info: 	process-weight-vel: +0.000000
Info: 	process-weight-pos: +0.000000
Info: 	process-weight-ang-vel: +1.000000
Info: 	process-weight-rot: +0.000000
Info: Tracker config for WM0
Info: 	min-report-time                  5.000000e-03
Info: 	light-error-threshold            1.000000e-01
Info: 	light-ignore-threshold           1.000000e+00
Info: 	report-ignore-threshold          1.000000e+00
Info: 	imu-acc-variance                 5.000000e-05
Info: 	imu-gyro-variance                1.000000e-02
Info: 	obs-pos-variance                 2.000000e-02
Info: 	obs-rot-variance                 1.000000e-02
Info: 	light-variance                   1.000000e-06
Info: 	process-weight-acc               1.000000e+01
Info: 	process-weight-vel               0.000000e+00
Info: 	process-weight-pos               0.000000e+00
Info: 	process-weight-ang-vel           1.000000e+00
Info: 	process-weight-rot               0.000000e+00
Info: Adding tracked object WM0 from HTC
Info: Available Posers:
Info: 	 Dummy
Info: 	 BaryCentricSVD
Info: 	*MPFIT
Info: 	 KalmanOnly
Info: Totals 4 Posers.
Info: Using 'MPFIT' for poser
Info: Attaching IMU/Lightcap/Buttons(0x81) for WM0
Info: All enumerated devices attached.
Info: Powered unit on.
Info: Loaded drivers: HTCVive
Info: Position found for LH 0(ID: 8644a1e9, mode:  2) -0.000000e+00   +8.388930e-01   +1.915680e-01	-1.024460e-01   -1.084970e-01   +7.269837e-01   +6.702447e-01
[... 70 lines cut ...]
Info: WM0 Unknown flag 0x40 bytes dropping rest of data 30 00 00 80 f4 17 82 1a 63 
Info: WM0 Unknown flag 0x40 bytes dropping rest of data 30 00 00 80 bc 0b 32 64 65 
Info: WM0 Unknown flag 0x40 bytes dropping rest of data 30 00 00 80 18 80 90 b7 2b bc 6d d7 23 a8 78 16 e0 1f b8 78 af 2f 61 5b 6f 
Info: WM0 Unknown flag 0x40 bytes dropping rest of data 30 00 00 98 87 02 ff 05 72 
Info: WM0 Unknown flag 0x40 bytes dropping rest of data 30 00 00 90 42 12 5a bc 04 b0 01 b8 d1 03 9f 01 ed 65 0b d0 23 0b 24 75 
Info: WM0 Unknown flag 0x40 bytes dropping rest of data 30 00 00 58 6a 40 60 c0 2b 98 33 09 c6 c5 07 d1 01 a4 02 c4 08 45 d8 77 
Info: WM0 Unknown flag 0x40 bytes dropping rest of data 30 00 00 98 c2 04 f8 da 77 
Info: WM0 Unknown flag 0x40 bytes dropping rest of data 30 00 00 18 b2 2f 59 44 7b 
Info: Discard WM0 1: '18 60 18 60 db 1f a5 7d 16 c1 2b a9 6d d6 23 a9 78 16 fd 17 b3 6d 87 '
Info: Discard WM0 1: '18 60 b5 2f c4 6d 16 eb 1b 4f 8a 8d '
Info: Discard WM0 1: 'a0 80 18 60 a8 33 d9 68 16 bd 27 ff 70 e5 1f d9 09 09 ed 0c 34 3e 96 '
Info: Discard WM0 1: '60 90 60 90 a7 33 b7 65 d1 23 ac 78 16 e3 1f e2 78 ad 2f cd 2f a0 '
Info: Discard WM0 1: 'a0 60 8c 01 c9 1b c9 01 d6 21 a1 '
Info: Discard WM0 1: '10 60 18 40 a0 05 ff 65 0b ec 1b f2 7c a1 33 b6 3f 09 af 08 77 b0 a8 '
Info: Discard WM0 1: '18 90 d9 0a bf 1f 0d cc 27 c0 14 ac '
Info: Discard WM0 1: 'a1 82 11 80 e3 1f 9a 7a 08 e1 05 8c 05 c4 04 f8 01 80 0b 5b a8 ae '
Info: Discard WM0 1: '8a 38 d9 03 9f 07 8a 06 a2 b4 ae '
Info: Discard WM0 1: '60 90 00 18 a0 33 b7 65 d6 23 ab 78 16 fa 17 c6 00 01 b0 2f f6 99 b8 '
Info: Discard WM0 1: '98 60 90 98 82 02 cb 18 0c ea 1b e7 7c a1 33 8f 41 09 b2 04 71 19 c1 '
Info: Discard WM0 1: '80 18 b3 2f 81 69 e0 1f 33 c7 c4 '
Info: Discard WM0 1: '90 00 bb 0b c7 06 0e b3 2b 88 9b ca '
Info: WM0 Unknown flag 0x40 bytes dropping rest of data 80 61 
Info: WM0 Unknown flag 0x40 bytes dropping rest of data 80 61 
Info: WM0 Unknown flag 0x40 bytes dropping rest of data 80 61 
```

jdavidberger, 2020-08-26:
@hortejak latest specifically uses python 3

@ash891228 for whatever reason that tracker has weird usb reports. Can you make a usb recording of it?

CoCa520, 2021-12-26:
what's the problem
/home/fang/.config/libsurvive/config.json
(  0.000) SimpleApi: libsurvive version v0.3-300-gf5d1eb3-dirty
(  0.000) SimpleApi: Config file is /home/fang/.config/libsurvive/config.json
(  0.000) SimpleApi: Args: 
(  0.000) SimpleApi: 	'./bin/api_example'
(  0.000) SimpleApi: 	'--v'
(  0.000) SimpleApi: 	'100'
(  0.000) SimpleApi: 	lh-light-variance: -0.010000
(  0.000) SimpleApi: 	lh-light-stationary-time: +0.100000
(  0.000) SimpleApi: 	lh-light-stationary-maxtime: +2.000000
(  0.000) SimpleApi: 	lh-light-variance: -0.010000
(  0.000) SimpleApi: 	lh-light-stationary-time: +0.100000
(  0.000) SimpleApi: 	lh-light-stationary-maxtime: +2.000000
(  0.000) SimpleApi: 	lh-light-variance: -0.010000
(  0.000) SimpleApi: 	lh-light-stationary-time: +0.100000
(  0.000) SimpleApi: 	lh-light-stationary-maxtime: +2.000000
(  0.000) SimpleApi: 	lh-light-variance: -0.010000
(  0.000) SimpleApi: 	lh-light-stationary-time: +0.100000
(  0.000) SimpleApi: 	lh-light-stationary-maxtime: +2.000000
(  0.000) SimpleApi: 	lh-light-variance: -0.010000
(  0.000) SimpleApi: 	lh-light-stationary-time: +0.100000
(  0.000) SimpleApi: 	lh-light-stationary-maxtime: +2.000000
(  0.000) SimpleApi: 	lh-light-variance: -0.010000
(  0.000) SimpleApi: 	lh-light-stationary-time: +0.100000
(  0.000) SimpleApi: 	lh-light-stationary-maxtime: +2.000000
(  0.000) SimpleApi: 	lh-light-variance: -0.010000
(  0.000) SimpleApi: 	lh-light-stationary-time: +0.100000
(  0.000) SimpleApi: 	lh-light-stationary-maxtime: +2.000000
(  0.000) SimpleApi: 	lh-light-variance: -0.010000
(  0.000) SimpleApi: 	lh-light-stationary-time: +0.100000
(  0.000) SimpleApi: 	lh-light-stationary-maxtime: +2.000000
(  0.000) SimpleApi: 	lh-light-variance: -0.010000
(  0.000) SimpleApi: 	lh-light-stationary-time: +0.100000
(  0.000) SimpleApi: 	lh-light-stationary-maxtime: +2.000000
(  0.000) SimpleApi: 	lh-light-variance: -0.010000
(  0.000) SimpleApi: 	lh-light-stationary-time: +0.100000
(  0.000) SimpleApi: 	lh-light-stationary-maxtime: +2.000000
(  0.000) SimpleApi: 	lh-light-variance: -0.010000
(  0.000) SimpleApi: 	lh-light-stationary-time: +0.100000
(  0.000) SimpleApi: 	lh-light-stationary-maxtime: +2.000000
(  0.000) SimpleApi: 	lh-light-variance: -0.010000
(  0.000) SimpleApi: 	lh-light-stationary-time: +0.100000
(  0.000) SimpleApi: 	lh-light-stationary-maxtime: +2.000000
(  0.000) SimpleApi: 	lh-light-variance: -0.010000
(  0.000) SimpleApi: 	lh-light-stationary-time: +0.100000
(  0.000) SimpleApi: 	lh-light-stationary-maxtime: +2.000000
(  0.000) SimpleApi: 	lh-light-variance: -0.010000
(  0.000) SimpleApi: 	lh-light-stationary-time: +0.100000
(  0.000) SimpleApi: 	lh-light-stationary-maxtime: +2.000000
(  0.000) SimpleApi: 	lh-light-variance: -0.010000
(  0.000) SimpleApi: 	lh-light-stationary-time: +0.100000
(  0.000) SimpleApi: 	lh-light-stationary-maxtime: +2.000000
(  0.000) SimpleApi: 	lh-light-variance: -0.010000
(  0.000) SimpleApi: 	lh-light-stationary-time: +0.100000
(  0.000) SimpleApi: 	lh-light-stationary-maxtime: +2.000000
(  0.001) SimpleApi: Available Posers:
(  0.001) SimpleApi: 	 Dummy
(  0.001) SimpleApi: 	 KalmanOnly
(  0.001) SimpleApi: 	 BaryCentricSVD
(  0.001) SimpleApi: 	*MPFIT
(  0.001) SimpleApi: Totals 4 Posers.
(  0.001) SimpleApi: Using 'MPFIT' for poser
(  0.001) SimpleApi: Available Disambiguators:
(  0.001) SimpleApi: 	*StateBased
(  0.001) SimpleApi: Totals 1 Disambiguators.
(  0.001) SimpleApi: Using 'StateBased' for disambiguator
(  0.001) SimpleApi: 	usb-hz-output: -1
(  0.008) SimpleApi: Device added 0x56392a4a59a0
(  0.008) SimpleApi: Device added 0x56392a4a5410
(  0.008) SimpleApi: Device added 0x56392a4a5350
(  0.008) SimpleApi: Device added 0x56392a4987b0
(  0.009) SimpleApi: Device added 0x56392a4986f0
(  0.009) SimpleApi: Device added 0x56392a4a2490
(  0.009) SimpleApi: Enumerating USB device 28de:2101 Watchman
(  0.009) SimpleApi: Successfully enumerated Watchman (1) 28de:2101 at 0.0085809
(  0.009) SimpleApi: Requesting config for Watchman (nil) 0
(  0.009) SimpleApi: Device added 0x56392a49baf0
(  0.009) SimpleApi: Enumerating USB device 28de:2101 Watchman
(  0.009) SimpleApi: Successfully enumerated Watchman (1) 28de:2101 at 0.0086520
(  0.009) SimpleApi: Requesting config for Watchman (nil) 0
(  0.009) SimpleApi: Device added 0x56392a499a80
(  0.009) SimpleApi: Enumerating USB device 0bb4:2c87 BRD
(  0.009) SimpleApi: Successfully enumerated BRD (1) 0bb4:2c87 at 0.0087159
(  0.009) SimpleApi: Mainboard class 0
(  0.009) SimpleApi: Submitting magic vive_magic_power_on at 0.000001 sec for BRD - BRD
(  0.009) SimpleApi: Requesting config for BRD (nil) 1
(  0.009) SimpleApi: Device added 0x56392a49ce90
(  0.009) SimpleApi: Device added 0x56392a4981e0
(  0.009) SimpleApi: Enumerating USB device 0bb4:2c87 BRD
(  0.073) SimpleApi: Successfully enumerated BRD (2) 0bb4:2c87 at 0.0730679
(  0.073) SimpleApi: Mainboard class 239
(  0.073) SimpleApi: Device added 0x56392a498890
(  0.073) SimpleApi: Enumerating USB device 28de:2000 HMD IMU & LH
(  0.073) SimpleApi: Successfully enumerated HMD IMU & LH (2) 28de:2000 at 0.0733929
(  0.073) SimpleApi: Requesting config for HMD IMU & LH (nil) 0
Segmentation fault (core dumped)

zghrairi, 2024-03-15:
hi, I'm connecting the tracker per USB to the VM. After running survive-cli, i'm receiving the same message repeated
Warning: Config request failed, trying again ..  T20 

I need you support.