collabora/libsurvive #200: Almost always fails to parse data with 2 index controllers

ChristophHaag, 2020-07-17:
Using a full Index Set with RF connected Index controllers and 2 Lighthouse v2 basestations. I remember this working quite well but for several months it has only been working well with 1 controller. When powering on the second controller, libsurvive almost always fails to parse data and spams unparsed data to stdout.

usbmon recording: [logs.zip](https://github.com/cntools/libsurvive/files/4938072/logs.zip)

If you restart it often enough, sometimes two controllers work fine.

jdavidberger, 2020-07-27:
The issue is that there are a few flags we don't know the lengths of, and there was a bug where it'd read off the end of the data packet. The packets tend to only happen at startup, or at least infrequently so I made it just bail early but also made it not read into invalid memory which it was doing. This should help; hopefully it doesn't drop a bunch of lightcaps unnecessarily though.

ChristophHaag, 2020-08-17:
New log with master 6debd6d84ef36a32a87d26d41de1aade0ebacfd3
[logs.zip](https://github.com/cntools/libsurvive/files/5085298/logs.zip)

ChristophHaag, 2020-08-17:
longer recording [logs.zip](https://github.com/cntools/libsurvive/files/5085367/logs.zip)

jdavidberger, 2020-08-17:
driver_vive.c:284; `MAGIC_CTOR(true, vive_magic_rf_raw_mode_0),` seems to be the issue

closed 2020-08-18 by jdavidberger