collabora/libsurvive #202: Makefile

hortejak, 2020-08-17:
Hi,
I am trying to run the latest commit, but it refuses to make. First it called `unknown argument -j` and `unknown argument 4`. I tried to delete those flags in makefile, it runs fine for a while up to I think 84%, then it finds undefined referencies in `simple_pose_test.c`.
```
[ 84%] Linking C executable simple_pose_test
CMakeFiles/simple_pose_test.dir/simple_pose_test.c.o: In function `DrawLineSegment':
simple_pose_test.c:(.text+0x588): undefined reference to `tdFinalPoint'
simple_pose_test.c:(.text+0x59b): undefined reference to `tdFinalPoint'
simple_pose_test.c:(.text+0x634): undefined reference to `CNFGTackPixel'
simple_pose_test.c:(.text+0x68d): undefined reference to `CNFGTackSegment'
CMakeFiles/simple_pose_test.dir/simple_pose_test.c.o: In function `GUIThread':
simple_pose_test.c:(.text+0x6fc): undefined reference to `CNFGSetup'
simple_pose_test.c:(.text+0x753): undefined reference to `CNFGClearFrame'
simple_pose_test.c:(.text+0x75d): undefined reference to `CNFGHandleInput'
simple_pose_test.c:(.text+0x767): undefined reference to `CNFGColor'
simple_pose_test.c:(.text+0x77a): undefined reference to `CNFGGetDimensions'
simple_pose_test.c:(.text+0x87f): undefined reference to `tdCross'
simple_pose_test.c:(.text+0x896): undefined reference to `tdCross'
simple_pose_test.c:(.text+0x8a2): undefined reference to `tdNormalizeSelf'
simple_pose_test.c:(.text+0x8df): undefined reference to `tdSetViewport'
simple_pose_test.c:(.text+0x8e9): undefined reference to `tdMode'
simple_pose_test.c:(.text+0x8f0): undefined reference to `gSMatrix'
simple_pose_test.c:(.text+0x8fb): undefined reference to `tdIdentity'
simple_pose_test.c:(.text+0x902): undefined reference to `gSMatrix'
simple_pose_test.c:(.text+0x942): undefined reference to `tdPerspective'
simple_pose_test.c:(.text+0x94c): undefined reference to `tdMode'
simple_pose_test.c:(.text+0x953): undefined reference to `gSMatrix'
simple_pose_test.c:(.text+0x95e): undefined reference to `tdIdentity'
simple_pose_test.c:(.text+0x965): undefined reference to `gSMatrix'
simple_pose_test.c:(.text+0x97c): undefined reference to `tdLookAt'
simple_pose_test.c:(.text+0x986): undefined reference to `CNFGColor'
simple_pose_test.c:(.text+0x9dd): undefined reference to `CNFGColor'
simple_pose_test.c:(.text+0xa34): undefined reference to `CNFGColor'
simple_pose_test.c:(.text+0xa8b): undefined reference to `CNFGColor'
simple_pose_test.c:(.text+0xab6): undefined reference to `CNFGColor'
CMakeFiles/simple_pose_test.dir/simple_pose_test.c.o:simple_pose_test.c:(.text+0xae1): more undefined references to `CNFGColor' follow
CMakeFiles/simple_pose_test.dir/simple_pose_test.c.o: In function `GUIThread':
simple_pose_test.c:(.text+0xb33): undefined reference to `CNFGSwapBuffers'
collect2: error: ld returned 1 exit status
CMakeFiles/simple_pose_test.dir/build.make:97: recipe for target 'simple_pose_test' failed
make[3]: *** [simple_pose_test] Error 1
```
Do I need to install something first? I did everything mentioned in `README.md`.
Thanks for feedback,
hortejak.

jdavidberger, 2020-08-17:
Should be fixed in latest

hortejak, 2020-08-18:
Makefile works just fine, thanks for the quick feedback. This issue can be closed.

closed 2020-08-18 by jdavidberger