collabora/libsurvive #203: Python bindings

hortejak, 2020-08-20:
Hi,
got a little question. I have tried (as is is the guide in `README.md`) set up python bindings and try it on `example.py`, but it misses module `pysurvive.pysurvive_generated`. Do I need to download something first or did it just not install well?
```
Traceback (most recent call last):
  File "bindings/python/graph-example.py", line 1, in <module>
    import pysurvive
  File "/home/hortejak/f2t_new/libsurvive/bindings/python/pysurvive/__init__.py", line 5, in <module>
    import pysurvive.pysurvive_generated
ModuleNotFoundError: No module named 'pysurvive.pysurvive_generated'
```

jdavidberger, 2020-08-20:
You need ctypesgen; I'll update the CI / build for it. In the meanwhile:

```
pip install ctypesgen
python setup.py clean install
```

That should get you up and running

hortejak, 2020-08-24:
Hi, I have still the same issue with the old version (commit `13c34f0`). I pulled the newest version (commit `f2ce25f`) and it refuses to even install, it produces output like this:
```
<_io.TextIOWrapper name='/home/hortejak/f1t_bind/libsurvive/.eggs/ctypesgen-1.0.2-py3.8.egg/ctypesgen/ctypesgen' mode='w' encoding='UTF-8'> ctypesgen 1.0.2
running clean
removing 'build/temp.linux-x86_64-3.8' (and everything under it)
running install
running bdist_egg
running egg_info
writing pysurvive.egg-info/PKG-INFO
writing dependency_links to pysurvive.egg-info/dependency_links.txt
writing top-level names to pysurvive.egg-info/top_level.txt
reading manifest file 'pysurvive.egg-info/SOURCES.txt'
writing manifest file 'pysurvive.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Looking for gattlib.h
-- Looking for gattlib.h - not found
-- Looking for zlib.h
-- Looking for zlib.h - found
-- Looking for X11/Xlib.h
-- Looking for X11/Xlib.h - found
-- Looking for fopencookie
-- Looking for fopencookie - found
CMake Warning at CMakeLists.txt:128 (find_package):
  By not providing "FindPython.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Python", but
  CMake did not find one.

  Could not find a package configuration file provided by "Python" with any
  of the following names:

    PythonConfig.cmake
    python-config.cmake

  Add the installation prefix of "Python" to CMAKE_PREFIX_PATH or set
  "Python_DIR" to a directory containing one of the above files.  If "Python"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Building python bindings file to "/home/hortejak/f1t_bind/libsurvive/bindings/python/pysurvive/"
-- ctypesgen found at /home/hortejak/f1t_bind/libsurvive/.eggs/ctypesgen-1.0.2-py3.8.egg/ctypesgen/ctypesgen
-- Configuring done
-- Generating done
-- Build files have been written to: /home/hortejak/f1t_bind/libsurvive/build/temp.linux-x86_64-3.8
Scanning dependencies of target mpfit
Scanning dependencies of target minimal_opencv
Scanning dependencies of target CNGFX
[... 121 lines cut ...]
  File "/usr/lib/python3.8/distutils/command/install_lib.py", line 109, in build
    self.run_command('build_ext')
  File "/usr/lib/python3.8/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/hortejak/.local/lib/python3.8/site-packages/setuptools/command/build_ext.py", line 87, in run
    _build_ext.run(self)
  File "/usr/lib/python3.8/distutils/command/build_ext.py", line 340, in run
    self.build_extensions()
  File "/usr/lib/python3.8/distutils/command/build_ext.py", line 449, in build_extensions
    self._build_extensions_serial()
  File "/usr/lib/python3.8/distutils/command/build_ext.py", line 474, in _build_extensions_serial
    self.build_extension(ext)
  File "/home/hortejak/f1t_bind/libsurvive/.eggs/cmake_setuptools-0.1.3-py3.8.egg/cmake_setuptools/cmake/__init__.py", line 57, in build_extension
    subprocess.check_call(['make', '-j', ext.name],
  File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '-j', 'all']' returned non-zero exit status 2.
```
Both installations say, that `gattlib.h not found` but only the newest version (commit `f2ce25f`) claims, that `could not find PythonConfig.cmake` and `could not find python-config.cmake` 
Just a side note, the newest version (commit `f2ce25f`) refuses to even calibrate, it does not create `config.json`. Commit `13c34f0` calibrates fine.
Lastly I am using python3.8, is it compatible with this one or do I need to use older versions?
Thanks,
hortejak

jdavidberger, 2020-08-24:
Python3.8 should be fine; I'm using 3.7.4 but there is no real difference for these purposes.

The `gattlib.h` thing is fine too; it would build an additional driver if you had it but you don't need it.

It might be your cmake is too out of date -- as seen here: https://github.com/jrl-umi3218/copra/issues/6. Can you upgrade your cmake and try again? If it is that, I will update the version requirement.

hortejak, 2020-08-24:
Okay, I updated the cmake from 3.10 to 3.18.2 and then I had to rewrite all `sys.maxint` to `sys.maxsize` (maxint is for python2, maxsize for python3). After that it seems to work pretty fine. I would open an issue, but when we are here ... flag --force-calibrate does not seem to work. I cannot recalibrate the new version (I have always just deleted `config.json` and I cannot find it now). I know it is in `XDG_CONFIG_HOME/libsurvive` but ...
Thanks,
hortejak

jdavidberger, 2020-08-24:
Where did you have to do those updates? All my generated stuff is `maxsize` which makes me worried something went wrong in the generation process. Possibly it found and used python 2. 

I'm not sure what you mean by it doesn't work. Are there no config files in that folder? Is it just reusing old calibration? Does it get anywhere at all? Is it a problem in just python or for the `survive-cli` too?

hortejak, 2020-08-25:
Okay, I do have python 2.7 as well on my computer, so it is possible it took something from it. I had to rewrite the `bindings/python/pysurvive/__init__,py`,there it is something like 5 times. Well first calibration went well, then I moved Base Stations and wanted to recalibrate. With the flag `--force-calibrate` it does print that this flag is set and instead of looking for LHs, it just freezes. This is all I get from `./bin/survive-cli --force-calibrate`:
```
Info: Adding tracked object WM0 from HTC
Info: All enumerated devices attached.
Info: Powered unit on.
Info: Loaded drivers: HTCVive
Info: Force calibrate flag set -- clearing position on all lighthouses
Info: Detected LH gen 2 system.
Info: LightcapMode (WM0) 1 -> 2
```
I found out, that even with the flag acknowledged, it is behaving like no calibration should be set, I calibrated with 3 LHs and now I am using just two, this is a production of `./bin/api_example --force-calibrate --v 100`:
```
SimpleApi: Config file is /home/hortejak/.config/libsurvive/config.json
SimpleApi: Adding LH 1 mode: 1 id: 647ecae5
SimpleApi: Adding LH 2 mode: 2 id: b424bcda
SimpleApi: Adding LH 3 mode: 0 id: 60ec8de6
SimpleApi: Available Posers:
SimpleApi: 	*MPFIT
SimpleApi: 	 Dummy
SimpleApi: 	 BaryCentricSVD
SimpleApi: 	 KalmanOnly
SimpleApi: Totals 4 Posers.
SimpleApi: Using 'MPFIT' for poser
SimpleApi: Available Disambiguators:
SimpleApi: 	*StateBased
SimpleApi: Totals 1 Disambiguators.
SimpleApi: Using 'StateBased' for disambiguator
SimpleApi: 	usb-hz-output: -1
SimpleApi: Successfully enumerated Watchman (1) 28de:2101
SimpleApi: Successfully enumerated Watchman 28de:2101
SimpleApi: 	report-ignore-start: 0
SimpleApi: 	light-required-obs: 10
SimpleApi: 	use-adaptive-imu: 0
SimpleApi: 	use-adaptive-lightcap: 0
SimpleApi: 	use-adaptive-obs: 0
SimpleApi: 	min-report-time: +0.005000
SimpleApi: 	light-error-threshold: +0.100000
SimpleApi: 	light-ignore-threshold: +1.000000
SimpleApi: 	report-ignore-threshold: +1.000000
SimpleApi: 	imu-acc-variance: +0.000050
SimpleApi: 	imu-gyro-variance: +0.010000
SimpleApi: 	obs-pos-variance: +0.020000
SimpleApi: 	obs-rot-variance: +0.010000
SimpleApi: 	light-variance: +0.000001
SimpleApi: 	process-weight-acc: +10.000000
SimpleApi: 	process-weight-vel: +0.000000
SimpleApi: 	process-weight-pos: +0.000000
SimpleApi: 	process-weight-ang-vel: +1.000000
SimpleApi: 	process-weight-rot: +0.000000
SimpleApi: Tracker config for WM0
SimpleApi: 	min-report-time                  5.000000e-03
SimpleApi: 	light-error-threshold            1.000000e-01
SimpleApi: 	light-ignore-threshold           1.000000e+00
SimpleApi: 	report-ignore-threshold          1.000000e+00
SimpleApi: 	imu-acc-variance                 5.000000e-05
SimpleApi: 	imu-gyro-variance                1.000000e-02
SimpleApi: 	obs-pos-variance                 2.000000e-02
SimpleApi: 	obs-rot-variance                 1.000000e-02
SimpleApi: 	light-variance                   1.000000e-06
SimpleApi: 	process-weight-acc               1.000000e+01
SimpleApi: 	process-weight-vel               0.000000e+00
SimpleApi: 	process-weight-pos               0.000000e+00
SimpleApi: 	process-weight-ang-vel           1.000000e+00
SimpleApi: 	process-weight-rot               0.000000e+00
SimpleApi: Adding tracked object WM0 from HTC
SimpleApi: Available Posers:
SimpleApi: 	*MPFIT
SimpleApi: 	 Dummy
SimpleApi: 	 BaryCentricSVD
SimpleApi: 	 KalmanOnly
SimpleApi: Totals 4 Posers.
SimpleApi: Using 'MPFIT' for poser
SimpleApi: Attaching IMU/Lightcap/Buttons(0x81) for WM0
SimpleApi: All enumerated devices attached.
SimpleApi: Powered unit on.
SimpleApi: Loaded drivers: HTCVive
SimpleApi: Force calibrate flag set -- clearing position on all lighthouses
Found 'WM0'
Found 'LH0'
Found 'LH1'
Found 'LH2'
SimpleApi: Got config data length 2194 for Watchman:0
SimpleApi: Read config for WM0
SimpleApi: Gen2 reason: WM0 Lightcap length >= 0x8000
SimpleApi: Detected LH gen 2 system.
SimpleApi: Invalid sensor 255 detected hit (24)
SimpleApi: LightcapMode (WM0) 1 -> 2
SimpleApi: 	max-error: +0.010000
SimpleApi: 	failures-to-reset: -1
SimpleApi: 	successes-to-reset: -1
SimpleApi: 	disable-lighthouse: -1
SimpleApi: 	sensor-variance-per-sec: +0.000000
SimpleApi: 	sensor-variance: +1.000000
```
It is quite long but it says it found 3 LHs, when just 2 are functional. I mentioned that it has `config.json` file path, so I deleted in manually and it calibrated nicely.