collabora/libsurvive #205: Sometimes "Segmentation faults"

807183087, 2020-08-24:
**Describe the bug**
I added some callback functions to original code, compiled libsurvive to .a .so , then i linked them to my c++ project. 
Everything goes okay, but sometimes it occurs "Segmentation faults", then the process stoped.

 - OS: Debian RaspberryPi
 - libsurvive Version: i tested two versions ( 2020.08.21 2060a85 and v0.1 )

**Additional context**
i attached the modified libsurvive and my project below, you can find my changes by finding "litao".

jdavidberger, 2020-08-24:
There is a lot going on in those modifications. What you probably need to do is install gdb to the raspberry-pi and run it through there; preferably as a debug release. This should give you a stack trace of the actual crash. It might also be worthwhile to turn on the `USE_ASAN` option. 

Those error numbers look pretty bad; so it wouldn't surprise me if it was going way off in tracking and that was the root of the issue. If it's an assert going off you could build as Release mode; but you will still have a problem with tracking in general.

I don't know exactly how you are sending the data; but a major warning for the general approach here: Those callbacks happen in the same thread as the raw data capture. If you are injecting a lot of latencies into that thread, it will drop data and give bad results. If you are sending that captured data out over a TCP socket, I think it'd be very likely to have random bouts of latency that would damage tracking output. The approach I'd recommend here is to copy out the data and use a semaphore to notify another thread. It's possible that is what you are doing -- NativeSendData is not defined in main.cpp. In theory the dynamic memory allocation in those callbacks could cause similar issues; but in practice it'd take a while to notice. 

Out of curiosity; what is the rationale for the custom callbacks? For the most part they seem duplicates of ones in the library; and I wasn't sure if there was a use case around that that could be better handled by libsurvive.

807183087, 2020-08-26:
Thanks jdavidberger.

My raspberryPi version is 3b+ (1.4GHz、4core) .
I found survive cost much cpu，maybe that is the reason.
I will run survive in other high performance hardware.

jdavidberger, 2020-08-26:
Maybe; but that is well within the specs of what libsurvive should run at and performance problems in general shouldn't cause seg faults. I think libsurvive running at full tilt on that system should only take like 30-40% of a core but I'm not sure what other processing you are doing.

closed 2020-08-26 by jdavidberger