collabora/libsurvive #207: Compilation failures: redefinitions, undeclared usages

PureTryOut, 2020-09-16:
**Describe the bug**
I have multiple compilation failures happening on a Musl based system.

**Compile log**

This is using GCC 10.2.0 and Musl 1.2.1.
```
/home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-0.2/redist/symbol_enumerator.c:47:8: error: redefinition of 'struct dl_phdr_info'
   47 | struct dl_phdr_info {
      |        ^~~~~~~~~~~~
In file included from /home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-0.2/redist/symbol_enumerator.c:40:
/usr/include/link.h:21:8: note: originally defined here
   21 | struct dl_phdr_info {
      |        ^~~~~~~~~~~~
/home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-0.2/redist/symbol_enumerator.c:57:6: error: conflicting types for 'dl_iterate_phdr'
   57 | void dl_iterate_phdr( void*, void*);
      |      ^~~~~~~~~~~~~~~
In file included from /home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-0.2/redist/symbol_enumerator.c:40:
/usr/include/link.h:47:5: note: previous declaration of 'dl_iterate_phdr' was here
   47 | int dl_iterate_phdr(int (*)(struct dl_phdr_info *, size_t, void *), void *);
      |     ^~~~~~~~~~~~~~~
/home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-0.2/redist/symbol_enumerator.c: In function 'gnu_hashtab_symbol_count':
/home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-0.2/redist/symbol_enumerator.c:43:25: error: '__WORDSIZE' undeclared (first use in this function)
   43 | #define UINTS_PER_WORD (__WORDSIZE / (CHAR_BIT * sizeof (unsigned int)))
      |                         ^~~~~~~~~~
/home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-0.2/redist/symbol_enumerator.c:62:78: note: in expansion of macro 'UINTS_PER_WORD'
   62 |     const unsigned int *const bucket = table + 4 + table[2] * (unsigned int)(UINTS_PER_WORD);
      |                                                                              ^~~~~~~~~~~~~~
/home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-0.2/redist/symbol_enumerator.c:43:25: note: each undeclared identifier is reported only once for each function it appears in
   43 | #define UINTS_PER_WORD (__WORDSIZE / (CHAR_BIT * sizeof (unsigned int)))
      |                         ^~~~~~~~~~
/home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-0.2/redist/symbol_enumerator.c:62:78: note: in expansion of macro 'UINTS_PER_WORD'
   62 |     const unsigned int *const bucket = table + 4 + table[2] * (unsigned int)(UINTS_PER_WORD);
      |                                                                              ^~~~~~~~~~~~~~
```
**Desktop (please complete the following information):**
 - OS: Alpine Linux
 - Version: edge

**Additional context**
I'm trying to build version 0.2 of libsurvive.

jdavidberger, 2020-09-16:
If this is a publicly available docker image; can you point me at it? If not, can you give me the output of:

`cat /etc/os-release`

and 

`apk info`?

PureTryOut, 2020-09-17:
This is not being done on Docker, I'm doing it with the usual Alpine Linux packaging tools, intending to get it in the Alpine repositories itself.

Because of this `apk info` is useless as that only gives information about the installed packages on my host, while the building happens in an `abuild` clean root (`abuild rootbld`). Same for `/etc/os-release`, it's not useful info.

```
✔︎ ~ cat /etc/os-release
NAME="Alpine Linux"
ID=alpine
VERSION_ID=3.13.0_alpha20200917
PRETTY_NAME="Alpine Linux edge"
HOME_URL="https://alpinelinux.org/"
BUG_REPORT_URL="https://bugs.alpinelinux.org/"
```

jdavidberger, 2020-09-17:
I'll try that out. I ran into package issues with blas on the latest version so I wanted to get by those issues. I'm not overly familiar with their packaging system but if you have a script/config you can share I'll make it work with that; otherwise I'll just get 3.13 and make it work with that.

PureTryOut, 2020-09-17:
I've made this APKBUILD, https://gitlab.alpinelinux.org/PureTryOut/aports/-/blob/libsurvive/testing/libsurvive/APKBUILD

As you can see, just regular CMake commands.

jdavidberger, 2020-09-19:
The latest master has fixes for this issue. You'll also want libusb-dev as a dependency. In theory you don't need the main vive driver to use the library -- you can build just for playbacks or for simulation -- but people will probably want it. 

I did run a linker issue:

```
/usr/lib/gcc/x86_64-alpine-linux-musl/10.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: /usr/lib/gcc/x86_64-alpine-linux-musl/10.2.0/../../../../lib/liblapacke.so: undefined reference to `LAPACK_zggsvp'
/usr/lib/gcc/x86_64-alpine-linux-musl/10.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: /usr/lib/gcc/x86_64-alpine-linux-musl/10.2.0/../../../../lib/liblapacke.so: undefined reference to `LAPACK_sggsvd'
/usr/lib/gcc/x86_64-alpine-linux-musl/10.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: /usr/lib/gcc/x86_64-alpine-linux-musl/10.2.0/../../../../lib/liblapacke.so: undefined reference to `LAPACK_zggsvd'
/usr/lib/gcc/x86_64-alpine-linux-musl/10.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: /usr/lib/gcc/x86_64-alpine-linux-musl/10.2.0/../../../../lib/liblapacke.so: undefined reference to `LAPACK_dgeqpf'
/usr/lib/gcc/x86_64-alpine-linux-musl/10.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: /usr/lib/gcc/x86_64-alpine-linux-musl/10.2.0/../../../../lib/liblapacke.so: undefined reference to `LAPACK_dggsvd'
/usr/lib/gcc/x86_64-alpine-linux-musl/10.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: /usr/lib/gcc/x86_64-alpine-linux-musl/10.2.0/../../../../lib/liblapacke.so: undefined reference to `LAPACK_sggsvp'
/usr/lib/gcc/x86_64-alpine-linux-musl/10.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: /usr/lib/gcc/x86_64-alpine-linux-musl/10.2.0/../../../../lib/liblapacke.so: undefined reference to `LAPACK_zgeqpf'
/usr/lib/gcc/x86_64-alpine-linux-musl/10.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: /usr/lib/gcc/x86_64-alpine-linux-musl/10.2.0/../../../../lib/liblapacke.so: undefined reference to `LAPACK_dggsvp'
/usr/lib/gcc/x86_64-alpine-linux-musl/10.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: /usr/lib/gcc/x86_64-alpine-linux-musl/10.2.0/../../../../lib/liblapacke.so: undefined reference to `LAPACK_sgeqpf'
/usr/lib/gcc/x86_64-alpine-linux-musl/10.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: /usr/lib/gcc/x86_64-alpine-linux-musl/10.2.0/../../../../lib/liblapacke.so: undefined reference to `LAPACK_cgeqpf'
/usr/lib/gcc/x86_64-alpine-linux-musl/10.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: /usr/lib/gcc/x86_64-alpine-linux-musl/10.2.0/../../../../lib/liblapacke.so: undefined reference to `LAPACK_cggsvd'
/usr/lib/gcc/x86_64-alpine-linux-musl/10.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: /usr/lib/gcc/x86_64-alpine-linux-musl/10.2.0/../../../../lib/liblapacke.so: undefined reference to `LAPACK_cggsvp'
```

which seems like a build / packaging issue with the alpine lapacke package?

PureTryOut, 2020-09-19:
> The latest master has fixes for this issue.

:tada: 

> In theory you don't need the main vive driver to use the library -- you can build just for playbacks or for simulation -- but people will probably want it.

Yeah the package is work in progress still, I first just want it compiling at all before I start adding more deps.

> I did run into a linker issue:

Hmm I can reproduce that issue, not sure what's happening there. Rebuilding lapack sadly doesn't help, I'll look into it more.

PureTryOut, 2020-09-21:
@jdavidberger: found the issue for the missing symbols, https://github.com/Reference-LAPACK/lapack/pull/367

The symbols are re-added in latest master of lapack, but as mentioned in the commit description the functions are deprecated. It might be good to port away from it.

I'm still working on getting this in Alpine. Once it's in, I'll close this issue.

PureTryOut, 2020-09-22:
Fixed, I'll be submitting the package to the Alpine repositories shortly.

Unrelated to this issue, but I'm now trying to get Monado building with libsurvive, but it fails to link to it.

```
/usr/lib/gcc/x86_64-alpine-linux-musl/10.2.0/../../../../x86_64-alpine-linux-musl/bin/ld: cannot find -lsurvive
collect2: error: ld returned 1 exit status
```

Should I make a new issue for it on this repository or on the Monado one?

closed 2020-09-22 by PureTryOut