collabora/libsurvive #209: Cannot find any valid Poser/Disambiguator. [LHT v2]

Virgile-Dauge, 2020-10-05:
**Describe the bug**
virgile@leon:~/libsurvive$ ./bin/survive-cli
Info: Error.  Cannot find any valid Poser.
Info: Error.  Cannot find any valid Disambiguator.

**Data**
No data.

**Compile log**
No compilation issue.

**Hardware setup**
- two Lighthouses V2 
- one tracker V2
(tried bot wired and via HTC dongle)

**Desktop (please complete the following information):**
 - OS: Ubuntu 
 - Version 18.04

**Additional context**
Steps to reproduce :

```git clone git@github.com:cntools/libsurvive.git
cd libsurvive
sudo cp ./useful_files/81-vive.rules to /etc/udev/rules.d/
sudo udevadm control --reload-rules && udevadm trigger
sudo apt-get install build-essential zlib1g-dev libx11-dev libusb-1.0-0-dev freeglut3-dev liblapacke-dev libopenblas-dev libatlas-base-dev cmake
sudo make
```
Plug in a headset / tracker / controller / etc and run:
```
./bin/survive-cli
```

jdavidberger, 2020-10-05:
Try latest now; should be resolved.

Virgile-Dauge, 2020-10-06:
Thanks, it fixed it.

closed 2020-10-06 by Virgile-Dauge